biemond/solaredge.modbus #1: Solar edge SE3000H Photovoltaic Inverter

yvoren, 2022-03-08:
Hello,

today they have installed my solar panels with a Solaredge SE3000H Inverter.
I want to make a connection with my Homey Pro, I have installed de app and I configured the parameters.
That all went ok, but in the app the value of SolarPower Heatsink Temperature and Inverter status does not show any value.
The solar edge  has a IP adres en the port is 1502. i can browse with a webbrowser the inverter.
What I'm doing wrong ?

biemond, 2022-03-08:
Hey, 

yes see https://community.homey.app/t/app-pro-solaredge/45487/250

I think upgrade does not work,  can you remove all the devices and add them again.

I optimised a lot and make it more   flexible and non blockable

thanks

yvoren, 2022-03-08:
Hey Edwin,
I removed all the devices.
I have also removed de solaredge app from my Homey. Then installed the latest version (again)
Then i have added a new inverter with the propper parameters. It was accepted , but still no result. Then i removed the device again, and installed a new one, stil no result. But what i discovered the added device name is different all the time but 
not my soloraedge SE3000H.
Hopefully you can solve this.

Met vriendelijke groet,

René Slisser.

biemond, 2022-03-08:
cool, I think from the app you can send a diagnostic log to me.

but the name changing is correct so you can more than 1 invertor. 

can you try unit 2 on your device. 

thanks

biemond, 2022-03-08:
and are you using 1502 or 502 as port number

yvoren, 2022-03-08:
I have tried both port numbers, with no result.

Met vriendelijke groet,

René Slisser.

biemond, 2022-03-08:
met de juiste settings , kun je na een paar minuten een diagnostics Report maken 

in homey , ga naar apps -> solaredge -> settings -> create Dia...   Ik krijg hem dan binnen.

gr

yvoren, 2022-03-08:
Hoi Edwin,

Hierbij diagnostische rapport code 3fafc914-44ff-4a6b-b65e-80c3988dcb4a

Met vriendelijke groet,

René Slisser.

biemond, 2022-03-08:
mooi , ik heb hem gezien. even kijken of ik iets kan zien

biemond, 2022-03-08:
lijkt dat ik hem niet kan bereiken.

```
[Device:5e2bc82f-d782-4532-90b0-faf7159eb4b6] 10.19.1.6
Error: connect ETIMEDOUT [10.19.1.6:1502](http://10.19.1.6:1502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1137:16) {
  errno: 'ETIMEDOUT',
  code: 'ETIMEDOUT',
  syscall: 'connect',
  address: '10.19.1.6',
  port: 1502
}
```

biemond, 2022-03-08:
toevalling een andere programma heeft de connectie.  

of doe
nc -z -v 10.19.1.6 1502

gebruik je wifi ?  er was een solaredge update waarmee ze modbus over wifi niet meer toelaten.

yvoren, 2022-03-09:
Hoi Edwin,

Ik heb nu een ethernet connectie, en nu werkt het.
Bedankt voor de support.
Modbus over wifi werkt dus idd niet
Groet

Verstuurd vanaf mijn iPad

yvoren, 2022-03-09:
Hoi Edwin,
Is er ook een mogelijkheid om de productie van vandaag en productie totaal te presenteren.

Met vriendelijke groet,

René Slisser.

biemond, 2022-03-09:
nice, ik zal eens kijken of ik dat betrouwbaar eruit kan halen en tonen, nu we eindelijk weer een beetje zon hebben.

biemond, 2022-03-13:
Ik heb de totale export toegevoegd , zie https://homey.app/nl-nl/app/solaredge.modbus/SolarEdge/test/

yvoren, 2022-03-13:
Top Edwin, super bedankt.

Is produktie vandaag ook nog mogelijk ?

Met vriendelijke groet,

René Slisser.

biemond, 2022-03-13:
Ik ga proberen door de 00:00 waarde op te slaan en dat dan te herleiden.

yvoren, 2022-03-14:
Hoi Edwin,

Is dit ook niet een gedefinieerd modbus adres welke dynamisch uit te lezen is ?

Met vriendelijke groet,

René Slisser.

biemond, 2023-04-03:
als je laatste update hebt dan moet modbus weer werken over wifi.

biemond, 2023-04-03:
maar volgens mij kun je altijd lan aansluiten. misschien moet de bescherming er wel tijdelijk af.

RoelvWijk, 2023-04-03:
Hoi Erwin,
Bedankt voor je snelle reactie.
Helaas krijg ik het niet aan de gang.
Ik heb de laatste versie (ten minste, die op de homey staat).
Heb net ook kan geprobeerd en werkt ook niet.
Er zit dus iets niet helemaal goed...

Hoe kom ik erachter... Heb al heel wat geprobeerd

Roel

biemond, 2023-04-03:
heb je de versie van je inverter gecheckt?

biemond, 2023-04-03:
https://www.solaredge.com/en/support/setapp-inverters-firmware
[image]

dit is de mijne , check cpu