biemond/solaredge.modbus #2: Status

Bakkerwj, 2022-03-16:
Graag de status weer in de Flow's..

[image]

biemond, 2022-03-16:
perfect, gaan we doen

biemond, 2022-03-17:
ben nu aan testen en wacht totdat het donker wordt voor een paar status veranderingen.  

ik stuur je later een link

Bakkerwj, 2022-03-17:
Super! Thanx..

Met vriendelijke groet,

Willem Jan Bakker

biemond, 2022-03-17:
kan je deze versie testen https://homey.app/nl-nl/app/solaredge.modbus/SolarEdge/test/

Bakkerwj, 2022-03-17:
Gedaan, maar kan morgen pas zien of t goed werkt. 

Met vriendelijke groet,

Willem Jan Bakker

biemond, 2022-03-17:
yep

biemond, 2022-03-17:
nog steeds producing voor mij

Bakkerwj, 2022-03-18:
Hij gaf vanmorgen keurig de melding Producing…

Alleen als ik NU een status opvraag geeft ie FAULT, hij geeft alleen de juiste status op t moment van wijzigen maar niet meer “tussendoor” als je daar om vraagt.

Met vriendelijke groet,

Willem Jan Bakker

biemond, 2022-03-18:
inderdaad,  dat is een trigger alleen als er iets wijzigt. anders wordt je gek en krijg je een trigger elke 20 seconden.  

maar wil je een ook condtie , zoiets   de zon schijnt en status = producing dan doe dit.

Bakkerwj, 2022-03-18:
“voorheen” kon ik met de flow iid bij een wijziging een status wijziging krijgen, maar als ik de “status” opvroeg kreeg ik m ook, dat wil nu niet meer.

Dus idd bijv, 10:00 uur status is producing, doe de zonneschermen naar beneden. oid..

Met vriendelijke groet,

Willem Jan Bakker
Boerestreek 1
8757JR Gaast
Mob. +31653217707

biemond, 2022-03-18:
mooi, dat ga ik toevoegen.

biemond, 2022-03-18:
je kan nu wel tijdelijk invertor is higher gebruiken , bijvoorbeeld meer dan 500

Bakkerwj, 2022-03-18:
Ja, Ik gebruik nu  ook “het vermogen is veranderd” EN "Solar+Batt power is groter dan 2500” 

En dergelijke flows, onder meer om met onze warmtepomp warmte te bufferen als er veel zon is..

Met vriendelijke groet,

Willem Jan Bakker
Boerestreek 1
8757JR Gaast
Mob. +31653217707

biemond, 2022-03-18:
nice

biemond, 2022-03-18:
kun je deze versie testen https://homey.app/a/solaredge.modbus/test/
[image]

Bakkerwj, 2022-03-28:
> interessant.
>
> zo hebben de 2 invertors echt een eigen id? Ik dacht dat ik dat ge-fixed had op de normale invertor , misschien op batt niet?
> ik zal nog eens hier naar kijken. batt en normale hebben wel een verschillende timing 18 vs 28 of zo. misschien kan ik nog een offset toevoegen..
>
> voor het laden gebruik je zeker een flow action en je zegt dat deze alleen werkt op de 1st invertor . zal er eens naar kijken, had verwacht dat de 2de invertor heeft dezelfde registers .
> wacht eens, je 2de invertor is zeker een normal invertor in de app . misschien moet ik doe control gewoon toelaten. het werkt of werkt niet eigenlijk en solaredge zal het weer terugzetten.

De inverteren hebben een eigen id ja, ik heb daar eind vorig jaar met Pieter contact over gehad, omdat het eerst niet werkte met 2 inverter’s.
Hij heeft toen e.e.a. aangepast en toen werkte het, met elke hun eigen ID. Sinds een update van jou werkte inverter 2 ineens niet meer, toen wat aan het uitproberen geweest en nadat ik 2 op 1 zetten werkten ze beide. Prima natuurlijk maar t is wat vreemd.

Met vriendelijke groet,

Willem Jan Bakker
Boerestreek 1
8757JR Gaast
Mob. +31653217707

biemond, 2022-03-28:
bedankt , Ik ga even kijken.

kun je trouwens op beide  invertor met batterij als device gebruiken. dan zou je flow action op beide kunnen gebruiken. misschien helpt dan de 2de invertor mee met opladen.

biemond, 2022-03-28:
kan je diagnostic report maken met verschillende unit ids , ben benieuwd wat ik dan zie.  Andere hebben het werkend met verschillende ids

Bakkerwj, 2022-10-11:
’t is hier donker….

Met vriendelijke groet,

Willem Jan Bakker

Bakkerwj, 2022-10-11:
Hierbij. 

1492b82d-664a-4bae-bb7f-2688c4a9e822

Met vriendelijke groet,

Willem Jan Bakker
Boerestreek 1
8757JR Gaast
Mob. +31653217707