biemond/solaredge.modbus #3: Missing variables

jowamedia, 2022-04-04:
Hi, 
thanks a lot for your app.
It use to work perfectly, but since several days, it miss some variables, like : 
- Power Export
- Power Import
- Consumption

I use these last variables to manage the car charger.
I had to reinstall the device, has it lost the connexion. The port on my Solaredge is 502 (not 1502 as default in the app).
Would it be possible to have theses variables again ?

Thanks a lot in advance.
Best,
Joel

biemond, 2022-04-04:
Hey, 

let me check this consumption.  I think I need  to calculate this.
I think latetly make sure you don't use wifi, they disabled modbus tcp over wifi , only over lan.

[image]

[image]

[image]

jowamedia, 2022-04-05:
Thanks looks great.
(I use LAN, not wifi).

For consumption, if you look on the app, it shows the power consumption. You may have an address for that in the modbus !?

biemond, 2022-04-05:
It is calculated. but I will add it, I got all the information.

biemond, 2022-04-10:
Hey , can you check this release. see if it works for you

https://homey.app/a/solaredge.modbus/test/

jowamedia, 2022-04-12:
Hey, I do have the v.2.0.13. When I click on the link ending with test, the
URL is redirecting to the normal one.
How to install this test version ?

biemond, 2022-04-12:
yes , good. homey already approved it,  does it work for you?

jowamedia, 2022-04-26:
Ok, good.
But I don't have additional values :
[image]

nor in flow.

Shell I reinstall the converter to get them ?

biemond, 2022-04-26:
can you try the battery invertor device instead.  I am checking it all, no worries when you don't have a battery. 
so you have meters installed without a battery?

jowamedia, 2022-04-26:
yes exactly, I don't have battery. I do have the Solaredge SE9k.

biemond, 2022-04-26:
ok I am checking it and everything is optional.  so batt inverter device should work for you then
can you try that 1

jowamedia, 2022-04-26:
now I am getting the values and they are updated every few seconds. I will look at it a few days and check if everything works fine. 
Thanks for the dev ,-)
I appreciate.

closed 2022-04-26 by jowamedia

jowamedia, 2022-10-11:
Ok, good.
But I don't have additional values :
[image: IMG_ECD60003DABC-1.jpeg]
nor in flow.

Shell I reinstall the converter to get them ?