biemond/solaredge.modbus #6: Since 2.0.13 update, last value unknown

Picsou621, 2022-05-11:
Hello, I am contacting you because I use your excellent SolarEdge application, which is very important to me and the operation of my home. However, since the 2.0.13 update on my Homey Pro (unless I'm mistaken which was yesterday), my SolarEdge inverter was no longer recognized. I reinstalled the app, added the inverter's IP address, but I don't know what to put under "Port" and under "Inverter ID", I never changed anything here. The device was added in Homey, but all values now say "Last value unknown". Can you help me? Thanks a lot in advance!

Picsou621, 2022-05-11:
Hello again, I found the beginning of a solution by indicating port 502 instead of port 1502. The value "Solar Power" is now indicated. However, my SolarEdge has a Smart Meter, which was also showing me the power consumption of my house, which is a value I absolutely need. However, this value is no longer displayed by Homey. Why is this, and is there a solution? Thank you very much!

biemond, 2022-05-11:
oh nice. you can use battery invertor device then .  this comes with meters plus I still check for batt when it is not there.  Code is flexible.

Picsou621, 2022-05-11:
Thank you very much for this quick answer, indeed it works with the battery version, great :-) Then, I think that many people have the Smart Meter but no battery (because it is not profitable at all, at least in Switzerland at the moment). So I wonder if it makes sense to have removed the Smart Meter data from the version without battery, but of course you know what is better ;-) I have been fully charging two electric vehicles for months with the excess power produced by my solar panels, thanks to your application and the Go-E Charger app. It would have been a disaster for me if the SolarEdge app had not fed back the data from the Smart Meter ;-)

biemond, 2022-05-11:
Cool. , I think for the batt , meters are a requirement.  We in NL , have mostly a P1 device on our smart meter.

Maybe we do both because it is totally flex , if it is there I will use it else not.