biemond/solaredge.modbus #7: Omvormer niet meer te benaderen

Geos3, 2022-05-12:
Goedemorgen,

Sinds de update kan ik mijn omvormer niet meer benaderen.

Dmv de solaredge app doet hij het gewoon.

Ik hoor het graag hoe te handelen.

Groetjes Floris

biemond, 2022-05-12:
Hey,

kun je de device eens verwijderen en weer toevoegen?   Ook gebruik je wifi of een lan kabel?  Solaredge support geen  modbus over wifi meer. 

Groet Edwin

Geos3, 2022-05-12:
Hoi Edwin,

Net het apparaat verwijderd.

Juiste waardes ingevuld.

Hij word toegevoegd maar waardes blijven op nul staan zegmaar.

Ook zie ik in de Homey app dat hij 6 omvormers toevoegd??

Ik hoor graag je reactie.

Groet Floris

biemond, 2022-05-12:
6 stuks ,  je hebt er toch 1 of niet. kan je alles verwijderen.  
1 device kan al de poort vasthouden, zodat de andere deze niet kunnen benaderen.

biemond, 2022-05-12:
ok , kun je op zend diagnostic report drukken.

Geos3, 2022-05-12:
4cdd1522-e286-4594-a221-ed4320fcb8c9

biemond, 2022-05-12:
```
[Device:b654bc74-4817-46b8-a2d9-a2ddceab9732] pollInvertor
[log] 2022-05-12 17:39:41 [ManagerDrivers] [Driver:invertor] [Device:b654bc74-4817-46b8-a2d9-a2ddceab9732] 192.168.1.153
Error: connect EHOSTUNREACH [192.168.1.153:10001](http://192.168.1.153:10001/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1157:16) {
  errno: -113,
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.1.153',
  port: 10001
}
Client closed
[log] 2022-05-12 17:39:59 [ManagerDrivers] [Driver:invertor] [Device:b654bc74-4817-46b8-a2d9-a2ddceab9732] pollInvertor
[log] 2022-05-12 17:39:59 [ManagerDrivers] [Driver:invertor] [Device:b654bc74-4817-46b8-a2d9-a2ddceab9732] 192.168.1.153
Error: connect EHOSTUNREACH [192.168.1.153:10001](http://192.168.1.153:10001/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1157:16) {
  errno: -113,
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.1.153',
  port: 10001
```

biemond, 2022-05-12:
cannot reach it , is it on LAN plus 10001 is not in use by some other program.

biemond, 2022-05-12:
wat is je ip?

```
Error: connect EHOSTUNREACH [192.168.1.153:10001](http://192.168.1.153:10001/)

    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1157:16) {
  errno: -110,
  code: 'ETIMEDOUT',
  syscall: 'connect',
  address: '192.168.3.130',
  port: 10001
}
Client closed
[log] 2022-05-12 17:57:08 [ManagerDrivers] [Driver:invertor] [Device:b654bc74-4817-46b8-a2d9-a2ddceab9732] pollInvertor
[log] 2022-05-12 17:57:08 [ManagerDrivers] [Driver:invertor] [Device:b654bc74-4817-46b8-a2d9-a2ddceab9732] 192.168.3.130
Error: connect ETIMEDOUT [192.168.3.130:10001](http://192.168.3.130:10001/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1157:16) {
  errno: -110,
  code: 'ETIMEDOUT',
  syscall: 'connect',
  address: '192.168.3.130',
  port: 10001
}
Client closed
```

biemond, 2022-05-12:
ok, timeout op poort 10001, is that correct ? plus je gebruikt een kabel of niet.
het kan ook zijn dat een ander progje deze poort in gebruik heeft.

biemond, 2022-05-12:
of een vlan en een firewall issue

Geos3, 2022-05-12:
Ik heb de poort 10001 idd ingesteld en ik gebruik een vaste internetkabel in de omvormer.
Het probleem is begonnen na de nieuwe update..

Ik heb er niet mega veel verstand van sorrie😪

Geos3, 2022-05-12:
En de polling op 20

biemond, 2022-05-12:
oh Ik support geen polling , heb je de juiste versie?
[image]

biemond, 2022-05-12:
id en poort kun je in solaredge menu zien.

Geos3, 2022-05-12:
Waar kan ik het id vinden ik zit nu in de omvormer te zoeken...

biemond, 2022-05-12:
unitId of device id . maar waarschijnlijk is het 1 of 2 .

biemond, 2022-05-12:
zal wel ergens bij modbus settings staan

Geos3, 2022-05-12:
Hoi Edwin,

Het is gelukt!!!!

Dank je wel voor je goede hulp en tijd.

Groet Floris

biemond, 2022-05-12:
super.  cool

Geos3, 2022-10-11:
Je bedoeld ip adres van omvormer?

Geos3, 2022-10-11:
Hoi Edwin,

Ik gebruik Lan

Als ik hem verwijder onthoudt hij de instellingen?

Groet Floris

Geos3, 2022-10-11:
De app van solaredge werkt wel gewoon prima.

biemond, 2022-10-11:
ja invertor pushed naar solaredge and app haalt het daar op.  homey moet de data ophalen.