biemond/solaredge.modbus #8: Solar Edge EV charger

nossors, 2022-05-18:
Hi Edwin
thank you for your very good app which is working perfekt. I habe bought the new solar Edge EV charger.
It would be fantastic if you could also integrate this new item into the app in order to generate for example a flow like.. If battery full and export of power is more than.. Switch on the EV charger.
Best
Remo

nossors, 2022-05-18:
> oh nice,
>
> I also want that.
> So I already allow the control flow and this can export back to the grid.
> Do you know if EV charger counts as internal consumption, plus you have more info on this. Like what can you do from the menu or solaredge app.
>
>  <https://user-images.githubusercontent.com/1085004/168989163-1486994e-1b55-4a19-9b8f-6d7f5c3077a3.png>
>
> thanks

He Edwin

In the solar edge app is only the following possibility.

After integration of the EV charger beside my existing battery, can you at the moment switch on or off
the ev charger by remote.

See attachment

 

Best regards

Remo

biemond, 2022-05-18:
you have some model number and is it connected to the invertor?

nossors, 2022-05-18:
Well it is connected to the network, where the invertor is also connected.
See the label attached