biemond/solaredge.modbus #9: SolarEdge SE5000-ER-01 Settings

MarcoVos5437, 2022-05-20:
Hi, I just started with e new Homey Pro. now i try to install my SolarEdge. I could install the related app but after entering the settings there are now readings. (i used the correct IP-address and default Port (1501) and Inverter ID (1).)
The converter is connected via a cable (RJ45) to my network (no Wif). what could be wrong?

MarcoVos5437, 2022-05-20:
in the past i used a other system (OpenHab) where i used a API Key with an ID-nr from the invertor itself ......

biemond, 2022-05-20:
Hey,

can you make sure the other is not connected to this and can you send me a diagnostic report from the application side. 

thanks

MarcoVos5437, 2022-05-21:
Where can i find the diagnostic report?

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden

MarcoVos5437, 2022-10-11:
Gevonden:
cd1dabb6-2f8f-41c7-ba5d-f563924adcc6

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden