biemond/solaredge.modbus #11: Connection Homey Pro-Solaredge

PotHomey, 2022-06-10:
Hi, I have a connection problem between my Homey Pro box and my Solaredge SE10K inverter. I have found the IP address, but I don’t know where to find the Port #. Regarding the inverter ID, the Solaredge app asks me a 2 digits number, and the ID I have found has 7 digits. Can you please help me? Regards, Christophe

biemond, 2022-06-10:
Hey,

Can you go the menu of the invertor , it should be there.

but I think for old ones port is 502 or new ones it is 1502.
plus invertor id  is always 1 when you only have one device.

Also use lan and not wifi.

biemond, 2022-06-10:
and this https://community.homey.app/t/app-pro-solaredge/45487

PotHomey, 2022-06-12:
Thank you very much for your reply and for the information. I will try again to connect my Inverter in a couple of days and I will come back to you to give you a feedback. Thank you !