biemond/solaredge.modbus #12: Zero export limit

djfanatix, 2022-06-14:
Hi,

Could you add a option to read and write the export limit?
It is address F001 

[[](url)](https://tweakers.net/i/_Vbw6h07uP3EtmVy-RP_ZXr0jnM=/full-fit-in/4000x4000/filters:no_upscale():fill(white):strip_exif()/f/image/IHKtvheNjManYE9TaabWoDh8.png?f=user_large)

Thanks

biemond, 2022-06-14:
yeah, will do, should be easy.

export limit is what we control or not?

By the way for your solcast,  is it possible to add a flow condition so I can charge the batt with dal tarief when needed (like when expected kwh for today is less ) , I run this after midnight so prediction should be in for that day..

biemond, 2022-06-14:
can you test this and see if this is what you want.

https://homey.app/nl-nl/app/solaredge.modbus/SolarEdge/test/

and https://github.com/biemond/solaredge.modbus/commit/80bd9ae7780b22c7404ca49af1f91bd3ff6b8dae

[image]

djfanatix, 2022-06-15:
Yes I see the same.

I found some more documentation:
[](https://user-images.githubusercontent.com/2674434/134251823-c404b894-037e-469b-bc9b-8787ebc1770a.png)

Can you make a flow card or a setting to change the power limit?

I think people with dynamic energy contracts can then enable zero export in case of negative prices

biemond, 2022-06-15:
really that is a thing in europe.  you have to pay when you do export to the grid. So you can only do this when you have batt or when it is stimulated and get paid for it.

biemond, 2022-06-15:
so an enable/disable flow with.   advanced power on 1 and reactive power config to 4 or for normal/enabled both back to 0 
is this also a menu setting/action? which one?

if you do this on the invertor = active power limit 0 % then?

djfanatix, 2022-06-15:
> really that is a thing in europe. you have to pay when you do export to the grid. So you can only do this when you have batt or when it is stimulated and get paid for it.

No most of the time you get paid to export, but on sunny and windy days, the rates are negative, so you have to pay.
(or you get paid to import)

djfanatix, 2022-06-15:
I found this in another repo:
I think I was wrong, the active power limit, limits the inverter, that's not what we need, we need to limit export (with a meter of course)

I don't think (for my needs) that E001 has to be included.

#0xE000 - 1 - Export control mode: 16bit_uint
     0: "Disabled",
     1: "Direct Export Limitation",
     2: "Indirect Export Limitation",
     4: "Production Limitation"
#0xE001 - 1 - Export control limit mode:16bit_uint
     0: "Total",
     1: "Per phase"
#0xE002 - 2 - Export control site limit: 32bit_float
["Export control site limit", "export_control_site_limit", 0xE002, "f", {"min": 0, "max": 10000, "unit": "W"}],

biemond, 2022-06-15:
yes that looks better , I know sometimes on US or AUS you cannot deliver back to the grid.

djfanatix, 2022-06-15:
> 
> By the way for your solcast, is it possible to add a flow condition so I can charge the batt with dal tarief when needed (like when expected kwh for today is less ) , I run this after midnight so prediction should be in for that day..

I checked this, but I am not planning to rewrite the app for sdk 3, as it too much work with the whole UTC and crontasks thing.
You can just use logic and the solcast tag and use it as a condition. (greater or smaller then..)

biemond, 2022-06-15:
I need this from oct/nov

biemond, 2022-06-15:
[image]

[image]

biemond, 2022-06-15:
so only when you have a meter.

biemond, 2022-06-15:
```
        "advancedpwrcontrolen": [0xf142, 2, 'UINT32', "Advanced Power Control En"],
        "reactivepwrconfig": [0xf102, 2, 'UINT32', "Reactive Power Config"],        
        "export_control_mode": [0xe000, 1, 'UINT16', "Export control Mode"],
        "export_control_limit_mode": [0xe001, 1, 'UINT16', "Export control limit Mode"],
        "export_control_site": [0xe002, 2, 'FLOAT32', "Export control site limit"],
        "powerreduce": [0xf140, 2, 'FLOAT32', "Power Reduce"],
        "maxcurrent": [0xf18e, 2, 'FLOAT32', "Max Current"]

rrcr_state 0 xxx RRCR State
active_power_limit 100 xxx Active Power Limit
cosphi 1 xxx CosPhi
advancedpwrcontrolen 1 xxx Advanced Power Control En
reactivepwrconfig 4 xxx Reactive Power Config
export_control_mode 0 xxx Export control Mode
export_control_limit_mode 0 xxx Export control limit Mode
export_control_site -3.4028234663852886e+38 xxx Export control site limit
powerreduce 100 xxx Power Reduce
maxcurrent 7071.06787109375 xxx Max Current
```

biemond, 2022-06-20:
tested it on 5000
```
rrcr_state 0 xxx RRCR State
active_power_limit 100 xxx Active Power Limit
cosphi 1 xxx CosPhi
advancedpwrcontrolen 1 xxx Advanced Power Control En
reactivepwrconfig 4 xxx Reactive Power Config
export_control_mode 1 xxx Export control Mode
export_control_limit_mode 0 xxx Export control limit Mode
export_control_site 5000 xxx Export control site limit
powerreduce 100 xxx Power Reduce
maxcurrent 7071.06787109375 xxx Max Current
```
[image]
[image]

[image]
[image]
[image]

biemond, 2022-06-20:
I think I only need to set these 3
```
export_control_mode 1 xxx Export control Mode
export_control_limit_mode 0 xxx Export control limit Mode  , already on site / not phase.
export_control_site 5000 xxx Export control site limit
```

biemond, 2022-06-20:
can you try this https://homey.app/a/solaredge.modbus/test/
for now only a menu item ( no flow ) and export limit fixed on 5000W

[image]
[image]

djfanatix, 2022-06-21:
Looks good, my site limit is 4500w, so I can’t test it at the moment.

biemond, 2022-06-21:
ok , 4500 is your energy provider restriction or do you get some error on solaredge. 

shall I change it to 4500 to find out if it works or give me some value.

djfanatix, 2022-06-21:
no, I only have a small inverter, 4kw.
Can you try 3000 or so?

biemond, 2022-06-21:
yes , let me do this now.

biemond, 2022-06-21:
done https://homey.app/a/solaredge.modbus/test/

djfanatix, 2022-06-21:
[image]

Works! Thank you.

I double checked in HA and I see the same data

biemond, 2022-06-21:
beter, mooi.  I will work on the flows and this export limit setting

biemond, 2022-06-21:
and the flows https://homey.app/a/solaredge.modbus/test/ plus you can change export limit watts

MTRKhn, 2022-12-03:
Hi,

could you please consider adding a 0W export limit setting ? I can’t find a way to enable zero export :(

thanks !

biemond, 2022-12-04:
oh , let me check

biemond, 2022-12-04:
@MTRKhn can you try this one https://homey.app/a/solaredge.modbus/test/

MTRKhn, 2022-12-04:
UI seems ok. Now the challenge will be to generate enough to check if it’s exporting or not 😅

[image]

biemond, 2022-12-04:
yes winter time.  it is hard to get something and go over your own consumption