biemond/solaredge.modbus #14: Peak Shaving

WillemsF, 2022-09-05:
Hallo,

Zou er een mogelijkheid bestaan om "Min Import Control" toe te voegen?
Via deze weg is het mogelijk om aan Peak Shaving te doen (België).
Ik stel deze nu manueel in op de omvormer (zie resultaat afbeelding).

[image]

Een Homey Pro heb ik nog niet in bezit, op vandaag kan ik dit nog niet testen.
alvast bedankt.

biemond, 2022-10-15:
Hey, just saw your message. 

you import or export control. Already added export https://github.com/biemond/solaredge.modbus/issues/12#issuecomment-1161409610

or can you show me what you did in the solaredge menu

thanks

WillemsF, 2022-10-15:
Hey Edwin,

is het oké als ik in het Nederlands verderga?
ik ben even door het menu gegaan van mijn SE5000H (met display)

POWER CONTROL -> ENERGY MANAGER -> LIMIT CONTROL -> CONTROL MODE -> MIN IMPORT CTRL -> MIN IMPRT LIM -> waarde in [KW] of DISABLE**

Onder POWER CONTROL -> ENERGY MANAGER -> ENERGY CTRL (deze blijft "steeds" op MSC)
 **dus enkel de waarde in kW aanpassen of op DISABLE plaatsen (zie hierboven).

Als dit reeds geïmplementeerd zit onder #12 zal ik zeker een Homey pro aanschaffen.

alvast bedankt voor het vele en mooie werk tot dusver.
Francis.

biemond, 2022-10-15:
hey, Ik ga even kijken in mijn menu.

import / export is lastig met wat je bedoelt.  Ik weet issue #12 was bedoelt om niet over je piek te gaan en dat je dan extra moet betalen aan je netbeheerder.

maar je bedoelt de export van je zonnepanelen beperken naar 1000, 2000 etc watt .  of praat je over import om je battery of telsa load te beperken. 

[image]

biemond, 2022-10-15:
[image]

biemond, 2022-10-15:
[image]

biemond, 2022-10-16:
step 1

[image]

WillemsF, 2022-10-16:
Hey,

By adjusting only this parameter "min import control" you can easily control the "charging" and "discharging" of the battery.
I have a LG Resu 9.8kWh.

example 1:
a winter's day with a solar yield of 5 kWh.
if you set this parameter (MIC) higher or equal to the import, all generated solar energy will flow to the battery. (backup for peak shaving)

example 2:
at 6 pm only a battery charge of 50% no more solar yield, status was on MSC, so switch to peak shaving -> MIC (min import control) at 0.2 kW, at 08:00 the sun rises and a sunny day is expected , so status from MIC to MSC, unfortunately at 13h we only have a battery charge of 60% so now back a MIC of 0.2 kW for example to pump more solar energy into the battery.

would like to find a screeshot of the SE monitoring of the day before yesterday with the "exact" values ​​read from the grid operator's meter on top.
I have made the manual adjustments (MIC and MSC) clear in colors. you can also see the battery charge.

[image]

WillemsF, 2022-10-16:
I suspect it is this parameter,
Unfortunately I can't test this yet.
[image]

biemond, 2022-10-16:
Thanks , I have the same batt , let me check if I can do this on my batt

https://github.com/biemond/solaredge.modbus/blob/main/drivers/invertorwithbatt/device.ts#L256

for now I only support this , let check and try option 11 and  also try to allow a minimal import  and use batt for the peaks.

```
 // 0 – Disabled
        // 1 – Export Control
        // 2 – Production Control
        // 11 – Minimum Import Control
```

biemond, 2022-10-16:
also I do this https://github.com/biemond/solaredge.modbus/blob/main/drivers/invertorwithbatt/device.ts#L319
I guess we do it by hand on the inverter and try to do the same from homey.

biemond, 2022-10-16:
[image]

biemond, 2022-10-16:
reading all this , it should be easy , only probably will do fixed limits like 500 , 1000 etc maybe will add 100,200,300, 400 also
https://github.com/biemond/solaredge.modbus/blob/main/drivers/invertorwithbatt/device.ts#L276

very hard to calculate those exact values.

WillemsF, 2022-10-17:
That's exactly what I had in mind.
fixed value, according to my consumption profile I would opt for the following limits:
DISABLE - 50 - 100 - 150 - 200 - 300 - 400 - 500 - 750 - 1000 - 1500 - 2000 - 2500.
depending on the charge of the battery and other logic (weather forecast, ...) the battery can be used dynamically.

this will be a big improvement compared to the profiles of SE

I suspect the "PV only" option was already there.
You need this if you want to hold a certain charge in the battery for a longer period of time.

[image]
[image]

biemond, 2022-10-17:
cool, I will start playing with this and reverse engineer the modbus settings.

biemond, 2022-10-18:
now checking the values
[image]
[image]

biemond, 2022-10-18:
[image]

biemond, 2022-10-18:
[image]
[image]
got the result, now I need to do some more action and flow test

biemond, 2022-10-18:
also works from flows 
[image]

biemond, 2022-10-18:
https://github.com/biemond/solaredge.modbus/commit/c41ff603b5eb5c0ccb7217241b89c2a01507596b

I would say with the new homey coming , get a cheap 2nd hand homey pro or try it from someone else

biemond, 2022-10-18:
to do, are in between values like 100 ,250 ,750 etc

WillemsF, 2022-10-18:
Hey Edwin,
I just became the owner of a 2nd hand Homey Pro :-)
The Solaredge App is already installed and I am now making the connection.

Amazing what you have done in such a short time. really great. :1st_place_medal: 

a small note that I tested this afternoon.

when I set the MIC to 0.00kW there will be "no" more export to the grid when the battery is (almost) 100%= total export limit.
to activate the export when the battery is almost full, you have to set the MIC to disable.

biemond, 2022-10-18:
oh never was thinking about stopping export (in NL we are a bit behind on you guys)  also 0 is not an flow option (just yet) , please advise what you do for this so I can do the same and reverse engineer it.
[image]
[image]

WillemsF, 2022-10-18:
zonet de flow gemaakt wanneer de batterij onder de 53% komt. dus even wat energie verbruiken :)

WillemsF, 2022-10-18:
just made the flow when the battery drops below 53%. so use some energy :)

biemond, 2022-10-18:
mooi. alles is waarschijnlijk nieuw voor je. insights is ook leuk voor alle statistieken. 
energy tab en ook deze tuya zigbee sockets met metingen is erg leuk om te weten welk device energie gebruikt plus erg goedkoop.  https://www.amazon.de/gp/product/B0054PSH6K/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

WillemsF, 2022-10-18:
modbus data did not arrive. got everything online now and everything works as it should.
I was just shocked by the SOH of my batt :( , what you use sometimes wears out faster than expected.
would it still be possible to add the lower values ​​to the flow? 50 - 100 - 150 - 200 - 300 - 400.
I will test everything thoroughly this weekend and delve deeper into the matter.
a wonderful appliance Homey.
big thanks in advance

biemond, 2022-10-19:
new version to test https://homey.app/a/solaredge.modbus/test/ 

with new import rates https://github.com/biemond/solaredge.modbus/commit/b3208cdc4464639ca95c4d802a22c93f25c0114c

WillemsF, 2022-10-19:
I am testing the flows and at the moment everything is working properly. 
I can't test the values ​​<1000W now because battery is almost empty. 
Is currently at Peak Shaving 1500W. 
tomorrow there will not be much solar power, ideal time to leave the MIC at 1500 and so that all solar power => battery.

biemond, 2022-10-19:
yep , winter time. in feb.march we are ok again.  I can do "salderen" over the whole year so I only have to pay for 50kwh .
I have "daltarief" but for this I won't sacrifice the batt health, went to 99% after 1 year.

maybe I will go for hour prices and charge batt once a day when it has the  lowest prices.

WillemsF, 2022-10-19:
Here we also had a principle of *salderen*, from next year we will be charged extra on the highest peak measured over a quarter of an hour that exceeds 2.5 kWh. We have a high consumption here at home (8MWh) of which I have 1MWh from the grid, so 7MWh directly from the Solaredge and LG Chem... the battery is in service from last year May

biemond, 2022-10-21:
Hey, 

new version with a performance boost by disabling not used registers, meters and battery. 
now I can read it all in 5 seconds and can bring back polling to every 30 seconds for meters and battery .

please try this.
https://homey.app/a/solaredge.modbus/test/

WillemsF, 2022-10-21:
Hey Edwin,

I will test this version tonight.
I did have something strange this night.
The battery has gone from 40% battery level to 1500W MIC.
This logic was in a flow I deleted on the homey 4 hours before.
Just to clarify, if I upload a new version do I have to delete the old one first or can I install this over it?
[image]

biemond, 2022-10-21:
you can do an update else you will loose the device and its data.

it can homey was hanging or could not connect but you check solaredge website. when it happens again send me a diagnostic report so I can see the settings.

also it can be if you do storage charge/discharge mode , it has timeout of 6 hours then it goes back to the default as protection;

```
      if (type == 'storagecontrolmode') {
        // 0 – Disabled
        // 1 – Maximize Self Consumption – requires a SolarEdge Electricity meter on the grid or load connection point
        // 2 – Time of Use (Profile programming) – requires a SolarEdge Electricity meter on the grid or load connection point 3 – Backup Only (applicable only for systems support backup functionality)
        // 4 – Remote Control – the battery charge/discharge state is controlled by an external controller
        const storagecontrolmodeRes = await client.writeSingleRegister(0xe004, Number(value));
        console.log('controlmodewrite', storagecontrolmodeRes)
      }

      if (type == 'storagedefaultmode') {
        const storagedefaultmodeRes = await client.writeSingleRegister(0xe004, Number(4));
        console.log('controlmodewrite', storagedefaultmodeRes)

        // set timeout to 6 hours, done in seconds
        const remoteTimeout = await client.writeMultipleRegisters(0xe00b, [ Number(21600), 0 ]);
        console.log('remote_control_command_timeout', remoteTimeout);
        // 0 – Off
        // 1 – Charge excess PV power only.
        // Only PV excess power not going to AC is used for charging the battery. Inverter NominalActivePowerLimit (or the inverter rated power whichever is lower) sets how much power the inverter is producing to the AC. In this mode, the battery cannot be discharged. If the PV power is lower than NominalActivePowerLimit the AC production will be equal to the PV power.
        // 2 – Charge from PV first, before producing power to the AC.
        // The Battery charge has higher priority than AC production. First charge the battery then produce AC.
        // If StorageRemoteCtrl_ChargeLimit is lower than PV excess power goes to AC according to NominalActivePowerLimit. If NominalActivePowerLimit is reached and battery StorageRemoteCtrl_ChargeLimit is reached, PV power is curtailed.
        // 3 – Charge from PV+AC according to the max battery power.
        // Charge from both PV and AC with priority on PV power.
        // If PV production is lower than StorageRemoteCtrl_ChargeLimit, the battery will be charged from AC up to NominalActivePow-erLimit. In this case AC power = StorageRemoteCtrl_ChargeLimit- PVpower.
        // If PV power is larger than StorageRemoteCtrl_ChargeLimit the excess PV power will be directed to the AC up to the Nominal-ActivePowerLimit beyond which the PV is curtailed.
        // 4 – Maximize export – discharge battery to meet max inverter AC limit.
        // AC power is maintained to NominalActivePowerLimit, using PV power and/or battery power. If the PV power is not sufficient, battery power is used to complement AC power up to StorageRemoteCtrl_DishargeLimit. In this mode, charging excess power will occur if there is more PV than the AC limit.
        // 5 – Discharge to meet loads consumption. Discharging to the grid is not allowed. 
        // 7 – Maximize self-consumption
        const remotecontrolwrite = await client.writeSingleRegister(0xe00d, Number(value));
        console.log('remotecontrolwrite', remotecontrolwrite);
      }
```

WillemsF, 2022-10-21:
When using the homey, should the storage mode always be set to "Remote Control" ? Correct?
I also deleted the storageprofile on the SE site (I have admin rights on my installation) this was still enabled
[image]

biemond, 2022-10-21:
remote control has this timer and moves to the default 
you need to use this 1 , which has no timer but I think export or import limit has no timer

[image]
[image]

biemond, 2022-10-21:
remote control timer is max 24h
[image]

biemond, 2022-10-24:
@WillemsF how is it going?  is it working for you

WillemsF, 2022-10-24:
Hey,

Everything works perfectly, it was a bit of puzzling with the homey but I have already been able to achieve a lot. I have tested everything and encountered no problem so far. Only the order of going to a state is very important.
So far a wonderful achievement from you!
[image]

biemond, 2022-10-24:
nice advanced flow, becoming a pro.

WillemsF, 2022-10-24:
if I may ask one more thing, would it be possible to move the control screen to the right? I've unintentionally changed the status a few times.
[image]

biemond, 2022-10-24:
sorry I know. I think this is done for other most common devices , like socket or lamps, on off etc 1st , then data and batt as last.

biemond, 2022-10-24:
but I see some devices start with data. let me check this

WillemsF, 2022-11-20:
Hey,
It works like a charm. 
I'm testing this every day now and so far everything works perfectly. 
There is one small error in the value of 1500W, it writes 1508 as a value.

[image]

biemond, 2022-11-21:
thanks , also remember this and let me fix this.

closed 2022-11-21 by WillemsF