biemond/solaredge.modbus #15: 3 phase energymeter and not blow the fuses

intetiffanypersson, 2022-10-15:
I have a Solaredge 10k with the modbus energy meter. It's 3-phase, since I'm in Sweden (not sure if all of Europe standardised on 3-phase). The Smartmeter/EnergyMeter is also 3-phase.

From this I can via Prometheus and scripts I found online, get a lot of values. But not into Homey, yet. I'm thinking/hoping it should be "easy" to add a few cards with these values.
Since I also have a "dumb" 3x16A car charger (Keba, another plugin) and my house has 3x20A import, I'd  like to measure each phase once a minute, from the Solaredge modbus data, and shut down or reduce my charging if any of the phases are exceeding 20Amp import from the grid. Otherwise, if someone starts cooking while the car is charging, the main fuse will blow. Or if someone showers, or starts a washing machine etc, etc. So many opportinitues for my family to yell at me...

So basically what I need is for the Solaredge plugin to present more cards with values from the modbus registers. Specifically for me, from https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf 
I THINK I want from Meter1:
40191 40192 1 M_AC_Current_A int16 Amps Phase A AC Current
40192 40193 1 M_AC_Current_B int16 Amps Phase B AC Current
40193 40194 1 M_AC_Current_C int16 Amps Phase C AC Current
Not a 100% sure how that works, but the Smartmeter is connected to the incoming Grid, not really the Solaredge, so I'm thinking this should be it...
With that I could make a flow that is like "If Phase A or B or C is greater that 20Amp, then set carcharger to 10A".

Thanks for a wonderful plugin, the modbus connection possibility is is just awesome!

KR
Magnus

biemond, 2022-10-15:
nice,

so I am looping over the 3 meters https://github.com/biemond/solaredge.modbus/blob/main/drivers/solaredge.ts#L93 but indeed only do meter 1 , extra meter is no issue 
https://github.com/biemond/solaredge.modbus/blob/main/drivers/solaredge.ts#L316

everything is flexible for me and easy to extend. currently meters are under battery , even if you have no batt you can still use it , I detect what is there.

but for the extra 2 phases can send me a diagnostic report from my homey app and I will get all possible values , so we can look together at the values register and see what you need.

thanks Edwin,

intetiffanypersson, 2022-10-15:
Hi Edwin,
Thanks for looking into this.
Never Done a diagnostic before. Do you need something more than the code?

54317F

/Magnus

Ev. stavfel beskylls min iPhone

biemond, 2022-10-15:
hey,

when you on hemoy , add device with battery on my app . if everything is working you can go to more, apps , select  my app , settings and do create diagnostics report.

I will get that and see your data.

intetiffanypersson, 2022-10-16:
Excellent,
I added it as with battery and created the report.

c62ba761-3d10-4222-94dc-7a7ca81e7d79

Thanks,
Magnus

Ev. stavfel beskylls min iPhone

biemond, 2022-10-16:
so comparing yours with mine ( got 1 phase meter)

mine
```
c_manufacturer SolarEdge  xxx Manufacturer
c_model SE4000H-RWS00BNO4 xxx Model
c_version 0004.0014.0228 xxx Version
c_serialnumber 74000868 xxx Serial
c_deviceaddress 1 xxx Modbus ID
c_sunspec_did 101 xxx SunSpec DID
current 420 -2 Current
l1_current 420 xxx L1 Current
l2_current -1 xxx L2 Current
l3_current -1 xxx L3 Current
current_scale -2 xxx Current Scale Factor
l1_voltage 2350 -1 L1 Voltage
l1_voltage_scale -1 xxx Voltage Scale Factor
l2_voltage -1 -1 L2 Voltage
l2_voltage_scale -1 xxx Voltage Scale Factor
l3_voltage -1 -1 L3 Voltage
l3_voltage_scale -1 xxx Voltage Scale Factor
l1n_voltage -1 -1 L1-N Voltage
l1n_voltage_scale -1 xxx Voltage Scale Factor
l2n_voltage -1 -1 L2-N Voltage
l2n_voltage_scale -1 xxx Voltage Scale Factor
l3n_voltage -1 -1 L3-N Voltage
l3n_voltage_scale -1 xxx Voltage Scale Factor
```

yours
```
current 242 -2 Current
l1_current 75 xxx L1 Current
l2_current 90 xxx L2 Current
l3_current 76 xxx L3 Current
current_scale -2 xxx Current Scale Factor
l1_voltage 4026 -1 L1 Voltage
l1_voltage_scale -1 xxx Voltage Scale Factor
l2_voltage 4050 -1 L2 Voltage
l2_voltage_scale -1 xxx Voltage Scale Factor
l3_voltage 4031 -1 L3 Voltage
l3_voltage_scale -1 xxx Voltage Scale Factor
l1n_voltage 2308 -1 L1-N Voltage
l1n_voltage_scale -1 xxx Voltage Scale Factor
l2n_voltage 2338 -1 L2-N Voltage
l2n_voltage_scale -1 xxx Voltage Scale Factor
l3n_voltage 2340 -1 L3-N Voltage
l3n_voltage_scale -1 xxx Voltage Scale Factor
power_ac 3200 -1 Power
power_ac_scale -1 xxx Power Scale Factor
```

biemond, 2022-10-16:
both have meter 1
mine
```
meter1-c_deviceaddress 2 xxx Modbus ID
meter1-c_sunspec_did 203 xxx SunSpec DID
meter1-current 27 -1 Current
meter1-l1_current 0 xxx L1 Current
meter1-l2_current 3 xxx L2 Current
meter1-l3_current 24 xxx L3 Current
meter1-current_scale -1 xxx Current Scale Factor
meter1-voltage_ln 23087 -2 L-N Voltage
meter1-l1n_voltage 23087 xxx L1-N Voltage
meter1-l2n_voltage 23326 xxx L2-N Voltage
meter1-l3n_voltage 23495 xxx L3-N Voltage
meter1-voltage_ll 0 xxx L-L Voltage
meter1-l12_voltage 0 xxx L1-l2 Voltage
meter1-l23_voltage 0 xxx L2-l3 Voltage
meter1-l31_voltage 0 xxx L3-l1 Voltage
meter1-voltage_ln_scale -2 xxx Voltage Scale Factor
meter1-frequency 5000 -2 Frequency
meter1-frequency_scale -2 xxx Frequency Scale Factor
meter1-power 369 0 Power
meter1-l1_power 0 xxx L1 Power
meter1-l2_power -40 xxx L2 Power
meter1-l3_power 410 xxx L3 Power
meter1-power_scale 0 xxx Power Scale Factor
meter1-power_apparent 552 0 Power (Apparent)
meter1-l1_power_apparent 5 xxx L1 Power (Apparent)
meter1-l2_power_apparent 71 xxx L2 Power (Apparent)
meter1-l3_power_apparent 537 xxx L3 Power (Apparent)
meter1-power_apparent_scale 0 xxx Power (Apparent) Scale Factor
meter1-power_reactive -408 0 Power (Reactive)
meter1-l1_power_reactive -5 xxx L1 Power (Reactive)
meter1-l2_power_reactive -59 xxx L2 Power (Reactive)
meter1-l3_power_reactive -343 xxx L3 Power (Reactive)
meter1-power_reactive_scale 0 xxx Power (Reactive) Scale Factor
meter1-power_factor -60 -2 Power Factor
meter1-l1_power_factor 1640 xxx L1 Power Factor
meter1-l2_power_factor 5419 xxx L2 Power Factor
meter1-l3_power_factor -7239 xxx L3 Power Factor
meter1-power_factor_scale -2 xxx Power Factor Scale Factor
meter1-export_energy_active 1246680 0 Total Exported Energy (Active)
meter1-l1_export_energy_active 0 xxx L1 Exported Energy (Active)
meter1-l2_export_energy_active 0 xxx L2 Exported Energy (Active)
meter1-l3_export_energy_active 1918603 xxx L3 Exported Energy (Active)
meter1-import_energy_active 1215552 xxx Total Imported Energy (Active)
meter1-l1_import_energy_active 6714 xxx L1 Imported Energy (Active)
meter1-l2_import_energy_active 1181079 xxx L2 Imported Energy (Active)
meter1-l3_import_energy_active 699681 xxx L3 Imported Energy (Active)
meter1-export_energy_active_scale 0 xxx Energy (Active) Scale Factor
meter1-export_energy_apparent 0 -32768 Total Exported Energy (Apparent)
meter1-l1_export_energy_apparent 0 xxx L1 Exported Energy (Apparent)
meter1-l2_export_energy_apparent 0 xxx L2 Exported Energy (Apparent)
meter1-l3_export_energy_apparent 0 xxx L3 Exported Energy (Apparent)
meter1-import_energy_apparent 0 xxx Total Imported Energy (Apparent)
meter1-l1_import_energy_apparent 0 xxx L1 Imported Energy (Apparent)
meter1-l2_import_energy_apparent 0 xxx L2 Imported Energy (Apparent)
meter1-l3_import_energy_apparent 0 xxx L3 Imported Energy (Apparent)
meter1-export_energy_apparent_scale -32768 xxx Energy (Apparent) Scale Factor
meter1-import_energy_reactive_q1 0 -32768 Total Imported Energy (Reactive) Quadrant 1
meter1-l1_import_energy_reactive_q1 0 xxx L1 Imported Energy (Reactive) Quadrant 1
meter1-l2_import_energy_reactive_q1 0 xxx L2 Imported Energy (Reactive) Quadrant 1
meter1-l3_import_energy_reactive_q1 0 xxx L3 Imported Energy (Reactive) Quadrant 1
meter1-import_energy_reactive_q2 0 xxx Total Imported Energy (Reactive) Quadrant 2
meter1-l1_import_energy_reactive_q2 0 xxx L1 Imported Energy (Reactive) Quadrant 2
meter1-l2_import_energy_reactive_q2 0 xxx L2 Imported Energy (Reactive) Quadrant 2
meter1-l3_import_energy_reactive_q2 0 xxx L3 Imported Energy (Reactive) Quadrant 2
meter1-export_energy_reactive_q3 0 xxx Total Exported Energy (Reactive) Quadrant 3
meter1-l1_export_energy_reactive_q3 0 xxx L1 Exported Energy (Reactive) Quadrant 3
meter1-l2_export_energy_reactive_q3 0 xxx L2 Exported Energy (Reactive) Quadrant 3
meter1-l3_export_energy_reactive_q3 0 xxx L3 Exported Energy (Reactive) Quadrant 3
meter1-export_energy_reactive_q4 0 xxx Total Exported Energy (Reactive) Quadrant 4
meter1-l1_export_energy_reactive_q4 0 xxx L1 Exported Energy (Reactive) Quadrant 4
meter1-l2_export_energy_reactive_q4 0 xxx L2 Exported Energy (Reactive) Quadrant 4
meter1-l3_export_energy_reactive_q4 0 xxx L3 Exported Energy (Reactive) Quadrant 4
meter1-import_energy_reactive_q1_scale -32768 xxx Energy (Reactive) Scale Factor
```
yours
```
meter1-c_deviceaddress 1 xxx Modbus ID
meter1-c_sunspec_did 203 xxx SunSpec DID
meter1-current 137 -1 Current
meter1-l1_current 115 xxx L1 Current
meter1-l2_current -8 xxx L2 Current
meter1-l3_current 30 xxx L3 Current
meter1-current_scale -1 xxx Current Scale Factor
meter1-voltage_ln 23089 -2 L-N Voltage
meter1-l1n_voltage 23089 xxx L1-N Voltage
meter1-l2n_voltage 23419 xxx L2-N Voltage
meter1-l3n_voltage 23439 xxx L3-N Voltage
meter1-voltage_ll 0 xxx L-L Voltage
meter1-l12_voltage 0 xxx L1-l2 Voltage
meter1-l23_voltage 0 xxx L2-l3 Voltage
meter1-l31_voltage 0 xxx L3-l1 Voltage
meter1-voltage_ln_scale -2 xxx Voltage Scale Factor
meter1-frequency 5002 -2 Frequency
meter1-frequency_scale -2 xxx Frequency Scale Factor
meter1-power -3230 0 Power
meter1-l1_power -2653 xxx L1 Power
meter1-l2_power 77 xxx L2 Power
meter1-l3_power -654 xxx L3 Power
meter1-power_scale 0 xxx Power Scale Factor
meter1-power_apparent 3272 0 Power (Apparent)
meter1-l1_power_apparent 2654 xxx L1 Power (Apparent)
meter1-l2_power_apparent 197 xxx L2 Power (Apparent)
meter1-l3_power_apparent 701 xxx L3 Power (Apparent)
meter1-power_apparent_scale 0 xxx Power (Apparent) Scale Factor
meter1-power_reactive -528 0 Power (Reactive)
meter1-l1_power_reactive -91 xxx L1 Power (Reactive)
meter1-l2_power_reactive -182 xxx L2 Power (Reactive)
meter1-l3_power_reactive -255 xxx L3 Power (Reactive)
meter1-power_reactive_scale 0 xxx Power (Reactive) Scale Factor
meter1-power_factor 7733 -2 Power Factor
meter1-l1_power_factor 10000 xxx L1 Power Factor
```

biemond, 2022-10-16:
besides the totals which you are already can see .

so you like to see these current ones or not. Also something from meter1 ?
```
current 242 -2 Current
l1_current 75 xxx L1 Current
l2_current 90 xxx L2 Current
l3_current 76 xxx L3 Current
current_scale -2 xxx Current Scale Factor
l1_voltage 4026 -1 L1 Voltage
l1_voltage_scale -1 xxx Voltage Scale Factor
l2_voltage 4050 -1 L2 Voltage
l2_voltage_scale -1 xxx Voltage Scale Factor
l3_voltage 4031 -1 L3 Voltage
l3_voltage_scale -1 xxx Voltage Scale Factor
l1n_voltage 2308 -1 L1-N Voltage
l1n_voltage_scale -1 xxx Voltage Scale Factor
l2n_voltage 2338 -1 L2-N Voltage
l2n_voltage_scale -1 xxx Voltage Scale Factor
l3n_voltage 2340 -1 L3-N Voltage
l3n_voltage_scale -1 xxx Voltage Scale Factor
```

biemond, 2022-10-16:
for If Phase A or B or C is greater that 20Amp
I see I don't do current, I do power and voltage but no current.

```
meter1-current 137 -1 Current
meter1-l1_current 115 xxx L1 Current
meter1-l2_current -8 xxx L2 Current
meter1-l3_current 30 xxx L3 Current
meter1-current_scale -1 xxx Current Scale Factor
meter1-voltage_ln 23089 -2 L-N Voltage
```

intetiffanypersson, 2022-10-16:
Hi,
I'm not 100% sure and it's hard to tell while looking at numbers that aren't realtime, but I think the ones in the last port are what I want.
meter1-l1_current 115 xxx L1 Current
meter1-l2_current -8 xxx L2 Current
meter1-l3_current 30 xxx L3 Current
I'm guessing that is my grid import/export, positive is importing, negative is overflow from the panels as export. So 1,15A, -0,08A, 3.0A.
Certainly, while you are in there, seeing the each pjhase solar production would be nice too, it would allow me to automate other things based on the panels.

Let's start with that and I will confirm once I can see them in Homey, in realtime, and compare with mySolarEdge app.
l1_current 75 xxx L1 Current
l2_current 90 xxx L2 Current
l3_current 76 xxx L3 Current

biemond, 2022-10-16:
cool, so tonight you will have a version to test this out.

biemond, 2022-10-16:
[image]

biemond, 2022-10-16:
here we go https://homey.app/nl-nl/app/solaredge.modbus/SolarEdge/test/

let me know if it works.

intetiffanypersson, 2022-10-16:
Yeeeeeeey! It works!!

Thanks so much! Awesome!!