biemond/solaredge.modbus #17: Reading data is not getting updated

SdeKoning, 2022-10-28:
Hi,

Since two days I am not getting updates on my generated energy any more.

It is dat and night now reading 539 w.

Please advice.

biemond, 2022-10-28:
Hey,  can you restart homey else remove and add the device again.  , or try to change some setting and put it back. 

sometimes with a homey or app upgrade , the timer is gone.   Or your ip has changed.  

thanks

you can also send me a diagnostic report.

SdeKoning, 2022-10-30:
Hi,

I have checked my solaredge but the parameters are looking ok.

I have made a diagnostic report
57b06ea2-9558-4052-9e55-eac260fb8adb

Thanks for helping

mvg
Steef de Koning

Peno11, 2022-11-01:
Hi, I also dont receive any data since the last update. Restarted, changed stuff. Nothing. Any advise?
br
Pär

SdeKoning, 2022-11-01:
No, i gave up.

mvg
Steef de Koning