biemond/solaredge.modbus #18: No data from inverters

Gottenbossie, 2022-10-29:
Hello, I am an new use whith an homey Pro early 2019
i have instaled the Solaredge app V2.1.9 on my homey pro and conected my two inverter to it.
at first sight looks like te conection is ok ( the solaredge inverters are conected to a wired network)
but de is no data in the solaredge app
do you have an idea what the problem coukld be?
[image]
[image]
[image]
[image]
[image]

biemond, 2022-10-29:
Hey , can you send me a diagnostic report so I can see if there is a connection.

plus only use 1 reader at a time , because modbus cannot allow multiple client .  Also use a LAN connection , no wifi.

Gottenbossie, 2022-10-29:
7d037b66-57e7-450e-aa04-0ffb8cf2789eDo you mean this?

biemond, 2022-10-29:
```
[Device:00b9962d-6bf7-4827-abe7-f18db40edfba] 192.168.2.4
Error: connect ECONNREFUSED [192.168.2.4:1502](http://192.168.2.4:1502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1157:16) {
  errno: -111,
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.2.4',
  port: 1502
}
Client closed
Error: connect ECONNREFUSED [192.168.2.3:1502](http://192.168.2.3:1502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1157:16) {
  errno: -111,
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.2.3',
  port: 1502
}
Client closed
```

moet iets zijn , ip of modbus staat niet aan of ander port of modbus id.  of een 2de programma leest dit ook uit en blokkeert homey