biemond/solaredge.modbus #19: no "energy" measurement?

Zakraket, 2022-11-02:
I just switched from Diederik's Bakker "Zonnepanelen" app to your SolarEdge app for my SE1500-M 4 panel installation. Works fine, however I'm not seeing a "daily generated energy" anywhere.
It's available in the other app so I'm assuming the convertor has it available?

[image]
[image]

I see it mentioned on the apppage in the reviews

biemond, 2022-11-02:
awesome timing,  check this one https://homey.app/nl-nl/app/solaredge.modbus/SolarEdge/test/

Zakraket, 2022-11-02:
Great! Works like a charm

One comment: it reports 0kWh now (just updated), while my app reports 0,9kWh for today. I don't know if the invertor "holds" the total day value by itself (so maybe it's not possible to get this from the invertor). This will of course be fixed tomorrow when it starts again at 0

biemond, 2022-11-02:
indeed, I have to store now when it is empty and on 0:01 it will use the new value and caculate from that moment

biemond, 2022-11-02:
also this calculated based on the export value. there is some diff with the app.  maybe solaredge has a better info then me

closed 2022-11-02 by Zakraket