biemond/solaredge.modbus #21: Power consumption of the house

kodak2022, 2022-11-08:
Hello,

I was using the homey app for solarpanels, with the webservice of SolarEdge. 

Your new app is much better and acurate, thank you, but the power consumption of the property is missing.

It's very usefull because in case you produce more than what you consume, you can charge a battery or a car :-)

Could you add this information ?

Here it's an idea of the other app :

[image]

Thank you :)

Stephen

biemond, 2022-11-08:
Hey, let me check this.  

Do you have smart meters installed?  if so use device with battery instead. even when you don't have a battery.  plus you will get way more meters and phase data.

This is what I current have on a default inverter.

[image]

example of inverter with meters and battery (optional)

[image]
[image]
[image]

kodak2022, 2022-11-08:
Ok, understood, thank you :-)

I don't have batteries, but when I select battery on install I have all informations.

Very cool app - thank you :-)

biemond, 2022-11-08:
cool