biemond/solaredge.modbus #23: Dynamic Commands

WillemsF, 2022-11-20:
Hey,
I am testing the possibilities to switch to a dynamic energy contract.
During testing it appears that all modes work except mode 3 => charge from PV+AC
I can't charge the battery from the grid...
In the manual I read to set the solaredge ac_charge policy to "1".
Could that be the reason why it doesn't work?

[image]

biemond, 2022-11-21:
Hey,

I think I had the same in the beginning , I remember that I did this from the menu and my battery was tested. After that I was allowed to do this. 

but let me check again.

biemond, 2022-11-21:
at least it still works.

[image]

WillemsF, 2022-11-21:
solved, activating a profile charge from AC for a while did indeed do the trick.
Is this a permanent condition?
[image]
NO LIMIT
[image]
1000W CHARGE LIMIT.

Perfect!!

biemond, 2022-11-21:
nice, in my case it was testing batt 1st. 

Also when batt is low , what also will happen it will slow charge from 8 / 9 until 10% before going full charge.

biemond, 2022-11-21:
yeah, if you switch to hour prices , sometimes it is free .  hour by the power app. you maybe can automate thus

WillemsF, 2022-11-21:
that's the idea... getting a contract now will be a bigger challenge.
to automate this day ahead prices I have already found the app on Homey
[image]

closed 2022-11-21 by WillemsF

biemond, 2022-11-21:
yep, 1 day before you can already  know the hours.