biemond/solaredge.modbus #25: Data incorrect

djvanas, 2023-01-10:
Hi,

I have an issue whereby the data retrived from the inverter is incorrect. The data in the Homey app shows a daily production of 205kWh (for today for example), the SE app shows a daily production of only 1.91kWh.

[image]
[image]

Tnx for your support

biemond, 2023-01-10:
Hey, 

yeah I try to reset it at 24:00 so I guess I missed your time / timezone.  see [
](https://github.com/biemond/solaredge.modbus/blob/main/drivers/solaredge.ts#L284)

is your homey up to date? 

and at 12:01 midnight can you send me a diagnostic report

thanks

biemond, 2023-01-10:
maybe you got 00:01 instead of 24:01 , I am also in NL.

djvanas, 2023-01-10:
Hi, 

Tnx. Homey s/w version is 8.1.1. I’m not sure what you mean with the timezone comment. I’m in NL, with correct timezone and time;

[image]

biemond, 2023-01-10:
I have it on english , maybe it is different 

[image]

biemond, 2023-01-10:
can you send me  at 0:01 or 0:02 a diagnostic report. very curious.  else also will change to dutch.

biemond, 2023-01-11:
I see it , 1 hour before compared to internal time.  let me check mine and I will also change the language to NL.

05:44
daily: 205.09800000000178
Client closed
[log] 2023-01-11 04:44:32 [ManagerDrivers] [Driver:invertor]

biemond, 2023-01-11:
and I will today or tomorrow make some test version which you can try

biemond, 2023-01-11:
I change to dutch. lets see what happens on midnight.

biemond, 2023-01-11:
fixed it https://homey.app/a/solaredge.modbus/test/ can you try this. on friday you will see it

djvanas, 2023-01-14:
Tnx, this has now been resolved!