biemond/solaredge.modbus #26: Hi, after upgrading to version 2.1.13 i got broken flows caused be missing action card "Solar Power is changed" and other cards with data changed.

FSmu, 2023-01-11:


FSmu, 2023-01-11:
After upgrade to version 2.1.13 I got broken flows caused be missing action card "Solar Power is changed" and some other cards that work with "data changed".

FSmu, 2023-01-11:
I got broken flows caused be missing action card "Solar Power is changed" and some other cards with data changed.

FSmu, 2023-01-11:
Sorry pressed the wrong button.

biemond, 2023-01-11:
oh let me check .  sorry for that.

FSmu, 2023-01-12:
> Van "Edwin Biemond" ***@***.***>
> Aan "biemond/solaredge.modbus" ***@***.***>
> CC "FSmu" ***@***.***>; "State change"
> Datum 11-1-2023 15:58:06
> Onderwerp Re: [biemond/solaredge.modbus] Missing cards after upgrade to
> version 2.1.13 (Issue #26)

Hi Card is back so it looks like everything is working again. Great App 
by the way and thank you for the fast fix.

Regards,

Frans