biemond/solaredge.modbus #28: No more decimals in Total values since last update

MTRKhn, 2023-01-12:
Total yield, Total export and total import have no decimal values anymore so calculating a daily consumption is not precise anymore... Is it possible to revert that change please ?

biemond, 2023-01-12:
Hey,

will do. 
by the way is the daily total yield not ok.   if not can you show me your calculation for this. Maybe I can add this.

thanks

MTRKhn, 2023-01-12:
Nope, Total Day Yield is not correct (or i generated 152kWh yesterday lol).

I'm just collecting the "meter-power" information in InfluxDb and then representing using Grafana for the desired period. Well, i'm collecting and representing a lot of SolarEdge data :D 

[image]

biemond, 2023-01-12:
that part is fixed , in NL they use 00:00 as midnight , others 24:00. 
can you try this https://homey.app/a/solaredge.modbus/test/ and on friday it should work.

biemond, 2023-01-12:
I will add the decimals back.

MTRKhn, 2023-01-12:
> that part is fixed , in NL they use 00:00 as midnight , others 24:00. can you try this https://homey.app/a/solaredge.modbus/test/ and on friday it should work.

I already installed .15 to check in the decimal issue was fixed. I'll update this thread tomorrow :)

MTRKhn, 2023-01-13:
> that part is fixed , in NL they use 00:00 as midnight , others 24:00. can you try this https://homey.app/a/solaredge.modbus/test/ and on friday it should work.

I checked this morning. "Total Day Yield" is 1.48kWh but SolarEdge app reports 2.06kWh :(

biemond, 2023-01-13:
yep I need to calculate this [ ](https://github.com/biemond/solaredge.modbus/blob/main/drivers/solaredge.ts#L277)
maybe solaredge has more info then me.

biemond, 2023-01-13:
https://github.com/biemond/solaredge.modbus/blob/main/drivers/solaredge.ts#L277

biemond, 2023-01-13:
published a new test version with 2 decimals

MTRKhn, 2023-01-14:
All good now. Thanks !