biemond/solaredge.modbus #29: Omvormer status word niet meer weergegeven, krijg nu een numer?

Bakkerwj, 2023-02-10:
Omvormer status werkt niet meer zoals voorheen, geeft nu een nummer ipv de echte status.

biemond, 2023-02-11:
Hey , dat is raar, de code is voor maanden niet aangepast en voor mij gaat het nog steeds goed.

zie je een bepaalde code of altijd alleen maar nummers

```
            if (result['status'] && result['status'].value != 'xxx') {
                if (parseInt(result['status'].value) < 9) {
                    this.addCapability('invertorstatus');
                    if (this.getCapabilityValue('invertorstatus') != result['status'].value) {
                        let status_str: { [key: string]: string } = {
                            "0": 'Undefined',
                            '1': 'Off',
                            '2': 'Sleeping',
                            '3': 'Grid Monitoring',
                            '4': 'Producing',
                            '5': 'Producing (Throttled)',
                            '6': 'Shutting Down',
                            '7': 'Fault',
                            '8': 'Maintenance'
                        }
                        // console.log(this.driver.id);
                        // console.log(status_str[result['status'].value]);
                        this.homey.flow.getDeviceTriggerCard('changedStatus').trigger(this, { status: status_str[result['status'].value] }, {});
                    }
                    this.setCapabilityValue('invertorstatus', result['status'].value);
                }
            }
```
https://github.com/biemond/solaredge.modbus/blob/main/.homeycompose/capabilities/invertorstatus.json

[image]

biemond, 2023-02-11:
anders kun je me ook een diagnostic report sturen.

Bakkerwj, 2023-02-11:
Hey Edwin,

Ik ben nog even aan het testen geweest:

Ik zie alleen maar nummers als ik via een flow een "inverteren status" opvraag, maar als ik een “status" opvraag geeft ie wel een omschrijving maar FAULT, terwijl ze aan het produceren zijn op dit moment.

￼￼

Ik heb beide omvormer’s vd week een firmware update gegeven, en e.e.a. in de instellingen vd omvormer gewijzigd omdat de SolarEdge omvormer’s steeds hun IP kwijt zijn en naar een “vreemd” ip gaan, (schijn meer voor te komen volgens het Unifi forum) dus heb ze op vast ip (192.168.1.10 en 192.168.1.12) gezet zowel in de omvormer zelf als in mijn Unifi systeem maar het probleem blijft zich voordoen.

Je kunt (sinds update, kon voorheen volgens mij niet) via de setapp de monitoring verbinding kiezen: Automatisch, TCP, WiFi, 485 Leider, en 485 volger.
Dat heb ik nu voor de “hoofd” omvormer SE10K-RWS (192.168.1.10) op TCP gezet en de “volger” via 485 volger, die zat voorheen ook rechtstreeks op homey via eigen IP  (192.168.1.12) maar nu via het zelfde ip nr als de hoofd omvormer met een ander poortnummer en Inverter ID 2.

￼

￼￼

Heb jij meer mensen gehoord die problemen hebben met “wandelende” Solaredge omvormer’s, alhoewel ze een vast ip hebben gaan ze vanzelf naar een heel andere reeks, ze zitten op een switch waar meer apparatuur op zit wat ’t wel goed doet, dus aan de kabel kan t eigenlijk ook niet liggen..

Met vriendelijke groet,

Willem Jan Bakker

biemond, 2023-02-11:
Ik heb ook alles van unify en op de switch maar geen problemen voor mij. Heb hem een vast ip gegeven in unify. Maar ben benieuwd naar je diagnostic report. Eens kijken wat de inverter terug geeft

Bakkerwj, 2023-02-11:
465889bf-31cc-44ea-8103-b52a6a30e462 Met vriendelijke groet,Willem Jan Bakker

biemond, 2023-02-11:
```
batt1-status 10 xxx Status
batt1-status_internal 1 xxx Internal Status
```

en 

```
status 4 xxx Status
vendor_status 0 xxx Vendor Status
```

biemond, 2023-02-11:
inverter is dus   '4': 'Producing',
batterij        '10': 'Unknown'  <-- ik heb geen idee what dat is , maar ik zie het soms

biemond, 2023-02-11:
zo  je inverter en inverter met batt vechten voor de connectie.
de ene keer krijgt de ene het , de volgende keer de ander

Bakkerwj, 2023-02-11:
Nr 2abde59c5-9c5e-4881-8f11-198cbc322804Batterij nu aan het laden gezet, (backup) is destatus hiervan nu anders?Met vriendelijke groet,Willem Jan BakkerBoerestreek 18757JR GaastMob. +31653217707

Bakkerwj, 2023-02-11:
h’m

Misschien dan toch inverteren ID 2 weer via z’n eigen ip doen?

Met vriendelijke groet,

Willem Jan Bakker
Boerestreek 1
8757JR Gaast
Mob. +31653217707

biemond, 2023-02-11:
ja denk het . anders moet ik kijken of ik iets met een off set kan doen. of 1 poller voor meerdere devices

biemond, 2023-02-11:
zeg je dat nr 10  voor batt status  = backup

Bakkerwj, 2023-02-11:
ja, dat is de batterij omvormer, SE10K-RWS

Met vriendelijke groet,

Willem Jan Bakker
Boerestreek 1
8757JR Gaast
Mob. +31653217707

biemond, 2023-02-11:
```
batt1-status 3 xxx Status
batt1-status_internal 3 xxx Internal Status
status 1 xxx Status
vendor_status 0 xxx Vendor Status
```

```
status 4 xxx Status
vendor_status 0 xxx Vendor Status
```

biemond, 2023-02-11:
is je status probleem voorbij?

Bakkerwj, 2023-02-11:
Jaaaaa.. werkt weer !

Thanx!

Met vriendelijke groet,

Willem Jan Bakker

Bakkerwj, 2023-06-26:
Hoi Edwin,

Het probleem dat ik als status weer een nummer krijg ipv, een “omschrijving” bijv. “Producing” is plotseling weer opgedoken, ik had een flow wat aangepast en ineens weer een nummer ipv omschrijving.

￼

Met vriendelijke groet,

Willem Jan Bakker

Bakkerwj, 2023-06-26:
9ac1d4d7-6ef5-42cc-a7a6-47b202fca039Met vriendelijke groet,Willem Jan Bakker

biemond, 2023-06-26:
hey, welke status en met welke waarde ? https://github.com/biemond/solaredge.modbus/blob/main/.homeycompose/capabilities/invertorstatus.json

misschien is het een status code die ik niet ken?

biemond, 2023-06-26:
```
status 4 xxx Status
vendor_status 0 xxx Vendor Status
batt1-status 6 xxx Status
batt1-status_internal 3 xxx Internal Status
```

Bakkerwj, 2023-06-26:
Zie afbeelding, onderaan gaf ie nog de omschrijving, Producing, en na wijziging flow ineens een nummer (4, 5 etc)

Ik heb nu een paar weken de nieuwe homey 2023, maar t werkte altijd prima totdat ik de flow zostraks aanpaste. sindsdien geeft ie een nummer ipv omschrijving dus bijv.  4 ipv producing, eigenlijk hetzelfde als in februari toen ik contact met je had.

￼

Met vriendelijke groet,

Willem Jan Bakker
Boerestreek 1
8757JR Gaast
Mob. +31653217707

biemond, 2023-06-26:
[image]

Bakkerwj, 2023-06-26:
> heb ook een homey 2023. wat is je flow startpunt

https://homey.app/f/xTP3Uz￼
Gedeelde Flow
homey.app

Met vriendelijke groet,

Willem Jan Bakker

Bakkerwj, 2023-06-26:
Heb het al gevonden, als je “inverteren status” kiest dan geeft ie een nummer, en als je “status” kiest de omschrijving.

￼

￼

Met vriendelijke groet,

Willem Jan Bakker

biemond, 2023-06-26:
interessant ,  ik ook als je inverter  status
ja gebruik de flag op status