biemond/solaredge.modbus #33: How to interface an old SolarEdge inverter with new Homey Pro

alechenne, 2023-05-07:
Dear Support,

I had VZ the past installed the app Solar Panels on my Homey 2019 to access my SokarEdge SE15k inverter which is unfortunately not equipped with TCP Modbus.

How should I proceed to have it back to Homey Pro 2023? Per my understanding, only SolarEdge with TCP Modbus are supported.

Thanks

biemond, 2023-05-07:
Hey, so you don't see or can enable modbus?  did you get any updates on your inverter?

Also are you saying solar panel app does not support homey 2023?

alechenne, 2023-05-07:
Thanks for your quick reply. My inverter hasn’t the Modbus TCP option (SolarEdge SE15k of 2017).

With Homey 2019, I used the Solar Panels App and give the API key of the my SolarEdge. Unfortunately Solar Panels App isn’t working on Homey 2023 Pro (still with SDK2 and not SDK3 ). I posted an issue on GitHub for Solar Panels.

I don’t have or receive any update from SokarEdge. My inverter cannot be setup
by the IOS App. I only see the information (Solar production …). 

Kind regards

biemond, 2023-05-07:
really , looks like the test version is https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/
converting to sdk v3 at the max takes a few hours, maybe it requires a lot of testing.

biemond, 2023-05-07:
I think solaredge or your installation guy can push the updates, we as end-user not . sadly

biemond, 2023-05-07:
if your installation guy cannot help , just create a support ticket on solaredge as a home owner . 2017 is not so old.

alechenne, 2023-05-08:
> really , looks like the test version is https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/ converting to sdk v3 at the max takes a few hours, maybe it requires a lot of testing.

THanks for your advise. I installed it on Homey Pro 2023. It works 50%. I could retreive some data but suddenly it stopped (last update 4 hours). It seems that there is still some bugs.

biemond, 2023-05-09:
ah, ok.  yep he need to fix that.