biemond/solaredge.modbus #34: Geen waarden van mijn SolarEdge omvormer

JDKnuivers, 2023-05-09:
Ik ben beginnend gebruiker van een Homey Pro 2023. Ik heb middels deze app mijn SolarEdge (3fase) omvormer toegevoegd met het correcte IP-adres. Toch ontvang ik geen waarden van mijn omvormer. 

Waar zou dit aan kunnen liggen? 

diagnostic report ID: D57323FF

biemond, 2023-05-09:
ja , hij kan niet connecten.  klopt de ip en de inverter gegevens. heb je de aanpassing gedaan in het menu?

plus draai geen andere app , want dan blokt hij.

JDKnuivers, 2023-05-09:
IP adres klopt. De andere gegevens weet ik zo niet, deze heb ik niet aangepast. Waar zou ik dat kunnen zien?

biemond, 2023-05-09:
heb je in de menu van de omvormer gekeken?  je kan ook mdns discovery bekijken , volgens kun je solaredge zien met de ip , port en modus id.

biemond, 2023-05-09:
[image]

JDKnuivers, 2023-05-12:
Hoi Biemond, 

Zojuist heb ik Discovery gedownload en ik krijg geen resultaten voor de modbus. Dien ik hiervoor nog instellingen te wijzigen op de omvormer toevallig? Ik heb een omvormer waar geen schermpje op zit (SE5K), dus ik weet niet of ik dan op afstand iets kan uitlezen of instellen?

[image]

biemond, 2023-05-12:
ik denk dat je het niet geactiveerd hebt.  kun je de 1st post volgen op de homey app forum. je moet iets doen in het menu van je omvormer

biemond, 2023-05-12:
te gek. mooi dat het is opgelost.

biemond, 2023-05-12:
zie je nu trouwens de discovery?

JDKnuivers, 2023-05-12:
Ja, hij staat er nu bij!

[image]

biemond, 2023-05-12:
mooi dat is perfecte check voor mij om te weten of alles ok is.