biemond/solaredge.modbus #36: Is this app working for homey 2023?

snuf72, 2023-06-14:
Is this app working for homey 2023?
doesn’t seem to connect

biemond, 2023-06-14:
Hey,

can you check your ip? if it is changed and you can check with a mdns discovery tool if it shows up 
also you can remove and re-add the device.

plus send me a diagnostic report.

thanks

snuf72, 2023-06-14:
Manual ip is set to 192.168.2.105 but still this is not working with the app I’m using. What do you meen with removing device? From homey-app I removed it several times

Best Regards,

Sylvain van Uffelen

biemond, 2023-06-14:
ok cool. can you send me a diagnostic report.

snuf72, 2023-06-14:
233f130e-65a7-4012-87f3-bd0d9b04ecf8

Best Regards,

Sylvain van Uffelen

biemond, 2023-06-14:
so is it up and do you see your inverter with discovery?  (did you enable it all like said on the 1st page of the solaredge forum)  see this 1st post https://community.homey.app/t/app-pro-solaredge-growatt-tcp-modbus/45487
[image]

```
[Device:269f2e19-2b59-43bd-9d4b-f79c0ed80e93] 192.168.2.105
Error: connect EHOSTUNREACH [192.168.2.105:1502](http://192.168.2.105:1502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1494:16) {
  errno: -113,
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.2.105',
  port: 1502
}
Client closed
```