biemond/solaredge.modbus #37: Solar edge SE3000H Photovoltaic

goofy1982, 2023-06-15:
Beste, 
Ik heb geprobeerd mijn Solar edge SE4000H te koppelen maar zonder succes. Ik heb vorige reviews gelezen maar ik weet niet waar ik mijn IP adres moet zoeken.  Is het mogelijk stap per stap mij hierin te begeleiden aub.

biemond, 2023-06-16:
Hey,

geen probleem , had je alles al aangezet, gedaan in the solaredge inverter menu?  that is stap 1

biemond, 2023-06-16:
als je dat hebt gedaan, dan heb je een app on je mobiel mDNS discovery.  dan zoek je op solaredge-modbus
dat heeft eigenlijk al jouw gegevens. zo niet check stap 1 of zorg ervoor dat je op hetzelfde netwerk zit.

[image]

OscarVesterink, 2025-07-10:
> Hoi,
> 
> Als ik hier op mag inhaken: Ik heb ook een SE4000H (oud beestje met nog een CPU in 3.19.*** met LCD). Ik heb de instructies in de README gevolgd en als ik naar het LAN-scherm ga krijg ik netjes de hele reeks aan '1' te zien plus een 'S_OK'. Als ik vervolgens door klik naar het MODBUS TCP overzicht zie ik dat de omvormer continu probeert contact te maken met de gateway welke faalt met de simpele melding 'Gateway ping failed'. Dit probeert de omvormer elke 5 seconden. Er is hier jammer genoeg bijzonder weinig over te vinden vanuit Solar Edge.
> 
> Dit verrast mij daar de LAN-check ook in ieder geval iets van een gateway ping bevat en succesvol is. Ik heb tevens nu drie LAN-kabels geprobeerd, elk gevalideerd op werking via PC's en een kabeltester. Bij mijn weten heb ik de MODBUS aangezet door specifiek de poort te definiëren (met een succesbericht), maar vanaf dat moment faalt de gateway ping dus in het MODBUS overzicht.
> 
> Heeft iemand enig idee wat hier aan de hand kan zijn?
> 
> Bij voorbaat dank!

Op de een of andere manier heeft een harde reset geholpen. Sterker nog, pas na het opnieuw koppelen van alle optimizers werd er een succesvolle verbinding opgezet. Ik kan dit zelf nog niet verklaren en ga de boel uitzoeken, maar het probleem is er in ieder geval niet meer. Toch bedankt.