biemond/solaredge.modbus #38: No more data?

Xybertecnic, 2023-06-19:
Till this morning communication with my SE4000H HD Wave worked fine, but since a couple of hours I can't get any data anymore. I have deleted and re-added the inverter without issues, but still no data, any idea what might be the problem?

[image]
[image]

biemond, 2023-06-19:
plus send me a diagnostic report. so I can see what is happening.

biemond, 2023-06-19:
```
[Device:03623de3-d317-46dd-bc50-c0a4ea8f7957] 192.185.1.14
Error: connect ECONNREFUSED [192.185.1.14:1502](http://192.185.1.14:1502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1494:16) {
  errno: -111,
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.185.1.14',
  port: 1502
}
```

biemond, 2023-06-19:
check mdns entry on your phone

Xybertecnic, 2023-06-19:
> check mdns entry on your phone

die vind hem niet

Xybertecnic, 2023-06-19:
I can see it in the solaredge app just fine

biemond, 2023-06-19:
yes there was an update but it cannot connect to your inverter. so when ip is ok , it can be solaredge menu setting like port or modbus id. else you would see a stacktrace.

Or network issue.  also there should be no other app who connects to it and blocks homey.

Xybertecnic, 2023-06-19:
well I don't know what happens with my Asus Router but I have problems with everything the last few days, before my P1 meter did not connect anymore, now the solaredge, but in the end even though I had set the IP of the SE fixed in my router, it still did not get it assigned, it was set to a completely different IP address. I had to go look into the inverter itself to find out. Everything works now again :-) Maybe next time buy another brand router...

Xybertecnic, 2023-06-19:
Would you mind removing the message with all the mac details etc? Don't know if it is harmfull but better safe than sorry...

biemond, 2023-06-19:
also had  this once on some p1 meter , I assign a fixed on mac address with dhcp but totally ignores it and does not connect.

probably some devices does not follow dhcp  or ehternet strictly or router is less strict.

Xybertecnic, 2023-06-19:
in any case, thanks for your help

biemond, 2023-06-19:
by the way do you see mdns entry now on your phome?

Xybertecnic, 2023-06-20:
> by the way do you see mdns entry now on your phome?

no, not found, everything else is, but not the inverter

closed 2023-10-24 by Xybertecnic