biemond/solaredge.modbus #39: How to connect SolarEdge inverter to Homey Pro 2023?

HansVermeerbachen, 2023-06-20:
Dear Biemond,

I've received my Homey Pro 2023 and I'm trying to get insights for/off/about my solar panels. I'm trying to add my SolarEdge Inverter (type S3680H).

[image]

Your Homey app (for which many thanks of course) asks me to enter the ip address, port and device ID of my inverter. I have been able to find the IP address by logging in at my router's dashboard and examining the connected devices. But I have no idea where I could find a port number and / or a device ID. 

Can you help me on my way? Where should I look for those? (With the default settings 1502/1 as suggested by your app, I get no data.)

Thanks a bundle.

biemond, 2023-06-20:
Hey, 

can you check this on your phone https://github.com/biemond/solaredge.modbus/issues/37#issuecomment-1594463919

and did you do the steps of the 1st post on the app forum. like enable it on the menu of solaredge.

thanks

HansVermeerbachen, 2023-06-20:
Thanks for your quick reply.

I needed to enable Modbus TCP in the setting of the inverter, which is indeed what you already suspected. It works now.

It was fairly easy once you know it, but it took me quite a while to get to this. I can via my Homey Pro app and the Homey Pro website and I landed here (https://homey.app/nl-nl/app/solaredge.modbus/SolarEdge-+-Growatt-TCP-modbus/), + browsed around a lot on the community and later on GitHub, but I did not easily land here (https://community.homey.app/t/app-pro-solaredge-growatt-tcp-modbus/45487). Perhaps I'm bad a browsing, but perhaps it can be made easier to find.

KR

biemond, 2023-06-20:
yes I know , homey don't allow  big app description.  it is on github repo https://github.com/biemond/solaredge.modbus/ and on the app forum.