biemond/solaredge.modbus #41: Omvormer geeft niks meer door

Geos3, 2023-07-17:
Goedenavond,

Ik heb helaas weer een probleempje met de app.

Hij geeft geen waardes meer door.

Heb de app verwijderd en opnieuw geïnstalleerd.

Maar hielp helaas niet.

Ik hoop op een oplossing.

Alvast top bedankt!

Groet Floris

biemond, 2023-07-17:
Hey,

heb. je al gekeken of je ip adres is veranderd?  en kijk of je inverter een mdns/bonjour entry laat zien met ip, port en modbus id. je hebt hiervoor een android of iphone app.

anders zend even een diagnostic report . maar het is meestal altijd het ip adres of een soort firewall rule als het ooit heeft gewerkt,

groet Edwin

Geos3, 2023-07-17:
9462757c-c59b-4e7c-b28a-0ced861944fd

biemond, 2023-07-17:
yep, kan hem niet bereiken.

```
[Device:d630a1d7-37a6-4d8a-beb1-560738ccda73] 192.168.1.53
Error: connect EHOSTUNREACH [192.168.1.53:1502](http://192.168.1.53:1502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1157:16) {
  errno: -113,
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.1.53',
  port: 1502
}
Client closed
```

Geos3, 2023-07-17:
Beste Edwin,

Idd. Je had gelijk ip adres is veranderd !!

Aangepast en het werkt.

Erg bedankt voor je snelle antwoord!!

Fijne avond 

Groet Floris

biemond, 2023-07-17:
kun je eens kijken op je telefoon of laptop  of je die mdns entry kan zien

Geos3, 2023-07-17:
Hoi Edwin,
Inmiddels opgelost !

Was ip nummer.

Erg bedankt voor jou snelle hulp!

Groet Floris

Geos3, 2023-07-17:
Beste Edwin,

Erg bedankt voor jou snelle hulp!

Was ip adres die was veranderd..

Groet Floris

closed 2023-07-17 by Geos3