biemond/solaredge.modbus #42: Values from Solaredge SE8K-RWS are

CaptainVoni, 2023-07-18:
I have a Solaredge SE8K-RWS inverter with a connect BYD Premium LVS 8.0 battery (installed last month). I installed this app on my Homey Pro 2023, activated Modbus TCP on the inverter via the Setapp, verified the port (1502) and device id (1), and then connected the app. Connection was immediately successful and it showed values, but the power consumption from my house is shown as "Solar Power", the actual solar power isn't anywhere shown, and the only flow action I have available is "Active Power Limit".

Any idea what has gone wrong?

I just bought Homey and my only installed app is this one, so maybe it's a simply me-issue ;-) I'm thankful for any help.

biemond, 2023-07-18:
hey, can you install the solaredge / storeedge device instead. 

[image]

biemond, 2023-07-18:
also you will see solar power/batt with battery charge and discharge

CaptainVoni, 2023-07-18:
On it. I told you, it's a me-issue.

biemond, 2023-07-18:
awesome. if it works also on the solaredge / growatt apps  side in homey you can press and send me a diagnostic report , just to verify it all.

biemond, 2023-07-18:
cool.  solaredge is not really doing this,  it is the sum of solar/batt with battery charge or at night it is solar/batt without batt discharge.
on the website they calculate this.

you can also calculate this. let me give you the flow for this.

CaptainVoni, 2023-07-18:
Nope, solar is zero now, consumption is 280 watts and battery discharge is constant at 70 watts - the rest comes from the grid.

biemond, 2023-07-18:
interesting , looks like meters are not providing the right load.  did you check on your smart meter?

if you go to homey apps on your mobile , can you go to solaredge app and send me a diagnostic report.

CaptainVoni, 2023-07-18:
Diagnostic report: c1b16859-9104-448a-b62e-7198983290f1

biemond, 2023-07-18:
cool so your inverter should adapt but 300w  diff is too much 
[image]
[image]

CaptainVoni, 2023-07-18:
[image]

biemond, 2023-07-18:
interesting. can you do a big load like a water cooker , oven etc.  batt discharge should handle this 

meters are fine. 
```
meter1-current 18 -1 Current
meter1-l1_current 4 xxx L1 Current
meter1-l2_current 4 xxx L2 Current
meter1-l3_current 9 xxx L3 Current
meter1-current_scale -1 xxx Current Scale Factor
meter1-voltage_ln 23221 -2 L-N Voltage
meter1-l1n_voltage 23221 xxx L1-N Voltage
meter1-l2n_voltage 23337 xxx L2-N Voltage
meter1-l3n_voltage 23324 xxx L3-N Voltage
meter1-voltage_ll 0 xxx L-L Voltage
meter1-l12_voltage 0 xxx L1-l2 Voltage
meter1-l23_voltage 0 xxx L2-l3 Voltage
meter1-l31_voltage 0 xxx L3-l1 Voltage
meter1-voltage_ln_scale -2 xxx Voltage Scale Factor
meter1-frequency 4999 -2 Frequency
meter1-frequency_scale -2 xxx Frequency Scale Factor
meter1-power -161 0 Power
meter1-l1_power -22 xxx L1 Power
meter1-l2_power -13 xxx L2 Power
meter1-l3_power -127 xxx L3 Power
```

CaptainVoni, 2023-07-18:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

biemond, 2023-07-18:
did you change the mode, can you put it on Maximize Self Consumption
now it is on remote control. 
```
storage_control_mode 4 xxx Storage Control Mode
storage_accharge_policy 2 xxx Storage AC Charge Policy
storage_accharge_Limit 0 xxx Storage AC Charge Limit
remote_control_command_mode 7 xxx Remote Control Command Mode
```

biemond, 2023-07-18:
this 1 , other on disabled. I think that will be auto when you the 1st
[image]
[image]

CaptainVoni, 2023-07-18:
Now I set it back to maximise self consumption: 79caaf3d-e36a-4784-af2d-1f173adb9f64

biemond, 2023-07-18:
nice, 

```
storage_control_mode 1 xxx Storage Control Mode
storage_accharge_policy 2 xxx Storage AC Charge Policy
storage_accharge_Limit 0 xxx Storage AC Charge Limit
remote_control_command_mode -1 xxx Remote Control Command Mode
```

biemond, 2023-07-18:
so now it is the same as me and should use batt to discharge to  match your internal consumption.

CaptainVoni, 2023-07-18:
This is crazy here. The settings jump all around the place. Limit mode should also be disabled?

No change in behaviour :-(

biemond, 2023-07-18:
yes also disable that  
[image]

biemond, 2023-07-18:
else do it from the solaredge inverter , maybe they detected some issue

CaptainVoni, 2023-07-18:
In the SetApp?

CaptainVoni, 2023-07-18:
It is set there to MSC. I only see that one option there, I guess the others are hidden, when this one is set to MSC.

biemond, 2023-07-18:
yep ,same .  but if now no discharge with a big load then call your installation guy or open a support ticket on solaredge. they can check it and see if something is holding it back.

CaptainVoni, 2023-07-18:
Ok, thanks, I guess I have to contact Solaredge then .

biemond, 2023-07-18:
at least you have some data to support your case besides the solaredge app.

CaptainVoni, 2023-07-19:
Good morning. Around midnight it started working normally again. No change from my side as I slept at that time 😉

If I want to use the different charge/discharge modes, how would you do that? I read that setting storage control mode to remote and then storage charge/discharge mode should do the trick. After yesterday I’m a bit unsure…

I want to disable battery discharging if grid prices are low.

biemond, 2023-07-19:
good, maybe it is some solaredge night job which fixed it all.  

never tried it but I guess storage control mode must be on disable , also no charge of batt
or do charge from pv and ac , fill your batt then when prices are low.

or do active power to 0 but also then no charge or discharge and only import.

mostly,  you want to control export of your PV when batt is full. you can control that with export limit .

CaptainVoni, 2023-07-19:
Thanks for all your help. I guess this doesn’t really have much to do with the initial problem, so I close this here.