biemond/solaredge.modbus #43: Mobile app tile vs web app tile

wsu-runner, 2023-07-21:
First - great app, thanks!

Noticed that the mobile app tile displays power generation but the web app tile displays inverter temp for my Solaredge inverter.

Temp is useful info but I'd prefer to see power generation on the web app tile.

Thanks for the consideration.

biemond, 2023-07-21:
hey, 

on the webapp you can do the 4 blocks and you will see it all.  
So in the phone app go to settings and change the status indicator. you can switch to another.  Sometimes you need to switch to something else and back. After that the webapp will use the same

wsu-runner, 2023-07-21:
Thanks. It was already the status indicator in the mobile app but I did as you suggested and switched to something else and back and the webapp picked up the right status indicator.

closed 2023-07-21 by wsu-runner