biemond/solaredge.modbus #44: Unexpected behaviour of "Battery is higher than xx" flow card

CaptainVoni, 2023-07-25:
Just take a look at the test run of this flow:
[image]

I assume that no words are necessary ;-)

My totally uneducated guess would be that the flow card "Battery is higher than xx" uses a battery SOC with decimal places, while the "Battery" variable is rounded to whole figures.

I would suggest to either add decimal places to the "Battery" variable or use the same rounded figure for the "Battery is higher than xx" flow card to achieve a consistent behaviour.

biemond, 2023-07-25:
[image]
works for me on both battery vars of solar I got 41.99 back.   Did you use the right batt variable.

CaptainVoni, 2023-07-25:
You used twice the logic flow card, which accesses the same variable. I used the logic flow to access the "Battery" variable (as you did in your example), but I also used the flow card "Battery is higher than" from the app itself. They behave differently.

biemond, 2023-07-25:
you can also use this, I don't specify any decimal , it is a number and nothing special on my side.  will check it again
[image]

CaptainVoni, 2023-07-25:
Maybe I wasn't clear in my initial report: I consider it problematic if there are two possibilities to check if the battery is above x%, but they behave differently. It's like asking the same question twice with different words and receiving a different answer.

Also I consider it problematic that the flow card "Battery is higher than x%" returns yes, but the Battery variable just shows that "x".

This all together might lead to errors.

biemond, 2023-07-25:
just always double check what works , most comes out of the box of homey and I have influence on.

mroeben, 2024-01-27:
I am with CaptainVoni. I've stumbled across this behaviour also and found it quite confusing. Refer to the [post on the Homey forum](https://community.homey.app/t/app-pro-solaredge-growatt-tcp-modbus/45487/795?u=jumpingtrees).

When I print the two variables to the timeline I do not see any difference.

[image]

biemond, 2024-01-27:
so I have 2 batt caps because else you won't see it on the device , only on the batt tab of the device

I can change the label of 1 so they are not exactly the same.
Also can add a delay on the changed batt trigger 

```
           if (this.validResultRecord(result['batt1-soe'])) { 
                this.addCapability('battery');
                this.addCapability('measure_battery');
                var battery = Number(Number.parseFloat(result['batt1-soe'].value).toFixed(2));
                if ( battery > 0 ) {
                    if (this.getCapabilityValue('battery') != battery) {
                        this.homey.flow.getDeviceTriggerCard('changedBattery').trigger(this, { charge: battery }, {});
                    }
                    this.setCapabilityValue('battery', battery);
                    this.setCapabilityValue('measure_battery', battery);
                }
            }

```