biemond/solaredge.modbus #45: No data from GroWatt Battery Inverter

Xybertecnic, 2023-07-29:
I recently installed a GroWatt Battery Inverter next to my SolarEdge Solar panel Inverter, and I was trying to connect it to Homey too. I bought a Modbus TCP converter (not the same brand/model because I wanted to connect through Wifi, but it is supposed to do the same thing), I have connected the RS485-3 port to the Modbus converter, and it is set up to connect to my wifi network. I can reach the device just fine, and the RS485 parameter is set to VPP. However I do not get any data inside homey, I can add the device, but the data fields stay empty. I set the communication parameters also as in the documentation explained in the dropbox link.

[image]
[image]
[image]
[image]
[image]

Also tried with inverter ID 1, but same result. Also set the RS485 converter to port 502, but didn't change anything either.
Lastly I tried modbus polling on, but also no change. I'm not sure what all the other settings are for.
Any hints or ideas?

Xybertecnic, 2023-07-29:
Uh, never mind, I should have put it as a server obviously and not as a client...
Can be closed, sorry to bother and thanks again for a great app!

closed 2023-07-29 by Xybertecnic

biemond, 2023-07-29:
Hey,

what is your inverter type/version?  plus what cable did you use ? on the
forum they use some special rj45 cable with only a few ones connected.
I know Phil bro is the expert on this.

I expect your cable is wrong.

thanks

biemond, 2023-07-29:
use the test version for more flows https://homey.app/nl-nl/app/solaredge.modbus/SolarEdge-+-Growatt-TCP-modbus/test/

biemond, 2023-07-29:
on the forum can you describe your setup so others can learn from this  https://community.homey.app/t/app-pro-solaredge-growatt-tcp-modbus/45487/607

thanks

Xybertecnic, 2023-07-29:
Does the test version change anything for the solar edge solar inverter? Or does it just add commands for the growatt battery inverter?
If so then I will try, I don’t want my flows for the SolarEdge to mess up. For the GroWatt I still have to figure out everything, I just recently have installed and connected it, and for now it is still running automatically with the included energy meter, I still have to figure out how to disable that.
Thanks

biemond, 2023-07-29:
yep, only for growatt , no change for solaredge. you can remove growatt device if you want to see all capabilities , but you get now a lot of flows.