biemond/solaredge.modbus #47: Value of "Storage Charge/Discharge Mode"

CaptainVoni, 2023-08-11:
The value of "Storage Charge/Discharge Mode" is defined as text, so I assumed it will return the label of the respective mode. But instead it returns a number, e.g. in case of Maximize self-consumption it will contain "7" as value.

Is this intentionally? If so, how can we verify if a specific mode is set, i.e. is there somewhere a documented mapping between the label and the number?

Shouldnt it be possible to set the mode to a specific value and when verifying the mode, that same value is returned?

biemond, 2023-08-11:
Hey, 

you can check here. for me it is a enum. don't know why homey does not show this https://github.com/biemond/solaredge.modbus/blob/main/.homeycompose/capabilities/storagedefaultmode.json

and it is a number on solaredge not a text.

biemond, 2023-08-11:
I think they do this because of locale and when you switch language it should still work and don't get in trouble.

CaptainVoni, 2023-08-15:
If I understand you correct it is not an issue of this app, but of Homey. I guess I should report it there then, correct?

I understand your point regarding switching the language, but it should be possible to solve this in some other way. It just doesnt make any sense, when you set some value to X, then retrieve that value and you get Y. That's just begging for troubles in my opinion.

biemond, 2023-08-15:
I think you used some logical value or not. this is direct on the capability.

I can add it , what do you need?  some  "And condition" so you can compare.  you need to provide the state and give back true / false.   Or you can get status changed trigger but that is only possible when you trigger it.

CaptainVoni, 2023-08-15:
That's exactly what I need. In some flows I need to know the current value of "Storage Charge/Discharge Mode". I could use the logic flow card "<Text/number> is equal to <value>", but the logical value is a number, while if I use the Then-flow-card "Storage Charge/Discharge Mode" to set the mode, it expects a text.

The same is (at least) also true for "Storage Control Mode": you set it as text, but when you retrieve it, you will get a number.

And btw: maybe adding a "Set " at the beginning of the Then-flow-card for e.g.  "Storage Charge/Discharge Mode" and " to " at the end would make the flows more readable. Example: "Set Storage Charge/Discharge Mode to" instead of just "Storage Charge/Discharge Mode" for the flow card. This way it would also be easier to distinguish between the flow cards and the values/capabilities - right now they have the same name.

CaptainVoni, 2023-08-15:
And: who is providing the translations? Right now German is a big mess - some English, some German - I could provide proper translations for the cards/variables if you want.

biemond, 2023-08-15:
interesting I don't do german , only english and some dutch. So I guess for the default stuff , homey try to do the locale.

biemond, 2023-08-15:
this is the main 1 but it generated from other places https://github.com/biemond/solaredge.modbus/blob/main/app.json but at least you get the idea

CaptainVoni, 2023-08-15:
Never used github, so I'm not sure how's the best way to provide you the translations. Should I just edit the file ("Edit in place") and you accept/reject the changes?

biemond, 2023-08-15:
you can download it and upload it here.  but it will be a lot of work

CaptainVoni, 2023-08-15:
See the attached file (as zip as json is not accepted here). I translated just the triggers for now, because before I translate everything I want to make sure that it's ok for you how I'm doing it. I may update some of this translations when I do the rest simply for consistency.

[app.zip](https://github.com/biemond/solaredge.modbus/files/12346561/app.zip)

CaptainVoni, 2023-08-17:
Should I interpet your thumbs up as "Do it"? ;-)

biemond, 2023-08-17:
will check it in the weekend and will make a test version for you which you can verify.

biemond, 2023-08-20:
here we go https://homey.app/a/solaredge.modbus/test/ and thanks again.

CaptainVoni, 2023-08-20:
Worked like a charm. The translations are - as expected - pretty inconsistent, so I have to take some time to sort that out.

Is it correct, that this test version only contains the german translations, but not the "and condition" we discussed? If it is included, I can't find it.

Where did you get the labels for the charge/discharge modes for the Solaredge inverters, e.g. "Charge excess PV power only"? They are different from the labels Solaredge itself uses in their [documentation](https://knowledge-center.solaredge.com/sites/kc/files/storedge_charge_discharge_profile_programming.pdf). Should I update them (the english ones) to fit the labels from Solaredge? (Especially the "Charge excess PV power only" is pretty misleading in my opinion)

And one final question if I may: where did the values that may be selected for "Set Discharge power" come from? I'm asking because I would need 0 as value (or 1 as fallback). And before you ask: no, there is no fitting battery mode for my case ;-)

I will start working on the translations in the coming days and let you know as soon as I'm finished.

biemond, 2023-08-20:
condition is still to do. 

everything comes from their modbus doc.

[image]
[image]

CaptainVoni, 2023-08-20:
Interesting... they use different labels in different documents for the same functionality. Do you have a link to that document?

biemond, 2023-08-20:
I calculate those vaules because it goes over 2 register .  so made it easier for myself  and easy to calculate those values

https://github.com/biemond/solaredge.modbus/blob/main/drivers/invertorwithbatt/driver.flow.compose.json#L629

biemond, 2023-08-20:
[Power Control Open Protocol for SolarEdge Inverters.pdf](https://github.com/biemond/solaredge.modbus/files/12389010/Power.Control.Open.Protocol.for.SolarEdge.Inverters.pdf)

biemond, 2023-08-20:
profiles I cannot support and it is done by solaredge NL. I know on BE some can do this. plus I don't have installation account. I know here they can do this and you need to use a fake business name https://www.solar-bouwmarkt.nl/solaredge-omvormers-solaredge-monitoring-portal-in.html and solaredge is not happy with this.

CaptainVoni, 2023-08-20:
> I calculate those vaules because it goes over 2 register . so made it easier for myself and easy to calculate those values
So it's not possible to set the discharging power to zero or do you have to make some calculations?

biemond, 2023-08-20:
yes 0 is possible. was thinking I already supported that. will add that

biemond, 2023-08-20:
oh it was for import / export limit https://github.com/biemond/solaredge.modbus/blob/main/drivers/invertorwithbatt/device.ts#L299

CaptainVoni, 2023-08-20:
> profiles I cannot support and it is done by solaredge NL. I know on BE some can do this. plus I don't have installation account. I know here they can do this and you need to use a fake business name [...]

Didn't really understand this.

biemond, 2023-08-20:
like this
[storedge_charge_discharge_profile_programming.pdf](https://github.com/biemond/solaredge.modbus/files/12389089/storedge_charge_discharge_profile_programming.pdf)

CaptainVoni, 2023-08-26:
I just saw that it was automatically installed last night. Will test it and report back.

FYI: I'm about 3/4 through the translations; I guess I will finish in the next days.

CaptainVoni, 2023-08-26:
Here my results:
- Setting the discharge power to zero results in resetting it to 5000W (seems to be the default), i.e. no effect.
- Setting the charge power to zero works fine, i.e. charging is prevented.
- Querying the charge/discharge mode works fine.

biemond, 2023-08-26:
thanks , shall  we do 50w discharge instead?

CaptainVoni, 2023-08-26:
May I vote for one Watt? We should have the absolute minimum as option. 

In case you want to know why: if the price from the grid is lower than from the battery, the inverter should create AC for self consumption, then charge the battery and feed any excess to the grid, but if the PV yield is too low to cover the consumption it should **not** use the battery. There is no charge/discharge mode for this scenario. Right now I switch between Max Self Consumption and Off depending on the PV yield, but to do so I need to trigger a flow whenever consumption or PV yield changes (i.e. a LOT!). If I could set the max discharge power to a very low value, I could use simply Max Self Consumption with the discharge power set.

biemond, 2023-08-26:
yep lets try that.

biemond, 2023-08-26:
in that case I would put active power to 0 and import it all from grid and don't do own consumption.

CaptainVoni, 2023-08-27:
> in that case I would put active power to 0 and import it all from grid and don't do own consumption.

But if you set Active Power to 0, you would lose the complete PV yield, as it just shuts down the PV inverter:

> Active Power Limit dynamic control: This register controls the active power limit of the inverter dynamically. It is set as the 
percentage of the inverter’s maximum power. For example: when setting the register of SE5000 to 20, it will limit the inverter to 
1000W which is 20% of 5000W.

And I always thought that there are two inverters: one to convert PV-DC to AC and one to convert Battery-DC to AC. This would explain why there are two individual limits for these two. If this is true, setting Active Power to zero wouldnt affect battery discharging.

biemond, 2023-08-27:
try it out because if the price goes very negative ,  import it all from grid.  else do self consumption and no export. if the price is ok, do self consumption and full export. 

the challenge is winter , to check with solcast or the price , the plan for the next day when to charge the batt and when not.

CaptainVoni, 2023-08-27:
Just tested it and this obviously is wrong:

> And I always thought that there are two inverters: one to convert PV-DC to AC and one to convert Battery-DC to AC. This would explain why there are two individual limits for these two. If this is true, setting Active Power to zero wouldnt affect battery discharging.

If Active Power Limit is set to zero (and Charge/Discharge Mode is set to Maximize Self Consumption and the battery is fully charged), all the power is taken from the grid. So Active Power Limit somehow limits also battery discharging, but it does not limit charging. I guess, there is only one DC-AC inverter, which is used for battery and PV together.

> in that case I would put active power to 0 and import it all from grid and don't do own consumption.

I guess you assume that the price for electricity **from** the grid is identical to the price **to** the grid at my place, which isn't the case. The price from the grid changes hourly and is negative from time to time; the price to the grid is determined monthly and has never been negative or even close to zero.
Also the price for electricity out of the battery is quite high if you consider the cost for the battery. Right now I calculate 0.217€/kWh for electricity out of the battery.
So whenever the grid price is below 0.217€, I want the battery to be charged (if possible), but not discharged (grid is cheaper). At the same time the PV production should not be restricted, because feeding excess PV production to the grid earns me money.

There might be some situations when it would be advantageous to disable PV production with Active Power Limit, but not only does the grid price has to be (very) negative, the consumption also has to be high. I can think of some rules for that, but they are quite complex. Thank you for giving me something to think about the next weeks :-D

CaptainVoni, 2023-08-27:
> the challenge is winter , to check with solcast or the price , the plan for the next day when to charge the batt and when not.

That's on my TODO list. Estimate how long the battery will last (done) and then charge it from the grid at the cheapest hour within that period.

biemond, 2023-08-27:
yep , here in NL it is around -19c , we had sometimes this summer -49c . yes it is unbelievable how much tax is added. I think we have added tax + energy tax and transport cost.

CaptainVoni, 2023-08-27:
I'm finished with the translations, but I'm uncertain with regard to "Total Yield" and "Total Day Yield": to me it seems that this isn't a "yield" but more a "consumption". E.g. meter_power.daily is displayed in Homey as Total Yield, but it changes when there is no PV production. Also judged purely from "meter_power", this should be the energy measured by the house meter, i.e. the electricity drawn from the grid.

I'm definitively no english native speaker, but at least to me this is confusing.

Do you have an opinion on this? Is the term "yield" correct in your view?

biemond, 2023-08-27:
I think today yield and day yield is correct.  don't see how  meter_power.daily is displayed in Homey as Total Yield. it show correct for me 
no it is total yield not consumption or smart meter. you have losses on everything etc.  but it is what measured and what should be possible. 
What does yield mean in solar? Energy yield is the amount of energy actually harvested from solar panels, taking into consideration external factors like heat, dirt, and shade, whereas efficiency refers to testing done in lab conditions.

[image]
[image]

CaptainVoni, 2023-08-28:
I'm a bit confused about that total day yield, because it shows (very) different numbers from what I see in the Solaredge app. But lets stick with "Yield"/"Ertrag".

See the attached [app.zip](https://github.com/biemond/solaredge.modbus/files/12451992/app.zip) for the complete german translation. As discussed, the file includes five lines starting with "TODO:" with some remarks/questions from me.

Let me know if anything is unclear.

biemond, 2023-08-28:
yep solaredge calculates it or has more data which we don't have this.

biemond, 2023-09-02:
done , can you test again https://homey.app/a/solaredge.modbus/test/

CaptainVoni, 2023-09-02:
Zis looks all weri German too me! 😄
Translations looks great, a handful flow cards are missing translation or the translation is missing „Setze“ at the beginning. I’ll download the latest json and try to figure out what went wrong.

Setting the discharge power to 10 W again doesn’t work - it is set to the 5000 W. 😞 Is there nothing in the documentation which specifies the accepted minimum?

Thank you for your effort!

CaptainVoni, 2023-09-09:
Yeah, I already know that you updated it. ;-)
Suddenly all the references in flows to the inverter were gone. After a reload they were back. :-)

Testing now.

biemond, 2023-09-09:
here we go , new version https://homey.app/a/solaredge.modbus/test/

biemond, 2023-09-09:
I added voltage and now this version has your locale

CaptainVoni, 2023-09-11:
Checked the voltages and verified all titles (English and German):
* All the voltages "measure_voltage.phaseX" are reported as value around 400V, which doesn't make much sense for a single-phase value. In contrast, the voltages "measure_voltage.meter_phaseX" report values around 230V, which seems to be fine.
* One capability had a title for an action. Fixed that.
* I added recommendations for the German and English titles of all actions in order to be in-line with the general titles of actions in Homey.

[app-3.0.43 (updated).zip](https://github.com/biemond/solaredge.modbus/files/12576485/app-3.0.43.updated.zip)

biemond, 2023-09-11:
yes meter ones are beter , for me somehow it is 240 for phase 1. 

All the voltages "measure_voltage.phaseX" are reported as value around 400V, which doesn't make much sense for a single-phase value

biemond, 2023-09-13:
pushed a new version

biemond, 2023-09-14:
thanks , yep can you find out what minimal discharge works for you , from the inverter menu.

CaptainVoni, 2023-09-14:
> thanks , yep can you find out what minimal discharge works for you , from the inverter menu.

What do you mean by "inverter menu"? I just checked everything I can find in SetApp and also on the monitoring website, but I can't find anything about the discharge power of the battery.

biemond, 2023-09-14:
ok will do some checks

CaptainVoni, 2023-09-22:
I guess I should report this to the Homey support, right?

[image]

- When I use a logic card to check if the battery discharge power ("Batterieentladeleistung") is greater than zero, it is not (correct).
- When I use the App flow card to check if the battery discharge power is greater than zero, it is (incorrect).
- The battery discharge power actually is zero. 

I'm under the impression that the Homey built-in flow cards for capabilities are buggy.

biemond, 2023-09-22:
oh let me verify this.  thanks for reporting.

biemond, 2023-09-23:
indeed I see the same, always true

biemond, 2023-09-23:
So greater than works for me, but you cannot test them from advanced flow. there is for me , no true or false output. and then the value is not tested

[image]

biemond, 2023-09-23:
so discharge 0 ,10,50 and 100 is working for me

<img width="498" alt="Screen Shot 2023-09-23 at 20 00 56" src="https://github.com/biemond/solaredge.modbus/assets/1085004/529c0816-5f60-478b-b155-e17c67ad16
[image]
[image]
[image]
[image]
[image]
0c">

CaptainVoni, 2023-09-23:
> so discharge 0 ,10,50 and 100 is working for me

Tested all of them right now: all are working for me except zero. But 10 W is also perfectly fine for me. Many thanks for your efforts! When do you think will you release this version?

CaptainVoni, 2023-09-23:
Are you planning on adding such trigger?

>     * A trigger for a change of the storage control mode and the storage charge/discharge mode is missing. If you want to add them you can use "Speicherkontrollmodus hat sich geändert" (storage control mode) and "Speicher-Lade/Entlade-Modus hat sich geändert" (storage charge/discharge mode) as translation.

I'm asking because my battery turns itself off every day between 21:00 and 22:00 and I want to react on that.

biemond, 2023-09-23:
no problem. also I think the default of remote action = 6 hours , than you need to do something else it will go to the default again.

let me ask to publish it to release

CaptainVoni, 2023-09-23:
> no problem. also I think the default of remote action = 6 hours , than you need to do something else it will go to the default again.

Oh, that's interesting! Will check that. Thanks for the hint!

[21 more replies]