biemond/solaredge.modbus #48: Any chance of support for Huawei Solar?

Melmixx, 2023-08-13:
Hi. There are currently major issues with the Huawei Solar app for Homey, because Huawei API changed and web also changes all the time. The developer can't keep up and mentioned modbus as a potential future option, but he doesn't have the time. I can see you have a modbus solution and you're already integrating with multiple inverters. Any chance you could include the Huawei Sun2000 inverter? I have no clue if this is of any help, but I found this on the modbus interface definitions:

https://www-file.huawei.com/~/media/CORPORATE/PDF/FusionSolar/HUAWEI_SUN2000_245KTL28KTL_MODBUS_Interface_Definitions_20150715_JP.pdf

Thank you and sorry if this is not the way to do it. I tried to add an enhancement tag, but can't figure out how :)

biemond, 2023-08-13:
oh let me check.

do you have the modbus option on your inverter?  also do you have some vpn so I can test it with yours ?

but it is doable.  

thanks

Melmixx, 2023-08-13:
Thank you for your speedy reply! I have found several outdated guides on how to enable modbus, so I'm sure I do. I can see that it requires me to have installer login access to the inverter, which I currently do not as my installer seems to have changed the default password. So I'm going to reach out to them and ask for the password, then I'll activate modbus and then I'll figure out a way to get you VPN access. I'll get back to you as soon as possible. Thanks again.

EV-BD, 2023-09-01:
I have ModBus enabled on my Huawei inverter and have the Ethernet adapter arriving next week I hope.

I'd be delighted to help test/get this working.

biemond, 2023-09-01:
nice. let me know, so we can test it.

EV-BD, 2023-09-01:
👍🏻

EV-BD, 2023-09-06:
So I got the FE connected and ModBus poll connects (anny suggestions for a better/free/cheap tool), however it reports "02 Illegal data address".

Any suggestion on how to proceed?

It also seems that when FE is connected, the Huawei app will not pull data over the internet.

If we can grab near realtime data and pull it into Homey that won't be an issue. ;)

EV-BD, 2023-09-06:
I have the FE connected, Modbus poll connects but reports "02 Illegal Data Address".

Any ideas or suggestions? Also for a better free/cheap tool than ModBus poll?

When FE is running the FusionSolar app does not read any data, but if we can get Homey pulling near-Realtime data that won;t be an issue. :)

biemond, 2023-09-06:
hey 

can you install node and this jsmodbus npm 

then run this script node test.js
https://github.com/biemond/solaredge.modbus/blob/main/test.js

you need to change or strip this script and looking at your pdf with registers plus check the return type etc. 
and see if you get some good values. start with 1 and expand

of give me vpn access and I will check it.

EV-BD, 2023-09-07:
Maybe on Monday I can look at that.
Alternatively, if I gave you my public IP and forwarded a port to the FE, would that work?

biemond, 2023-09-07:
yeah , I can give you my ip so you can use that on the firewall.  modbus is not really protected

send a mail to  biemond at gmail

biemond, 2023-09-09:
have some success

used these registers
https://gitlab.com/Emilv2/huawei-solar/-/blob/master/src/huawei_solar/registers.py

node test2.js
-------------------
Connected ...
UserRequestError {
  err: 'Timeout',
  message: 'Req timed out',
  response: undefined
}
Model: SUN2000-6KTL-L1
Version: 343
Serial: aaaa
PN: aaaa
grid voltage: 2346