biemond/solaredge.modbus #49: function descriptions

kicksev, 2023-08-20:
I recently got a Solaredge inverter 10.5K 3-phase with a 9 KWh battery system and a dynamic energy contract. For this, I'm looking for a way to control the inverter, and to my great delight, I found this app for my Homey. Now, I'm looking for some information on what the provided functions do and the meaning of the different statuses. Probably, this info is available somewhere, but I can't find it. Can you give me a hint?

biemond, 2023-08-20:
hey, yes please check the solaredge manuals but you can basically limit import (use batt more) , export (with negative prices)  . or charge / disacharge the battery to grid when prices are ok. 

[SolarEdge_StorEdge_Manual_V2.pdf](https://github.com/biemond/solaredge.modbus/files/12388068/SolarEdge_StorEdge_Manual_V2.pdf)

biemond, 2023-08-20:
[image]
[image]

kicksev, 2023-08-20:
Thank you for your help.

closed 2023-08-20 by kicksev