biemond/solaredge.modbus #50: Invert energy readings

lovethyresson, 2023-08-21:
In order to fit into Homeys energy feature it would be great with a switch to let you invert the value sent by the inverter.

eg if my SE inverter produces 6500W I want the app to say -6500W

biemond, 2023-08-21:
https://apps.developer.homey.app/the-basics/devices/energy#generating-devices

Devices, such as solar panels, can generate their own energy. Homey shows these devices in the outer ring in the Energy tab.

Homey will show this device separately in the UI. **You should provide the generated power as a positive value**. When providing a negative value, e.g. -13 watt, Homey assumes the solar panel is currently consuming instead of generating energy.

lovethyresson, 2023-08-25:
Hey, oh. Thanks! Totally misunderstood that then :)

closed 2023-08-25 by lovethyresson