biemond/solaredge.modbus #51: Connect Solar Edge

BartVlaar, 2023-08-23:
Beste Edwin,

Ik ben een nieuwe gebruiker van Homey Pro 2023 en probeer een connectie te maken met mijn SolarEdge omvormer SE3500H (zonder display). Ik heb hiervoor jouw app "SolarEdge + Growatt TCP modbus" geïnstalleerd maar heb geen idee hoe ik aan de juiste inverter settings kom: IP address, Port en Inverter ID

Hopelijk kun je me hier mee helpen, alvast bedankt voor je reactie.

Bart

BartVlaar, 2023-08-23:
Hi Edwin, excuses ik kom net een verwijzing naar een handleiding tegen op de community page. Deze zal ik eerst raadplegen, als ik er dan nog niet uitkom dan meld ik me weer.

Bart

BartVlaar, 2023-08-28:
Inmiddels heb ik het IP address achterhaald en de instellingen uit de eerste post op de omvormer doorlopen. Helaas wordt hij nog niet gevonden. De omvormers is via wifi met mijn netwerk verbonden en niet via een kabel, zou dat het probleem kunnen zijn?

Alvast bedankt voor een reactie.

Bart

biemond, 2023-08-28:
hey,

kun je eens kijken of je poort open is op dit ip adress?  
Voor wifi , er is een versie waar ze modbus hebben disabled. bij recente versies laten ze het weer toe.

gr Edwin

BartVlaar, 2023-08-30:
Hi Edwin,

Inmiddels heb ik het volgende uitgevoerd maar nog zonder succes.

- IP adres van de omvormer statisch gemaakt en de poort 1502 open gezet in mijn ziggo smartwifi modem. Al ben ik niet zeker of het poort 1502 betreft. 
- Gecontroleerd met Angry IP scanner of de poort open staat op het toegewezen IP adres. De poort staat niet open ondanks dat ik deze dus is de Ziggo modem heb open gezet. Ik vermoed dus dat het hem in de omvormer zit. _(Alive host - is the host, responding to pinging. These are blue in the results list. Open port - a TCP port, responding to connection attempts. Hosts with open ports are green in the results list.) zie screendump bijlage.
- In een andere post had ik gezien dat je verwees naar mDNS discovery, deze heb ik geïnstalleerd en gezocht op: solardedge-modbus maar dat gaf geen resultaat.
- Ook de stappen voor het instellen van de RS485-2 in de omvormer heb ik nogmaals doorlopen, maar nog altijd geen resultaat helaas. Mijn kennis op dit vlak is nihil, maar ik vroeg me af waarom de RS485-2 instellen als ik contact maak via wifi.
- Ik heb een aantal screenshots toegevoegd van Angry IP en de SolarEdge instellingen.

Hopelijk heb je nog wat tips want ik kom er niet uit. Als je extra informatie nodig hebt om er iets zinnigs over te kunnen zeggen dan hoor ik het ook graag.

Alvast bedankt voor je reactie.

Bart
[image]
ge.modbus/assets/65311699/8d80d25d-1398-41bf-aaf0-cc398dfa1969)

[image]

biemond, 2023-08-30:
kun je je installateur or solaredge vragen om je firmware te updaten. ik denk dat je niet de meest recente versie hebt.
of je moet even een lan kabel gebruiken of ethernet over je elektrische leidingen gebruiken.

biemond, 2023-08-30:
ipv wifi.  solaredge had op een bepaald moment modbus over wifi uitgezet.   kan ook zijn dat ziggo wifi clients onderling connectie uit heeft gezet.

BartVlaar, 2023-08-30:
Okay ik ga kijken of ik de firmware kan updaten. 
Wederom bedankt, ik laat je weten hoe het afloopt.

green-us, 2024-05-21:
settings die iig. voor even hebben gewerkt twee weken terug, via Wifi ook nog!) (2022 HDwave), echter momenteel werkt het niet meer. Mogelijk dat het verband houdt met de tariefoptimialisatie functie die SE nu zelf in de app heeft geïntroduceerd en recent ook verbetered (dat werkt nu goed).:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]