biemond/solaredge.modbus #52: Solar Edge app not sending live production

Gits74, 2023-08-26:
My solar edge inverter is not sending live production to my homey therefor my solar edge app is not starting flows.

biemond, 2023-08-26:
hey, can you send me a diagnostic report. mostly the ip is not correct or modbus is not enabled.

Gits74, 2023-08-26:
Hi
Yes ip is probably wrong. How do i find the ip address. The inverter is
connected with lan cable.

Stig

biemond, 2023-08-26:
sometimes you see on mdns a solaredge entry , else check your router or inverter which ip you got.

RampyLou, 2023-08-27:
[image]
only 1 response at 22:35 from this flow -> see below

"Homey
U
SE accu = 83.4, 22:35 - 26-08-2023"

I think I only get a connection to the modbus of the SE once, this is set and active, everything worked perfectly on the Homey2019, now not on the H2023
Diagnostic raport : Oeps, zo te zien werkt de app niet

biemond, 2023-08-27:
did you remove it from homey 2019? you cannot do both. 
and we all got 2023 homey and it works perfectly on both .

RampyLou, 2023-08-27:
yes removed from  the 2019, deinstall - reinstall still the same diagreport oeps....

RampyLou, 2023-08-27:
deinstall - reinstall on the 2023 :)

tonsemail, 2023-09-29:
Hi I have a Home Pro 2023. Modbus is working. I get current, temperature and voltage. The watts are doing nothing
What is the ID used for, or where can i find it

RampyLou, 2023-09-29:
norm your ip, port 1502, ID 1

biemond, 2023-09-29:
id is the modbus identifier , mostly the inverter = 1 . but can you send me a diagnostic report.  don'tknow why you got  temp but not the rest.

RampyLou, 2023-10-07:
> did you remove it from homey 2019? you cannot do both. and we all got 2023 homey and it works perfectly on both .

is it possible because a FlexiO is also connected to the SE (raspPi) that the modbus ignores the Homey app ?

biemond, 2023-10-07:
you mean Pi blocks homey?  what homey does it connects for a few seconds and then not for 20 seconds or so. so with 2 devices on homey or 2 homey you can still do both or connect to it yourself.  sometimes a conflict but it retries at least.

So no clue how other apps are doing things.  probably they keep the connection alive.

I know there are some modbus proxy which connects to the inverter  and others connect to this proxy.