biemond/solaredge.modbus #54: Is it possible to link the voltages of all 3 phases to the app?

Burpman2000, 2023-09-06:
I have a Solaredge 3-phase inverter, Phase 3 generally has a much higher voltage than the other 3 phases (Often above 253 volts, thus switching off the inverter), if I charge the car on this phase, phase 1 will still run high.
So I will have to create an advanced flow that goes something like: if the voltage of Phase 1 is higher than 252 volts OR if the voltage of Phase 2 is higher than 252 volts OR if the voltage of Phase 3 is higher than 252 volts THEN reduce production output to 20%

Below is my log of the past 24 hours, voltage per phase per minute put in a DB and graphically displayed via Highcharts, huge differences over the day.

https://wilco-boerman.nl/media/voltage24uur.jpg

Sincerely, WilcoB

biemond, 2023-09-06:
Hey, 

I think I show you some voltages and with logical values you can check the values. and probably do active power to 80% of max of your inverter output.

So you want to know the l1,l2,l3 voltages?

thanks

Burpman2000, 2023-09-07:
Dear Edwin

It would be nice if 80% would work, but in the afternoon between 13:00 and 16:00 our neighborhood is so heavily overloaded that even at 20% 253 volts are sometimes reached ;-(

See yesterday's graph as an attachment, it doesn't make you very happy.

Panels were installed just three weeks ago (18 panels, 9 east and 9 west, 500 wp Bisol, and everything is already prepared for the arrival of a battery)
Even in September the voltage was too high, so that doesn't look good for next year.

Enexis reports "Voltage problems are known in your neighbourhood".

I am already emailing Enexis, but get the standard answer, the problem is known, and we will keep you informed, but plus and minus 10% (of 230 Volt) is allowed, But with us it sometimes taps 204 Volt , goes beyond that 10%

Yes, I would really like to know all three voltages in my Homey.
Via the raspberry I already read this out via the modbus of the Solaredge.
But if I could also see it in the Homey and use it in flows that would be great.

Regards Wilco

[image]

biemond, 2023-09-07:
will do.  expect in the weekend to add them.

Burpman2000, 2023-09-07:
Ha, that's nice!!!

Thank you.

biemond, 2023-09-09:
here we go , can you test this 1 https://homey.app/a/solaredge.modbus/test/

Burpman2000, 2023-09-09:
Hello Edwin.

I think we're almost there.
I think the Homey app now shows lX_voltage and I would really like to see the lXn_voltage, see images below.

Is the lX_voltage perhaps the voltage of the Current transformer (CTs) meter, and the lXn_voltage the converted (actual) voltage?

[image]

[image]

Regards Wilco

biemond, 2023-09-09:
so the ones from your meter.  will do it now.

biemond, 2023-09-09:
please try this one https://homey.app/a/solaredge.modbus/test/

Burpman2000, 2023-09-09:
jesss, in version 3.0.43 I can read the voltages as I wanted, great that you did this so quickly, thank you!!!

[image]