biemond/solaredge.modbus #55: Active power limit

MarcovBarneveld, 2023-09-10:
I have 2 SolerEdge inverters.
First one is a SE3500H
Last week I commisioned a SE3000H.
Now I want to set an active power limit on the SE3500H with homey.
That is not working correctly.
When I set the limit for the SE3500H it is executed on the SE3000H.
When I set the limit for the SE3000H it is excecuted correctly on the SE3000H.
Bothe have different IP adresses and show correctIy in Homey.
I already changed modbus port to 1503 and the inverter id to 2, but I still can't set the limit on the SE3500H.

biemond, 2023-09-10:
ah, let me check if I pick up right info from the device.  I think the 2 are linked or not?

biemond, 2023-09-10:
don't see it but because they linked together, 1 is master and other follows.
can you make a diagnostic report  when you do SE3500H .

I think I need to use a different register ,  also when you do it on SE3500H from the menu, do you see it correctly on homey?

MarcovBarneveld, 2023-09-10:
The two are not linked (yet).

biemond, 2023-09-10:
ok maybe they removed it or your firmware is behind.

can you do it again on se3500 and then right after send me a report. like to see if it is denied or not.

MarcovBarneveld, 2023-09-11:
Firmware is up-to-date on both.
SE3500H:
Communicatie:ETHERNET
DSP1:1.100.1515
DSP2:2.100.1302
DSP3:NA
CPU-Versie:4.18.32

SE3000H:
DSP1:1.100.1515
DSP2:2.100.1302
DSP3:NA
CPU-Versie:4.18.32

Made two reports.
One for the homey itself
"3D686EF8"
and for you app
4c142be8-6eea-4603-b1cb-671543e4790c
Is that OK? Don't know wich one you need.

You say "_denied_".
does that mean that when I execute both inverters should go to the same limit?

biemond, 2023-09-11:
61

```
Client closed
{
  host: '192.168.1.61',
  port: 1502,
  unitId: 1,
  timeout: 15,
  autoReconnect: false,
  logLabel: 'solaredge Inverter',
  logLevel: 'error',
  logEnabled: true
}
Connected ...
activepowerlimit {
  metrics: UserRequestMetrics {
    createdAt: 2023-09-11T13:26:58.982Z,
    startedAt: 2023-09-11T13:26:58.983Z,
    receivedAt: 2023-09-11T13:26:59.013Z
  },
  request: ModbusTCPRequest {
    _id: 1,
    _protocol: 0,
    _length: 6,
    _unitId: 1,
    _body: WriteSingleRegisterRequestBody {
      _fc: 6,
      _address: 61441,
      _value: 50
    }
  },
  response: ModbusTCPResponse {
    _id: 1,
    _protocol: 0,
    _bodyLength: 6,
    _unitId: 1,
    _body: WriteSingleRegisterResponseBody {
      _fc: 6,
      _address: 61441,
      _value: 50
    }
  }
}
```
```
c_model SE3000H-RW000BEN4                xxx Model
active_power_limit 50 xxx Active Power Limit
advancedpwrcontrolen 1 xxx Advanced Power Control En
```

biemond, 2023-09-11:
can you remove SE3000H device from homey.  and do it again.  don't .60 ip one. 
send me an app  report , no need for a homey 1.

MarcovBarneveld, 2023-09-12:
That will leave a whole bunch of broken flows :-((
If I understand correctly you want the SE3500H on an other IP so I have to change that to?

I let you know when I have done so.

biemond, 2023-09-12:
maybe  you just selected the wrong device on your flow.  or maybe change the ip on se3000h for now so it fails 
You say the inverters are not connected to each other but also you don't see it in the menu of 3500.  but I see the modbus values at least.

MarcovBarneveld, 2023-09-12:
I gave the SE3000H (xx.61) an other IP in Homey and executed the command on the SE3500H again.
The SE3000H does not react as far as I can see, nor does the SE3500H
f03b88f4-c4d3-4de8-9b4e-22c576d797d7
Hope you see what goes wrong.

I do'nt know if I can ask this in this ticket also, but can you make the voltage of the inverter vissible in de Homey app?

biemond, 2023-09-12:
clearly it is doing the wrong ip.  are you certain you do the right device  on your flow? the 3500 device and not the 3000 1.

MarcovBarneveld, 2023-09-13:
[image]
[image]

MarcovBarneveld, 2023-09-13:
[image]
[image]

MarcovBarneveld, 2023-09-13:
I gave the two inverters an other IP adress.
You can beleave me or not, but ...-0C is the SE3000 and the ...-9C is the SE3500.

[image]
[image]

MarcovBarneveld, 2023-09-13:
[image]
[image]
[image]

MarcovBarneveld, 2023-09-13:
Executed the command on the 3500H and the 3000H drops. (hoe dan?!)
ec8f270f-a32f-465e-8765-2ee612c87a5f
I can't see any problem on my side, or I'm greatly mistaken.
Perhaps this data helps you to pinpoint something

biemond, 2023-09-13:
yeah super weird , because you get the right data on the right homey device and use the same settings for the update action. 
Plus you are saying inverters are in-depended of each other. 

let me check again

MarcovBarneveld, 2023-09-13:
Only I have an other problem now!
I can't get it back to 100%! the way I tested it before is not working anymore!!!
This is not funny :-(

biemond, 2023-09-13:
from the solaredge menu you should be able to . 
from homey when you do this , can you give me a diagnostic report.

MarcovBarneveld, 2023-09-13:
I know what I did.
By way of testing I added a new SE inverter, with dummy IP to see if i missed something on installation. The SE3000 was the last one to install and it picks alle the commands. Is there any connection in this.
Now I'm going to remove and add the SE3000 to see what happens.

biemond, 2023-09-13:
at least it looks like last device added is used for the updates.

biemond, 2023-09-13:
I don't have an 2nd inverter but can you setup some vpn so I can check it from my laptop.

biemond, 2023-09-13:
at least let me add more logs , which device and ip is doing the action.

MarcovBarneveld, 2023-09-13:
Somhow I don't see the power limit in setapp webinterface? I resetted the power settings, but no change.
You perhaps know where to find it in setapp web?

Because I could not reset it that way and I mus do other things to quick resolve it I removed the device from homey and added it again. So it indeed seems the last added picks up the commands
I executed the command and it worked 👍 
But now I lost my day total :-( But at least I got my power back!!!

biemond, 2023-09-13:
yep, now adding more logs. I think the wrong device get the update event.

biemond, 2023-09-13:
so added a double device and only see 1 invocation. only getting ip can be an issue.  

added more logs , can you do this 1 https://homey.app/a/solaredge.modbus/test/ 
and send me a report again.

MarcovBarneveld, 2023-09-14:
OK, I keek an eye on a new beta.

MarcovBarneveld, 2023-09-14:
I see you have also added voltage!! Great!!!!!

biemond, 2023-09-14:
yep, I think I need to filter on device 
```
2023-09-14T18:40:42.657Z [log] [ManagerDrivers] [Driver:invertor] [Device:15cd6f4f-3648-4ef8-a79d-7167f77f4be5] device name id j7pt4
2023-09-14T18:40:42.660Z [log] [ManagerDrivers] [Driver:invertor] [Device:15cd6f4f-3648-4ef8-a79d-7167f77f4be5] device name SE3000H
2023-09-14T18:40:42.661Z [log] [ManagerDrivers] [Driver:invertor] [Device:15cd6f4f-3648-4ef8-a79d-7167f77f4be5] SE3500H
2023-09-14T18:40:42.662Z [log] [ManagerDrivers] [Driver:invertor] [Device:15cd6f4f-3648-4ef8-a79d-7167f77f4be5] device name id j7pt4
2023-09-14T18:40:42.662Z [log] [ManagerDrivers] [Driver:invertor] [Device:15cd6f4f-3648-4ef8-a79d-7167f77f4be5] device name SE3000H
{
  host: '192.168.1.65',
  port: 1502,
  unitId: 1,
  timeout: 15,
  autoReconnect: false,
  logLabel: 'solaredge Inverter',
  logLevel: 'error',
  logEnabled: true
}
Connected ...
activepowerlimit {
  metrics: UserRequestMetrics {
    createdAt: 2023-09-14T18:40:42.678Z,
    startedAt: 2023-09-14T18:40:42.678Z,
    receivedAt: 2023-09-14T18:40:42.705Z
  },
  request: ModbusTCPRequest {
    _id: 1,
    _protocol: 0,
    _length: 6,
    _unitId: 1,
    _body: WriteSingleRegisterRequestBody {
      _fc: 6,
      _address: 61441,
      _value: 100
    }
  },
  response: ModbusTCPResponse {
    _id: 1,
    _protocol: 0,
    _bodyLength: 6,
    _unitId: 1,
    _body: WriteSingleRegisterResponseBody {
      _fc: 6,
      _address: 61441,
      _value: 100
    }
  }
}
disconnect
```

MarcovBarneveld, 2023-09-15:
OK. Let me know when you have a new (bèta) driver availiable?

biemond, 2023-09-15:
done,  was thinking wrongly , action is global and need to use device settings which in the arguments 

https://homey.app/a/solaredge.modbus/test/

MarcovBarneveld, 2023-09-16:
YES!!! It is working!!! THNX!
When is the new driver going live?

biemond, 2023-09-16:
cool, same version will go live as release , not much  difference.