biemond/solaredge.modbus #57: charge battery from the grid

kicksev, 2023-10-22:
Hi, I have a Solaredge 3f 10,5K system with a battery and I am using the SolarEdge + Growatt TCP modbus app in my Homey Pro. I am looking for a way to charge the battery from the grid. When setting Storage Control to "remote control"and Storage charge/discharge mode to "charge from PV+AC", the house is charged from the grid and the battery is only charged from the PV. As the PV power becomes 0, the house is still charged from the grid and the battery is discharging to the house with a tiny 0,05kW. The battery never gets charged from the grid. What modes should I set to make the battery being charged from the grid?

biemond, 2023-10-22:
works for me 

[image]
[image]
[image]

biemond, 2023-10-22:
maybe ask for a firmware update

CaptainVoni, 2023-11-11:
I have the same situation here. Grid charging is simply not enabled in some regions it seems. I don't even have the option to enable grid charging in the SetApp.

@biemond What inverter and what firmware do you have exactly?

kicksev, 2023-11-11:
A few days ago out of the blue all functions started to run. I assume there is new software loaded into my inverter from remote or some magic happened. With Storage Contol status on “Remote Control” changing the Storage Charge/discharge Mode is almost immediately reflected in the inverter, functions as expected and is visible in the Mysolaredge app.

biemond, 2023-11-11:
indeed, but at same point in the beginning the inverter wants to do test and after that it was allowing remote mode.

https://github.com/biemond/solaredge.modbus/blob/main/drivers/invertorwithbatt/device.ts#L459 plus I set it in remote mode 1st.

I think solaredge enabled it for you. mostly in some night batch after you try to do this.  Don't expect your installaltion guy is doing things proactive.

CaptainVoni, 2023-11-12:
@kicksev  Could you tell me what firmware version you have on your inverter? Do you have the SetApp available?

kicksev, 2023-11-13:
I don't have Setapp, but use the inverters wifi to browse to 172.16.0.1.
Firmware versions of my inverter:
CPU  4.17.136
DSP1  1.20.1551
Dsp2  2.20.1104

CaptainVoni, 2023-11-13:
I got 4.17.136/1.20.1532/2.20.1103.

Can you check what you see under Commissioning, Power Control (the lightning icon), Energy Manager, Storage Control? I only have one entry there: AC Charge Limit. According to [this video](https://youtu.be/w2ap784QGO8?feature=shared&t=32), there should be another option "Grid Charge" that I simply don't have.

May I ask what country you are from?

biemond, 2023-11-13:
menu is restricted . also for me. modbus allows me more. also if you want to see more you need an installation accounts.

CaptainVoni, 2023-11-13:
Why don't you register such an account? I have two ;-)

kicksev, 2023-11-13:
I am from Holland. My answer to the Commissioning question has to wait for a replacement of the communication board in the inverter. the system stopped communicating a few days ago, even the inverter wifi is not working.

CaptainVoni, 2023-11-13:
@kicksev  Thanks for the info. I would be really interested to see what settings you have there. I'm pretty sure, that my inverter was able to charge from the grid up at the very beginning, but then it was disabled. I really wonder why and suspect some issue with (country-specific) regulation.

@biemond  You yourself might have given the answer to your own question ;-)

biemond, 2023-11-13:
yep https://www.solar-bouwmarkt.nl/budget-omvormers-solaredge-monitoring-portal-in.html

CaptainVoni, 2023-11-21:
I just got information from Solaredge directly: it is right now impossible to charge from the grid, it's not implemented yet. 😆 

I'm not sure they know how their products actually work...

CaptainVoni, 2023-12-01:
> I am from Holland. My answer to the Commissioning question has to wait for a replacement of the communication board in the inverter. the system stopped communicating a few days ago, even the inverter wifi is not working.

Just checking: is your inverter still without communication? I'm looking for some "proof" that the Solaredge inverter actually can charge from the grid for my installer. So that he at least tries to find a way to enable it. :grin:

kicksev, 2023-12-02:
@CaptainVoni My system is running again and I have looked at the Storage Control. Also in my system there is only 1 entry : AC charge limit which is set to no limit. To start charging and uncharging to AC I had to set the import and export limits to 5000 again,  (were reset to 0)

CaptainVoni, 2023-12-02:
@kicksev Thanks for the hints. I tried pretty much everything, but I can't get it to charge. What exactly do you do to set the import and export limits? (Just to be safe we are talking about the same settings) Can you show me a flow where you set all the necessary setting to get it charging?

kicksev, 2023-12-02:
@CaptainVoni  what country you are from? 
I have 2 48V batteries with a maximum charge/discharge load of 5000W

[image]

To check the inverter and battery status: 
[image]

Inverter status:

0.	Undefined
1.	Off
2.	Sleeping
3.	Grid Monitoring
4.	Producing
5.	Producing (Throttled)
6.	Shutting Down'Fault
7.	Maintenance

Battery Status:

0.	Off
1.	Standby
2.	Initialising
3.	Charging
4.	Discharging
5.	Fault
6.	Idle
7.	Idle
8.	Idle
10.	Idle

kicksev, 2023-12-02:
I will try to make a dashboard screen dump when my battery is loading from the grid at 5000W for your installer Too bad, at this moment the system is not listening to my commands, It does not react at all, may be I have played too much today.

CaptainVoni, 2023-12-03:
That would be great, thanks.

As I wrote, right at the beginning I was able to charge from the grid. But that was when I installed and played around with this app and I think that playing-around triggered something to prevent future charging.

kicksev, 2023-12-03:
@Biemond
On Oct 22 you showed a picture like this with charging 1000w and discharging 5000w, values my inverter is now also showing:
[image]
You also showed a picture where charging and discharging settings were 5400W. How did you change these values? I tried setting charge power and discharge power with no success

kicksev, 2023-12-06:
@CaptainVoni  I Think I start to understand the system a bit. My latest problem was caused by a minimum import limit set to 3000W. When set to zero, battery discharging started again. This limit seems to be the minimum power imported from the grid before the battery will discharge. As far as I can see are all modes of storage charge / discharge mode seem to work. This is a screendump when the battery is loaded with 4000W from AC according to the charge settings. 
[image]

CaptainVoni, 2023-12-07:
@kicksev Thanks for all the info! I can't tell you how frustrating this is for me. No matter what I try, it doesn't charge from the grid. But now I have a screenshot from the Solaredge app I can pass along. Thanks again, I'll report back if I find any solution.

kicksev, 2024-01-05:
@CaptainVoni Now testing the new TOU option in the Solaredge app. According to Solaredge: “_In Time of Use mode, we use your rate plan and smart forecasting and learning algorithm which optimizes your energy sources and battery usage to reduce your electric bills.
Based on your consumption habits and energy production, our algorithm predicts your consumption and energy generation potential for the following day, and dynamically charges or discharges the battery accordingly.
Time of Use mode automatically uses your battery when electricity rates are at their peak and solar is not available_.”
 In fact, at lowest hour price my battery is charged from the grid at full speed (5kW) but then it is discharged at random times. I am trying to find out what smart algorithms this behavior is based upon. But in your case, maybe the system will be forced to charge at 5kW.

biemond, 2024-01-05:
oh, only was thinking they now support auto disabling solar export when hour price is negative 

is this a new option in your inverter menu?

kicksev, 2024-01-05:
it in the Mysolaredge app. info: https://marketing.solaredge.com/tou-mode 
[image]

kicksev, 2024-01-05:
First week TOU. Charging starts at minimum hour price, discharging starts random? It does not look like the result of smart algorithms as promised.
[TOU test.pdf](https://github.com/biemond/solaredge.modbus/files/13845636/TOU.test.pdf)

biemond, 2024-01-05:
interesting so they do this from their datacenter and gather all the data.  Did you see this menu option on your inverter? 
also like to see your change triggers, I guess they do the same trigger actions

kicksev, 2024-01-05:
I have asked them where the algorithms run, but they told me to consult my installer who probably does not even know of the option. I saw the option on my mobile not on the inverter menu. What do you mean with change triggers?

kicksev, 2024-01-05:
I have asked them where the algorithms run, but they told me to consult my installer who probably does not even know of the option. I do not understand your questions

biemond, 2024-01-05:
oh, now I know this time of use.  I did this 2 years back . only NL as user you were not allowed to make these profiles, my installer did this for me and there were some profiles like day and night tariff

looks like they make it more advanced

biemond, 2024-01-05:
[storedge_charge_discharge_profile_programming.pdf](https://github.com/biemond/solaredge.modbus/files/13845690/storedge_charge_discharge_profile_programming.pdf)

biemond, 2024-01-05:
you need to click on the batt  and I see the same as the pdf but now from the app and I need to set all my peak moments. 
you can also do utility rate plan but I don't have this.so I got peak hours only

kicksev, 2024-01-05:
In what country you are? Have you selected your tariff in the app?: 
[image]

biemond, 2024-01-05:
like you in NL but did not select dynamic, still do salderen.

CaptainVoni, 2024-01-07:
@kicksev I tried the TOU option - didn't do anything. Just for fun I tried to charge from the grid:
[image]

I didn't change anything. No update for the inverter. No idea - just works for now.

kicksev, 2024-01-07:
The system is charging with 5kW. This should only take place at lowest rate hour. This screenshot should be taken at about 4 
 o'clock this morning, did you?  Please let me know if discharging will start at the peak hours (if you entered some). 

When I set the Storage charge/ discharge to Charge from PV+AC, it sometimes charges from PV+AC with 5kW and sometimes only with the available PV. Is there be a variable prohibiting/allowing charging from the grid besides the max discharge setting?

kicksev, 2024-01-09:
in TOU mode last week my system charged from the grid at 5kW. This week it does charge from the grid at all, not by TOU mode nor when manually set to charge from PV+AC.  I wonder why. There is a register (Storage AC Charge Policy) enabling or disabling charging from the grid. do you know a way to read this register value or even set it?

Storage AC Charge Policy is used to enable charging for AC and the limit of yearly AC charge (if applicable).
0 – Disable
1 – Always allowed – needed for AC coupling operation. Allows unlimited charging from the AC. When used with Maximize
self-consumption, only excess power is used for charging (charging from the grid is not allowed).
2 – Fixed Energy Limit – allows AC charging with a fixed yearly (Jan 1 to Dec 31) limit (needed for meeting ITC regulation in
the US)
3 – Percent of Production - allows AC charging with a % of system production year to date limit (needed for meeting ITC
regulation in the US)

CaptainVoni, 2024-01-10:
@kicksev This sounds like you found the reason why it sometimes charges and sometimes not! 🎉 Do you have a link to the document where you found that register mentioned? I only found it mentioned [here](https://github.com/evcc-io/evcc/discussions/6788) and it's stated that it is read-write, so we should be able to set it.

@biemond Can you add that register to the app? For the options 2 and 3 the following register (Storage AC Charge Limit) would be needed,  but I doubt that there is much use outside the US for those options.

CaptainVoni, 2024-01-10:
@biemond The register "Storage Backup Reserved Setting" might also be worth considering. It seems to relate to the minimum SOC of the battery.

biemond, 2024-01-10:
let me check the docs which register this is plus what is my value for this.

biemond, 2024-01-10:
can you send me diagnostic reports, I already added this 

```
        "storage_control_mode": [0xe004, 1, 'UINT16', "Storage Control Mode"],
        "storage_accharge_policy": [0xe005, 1, 'UINT16', "Storage AC Charge Policy"],
        "storage_accharge_Limit": [0xe006, 2, 'FLOAT32', "Storage AC Charge Limit"],
```

CaptainVoni, 2024-01-10:
07d74748-6c83-4ab2-b214-10d109151f0f

biemond, 2024-01-10:
```
storage_control_mode 4 xxx Storage Control Mode
storage_accharge_policy 2 xxx Storage AC Charge Policy
storage_accharge_Limit 0 xxx Storage AC Charge Limit
```

CaptainVoni, 2024-01-10:
With what version did you add that?

biemond, 2024-01-10:
what do you mean ? I just already read this 

2 – Fixed Energy Limit – allows AC charging with a fixed yearly (Jan 1 to Dec 31) limit (needed for meeting ITC regulation in
the US)

biemond, 2024-01-10:
maybe in the new year you have some new quota.

biemond, 2024-01-10:
it was always there , never showed this on the device.

CaptainVoni, 2024-01-10:
> maybe in the new year you have some new quota.

That was also my first thought. But at least we might have an idea right now what’s going on.

biemond, 2024-01-10:
will make a device action so you can change it.

biemond, 2024-01-10:
nop , I never touch or set this. just reading. there is no code which can do this on the app.

biemond, 2024-01-10:
I got these 
```
storage_control_mode 1 xxx Storage Control Mode
storage_accharge_policy 1 xxx Storage AC Charge Policy
storage_accharge_Limit 0 xxx Storage AC Charge Limit
remote_control_command_mode -1 xxx Remote Control Command Mode
remote_control_charge_limit 5000 xxx Remote Control Charge Limit
remote_control_command_discharge_limit 5000 xxx Remote Control Command Discharge Limit
remote_control_command_timeout 3600 xxx Remote Control Command Timeout
remote_control_default_command_mode 0 xxx Storage Charge/Discharge Default Mode
```

biemond, 2024-01-10:
I think a hour is the default for reset. 

I do more 

```
      if (type == 'storagedefaultmode') {
        const storagedefaultmodeRes = await client.writeSingleRegister(0xe004, Number(4));
        console.log('controlmodewrite', storagedefaultmodeRes)

        // set timeout to 6 hours, done in seconds
        const remoteTimeout = await client.writeMultipleRegisters(0xe00b, [ Number(21600), 0 ]);
        console.log('remote_control_command_timeout', remoteTimeout);
```

kicksev, 2024-01-11:
@CaptainVoni @biemond I may be too late with this link to the document you asked for: https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/ 
I see Biemond wrote some code to set some registers. Although I have programmed a lot of my life, I'm not familiar with this coding, so could you tell me what is done and if this can help me to charge from AC?

CaptainVoni, 2024-01-11:
@kicksev If I’m not mistaken, @biemond will update the app in the next time, so that it will be possible to change the policy setting.

Patience, young padawan! 😉

biemond, 2024-01-14:
I think we don't care  about Storage AC Charge Limit  if we can set this Storage AC Charge Policy   

Storage AC Charge Limit is used to set the AC charge limit according to the policy set in the previous register. Either fixed in kWh or percentage is set (e.g. 100KWh or 70%). Relevant only for Storage AC Charge Policy = 2 or 3.

biemond, 2024-01-14:
Can you try the menu action, it works for me https://homey.app/a/solaredge.modbus/test/
[image]
[image]

CaptainVoni, 2024-01-14:
@biemond IT WORKS! 🎉 Thank you so much for solving this!
@kicksev Also a big thank you to you for identifying the problem!

biemond, 2024-01-14:
cool , in a few days when I don't hear any issues , I will publish it to release..

how did you like my German?  bit rusty on that but google helped me a bit.

biemond, 2024-01-14:
By the way is device action enough or do you want a flow action?

[17 more replies]