biemond/solaredge.modbus #59: Gegevens uit mijn Solaredge halen

gelde1957, 2023-11-27:
Heb sinds kort een Homey pro, en wil graag mijn Solaredge er op connecten.

op dit moment werkt mijn Solaredge prima en is uit te lezen met o.a. homewizard.
Heb de app geïnstalleerd voor de Homey pro en apparaat toegevoegd met IP adres, poort 1502 en divice ID 1. Ik kan hem nu openen in apparaten van Homey echter er komt geen dat binnen😩.

heb je een idee wat en hoe ik het kan oplossen want bij de homewizard komt wel info binnen.

biemond, 2023-11-27:
Hey, heb je modbus wel aangezet in de inverter? Zie 1ste forum post of home repository readme.

Home wizard gebruikt internet api , dit is rechtstreeks en elke 25 seconden

gelde1957, 2023-11-28:
Beste Edwin,Het is gelukt om TCP enable te zetten in de omvormer👍Echter hij geeft nog steeds geen data door aan de app😳.Moet ik nog iets anders instellen of resetten?.Groet,Antonio

gelde1957, 2023-11-28:
Edwin,Nog even een nabrander, ik heb ook via discovery gekeken maar daar staat de Solaredge ook niet tussen. Groet,Antonio

biemond, 2023-11-29:
Ok, misschien is je ip adres anders? Of poort nummer klopt niet

Heb ook wel vaak meegemaakt dat de firmware van de inverter achter loopt

gelde1957, 2023-11-29:
Sorry maar heb alles gecontroleerd, IP adres klopt, TCP modus staat enable, port klopt 1502. Firmware is gisteren geüpdatet. Kan het zijn dat inverter ID anders moet zijn dan 1? En hoe zou ik dat kunnen uitzoeken?Groet,Antonio

gelde1957, 2023-11-29:
> Apart, heb je toevallig nog een programma geïnstalleerd die deze uitleest en homey blokkeert
> Anders een poort scan doen op de poort nummer , wel eerst de homey device verwijderen.
> Als hij niet verschijnt dan is er iets mis. Zo ja dan kan je mij dan een diagnostic report sturen

Beste Edwin,Bedankt voor je snelle reactie, ik ga straks wel kijken naar die poortnummer, moet eerst uitzoeken hoe ik de poortadressen kan uitlezen maar zal wel lukken.Laat het je nog weten.Groet,Antonio

biemond, 2023-11-30:
on mac doe ik dit
https://www.cyberciti.biz/faq/linux-port-scanning/
```
edwinbiemond@Edwins-MacBook-Pro ~ % nc -z -v 192.168.107.25 1502
Connection to 192.168.107.25 port 1502 [tcp/shivadiscovery] succeeded!
edwinbiemond@Edwins-MacBook-Pro ~ %
```