biemond/solaredge.modbus #60: Storede/ accu en enery meter SolarEdge uitlezen

Vdhham, 2023-11-29:
He, ik heb een SolarEdge storedge met accu’s en een smart meter van SolarEdge in gebruik en de sturing werkt allemaal. Alleen de import en export gegevens uit de smart meter van SolarEdge krijg ik niet op homey binnen. Deze is wel verbonden via het draadloze home energy netwerk en niet via RS485, zou dat de oorzaak kunnen zijn dat de gegevens niet doorkomen na homey?

alvast bedankt

biemond, 2023-11-30:
Hey,

ik zag je rapport  en ik zie alleen phase 1 meter data , niet de andere 2. ik heb de andere 2 uitgezet vanwege timeouts.

```
meter1-c_deviceaddress -32768 xxx Modbus ID
meter1-c_sunspec_did -1 xxx SunSpec DID
meter1-current -1 -1 Current
meter1-l1_current 2260 xxx L1 Current
meter1-l2_current -1 xxx L2 Current
meter1-l3_current -1 xxx L3 Current
```
je bent de eerste die ik zie met deze smart meter van SolarEdge. misschien heb je 3 aparte meters ipv 1.

zie wel 

```
meter1-export_energy_active 4294967295 0 Total Exported Energy (Active)
meter1-import_energy_active 65535 xxx Total Imported Energy (Active)
```

biemond, 2023-11-30:
heb je toevallig een vpn waar ik je inverter kan uitlezen , fixen en de timeouts kan tunen.

Vdhham, 2023-11-30:
Nee heb helaas zo geen vpn thuis, ben daar dan ook net weer niet in thuis🙈

Vdhham, 2023-11-30:
Het is wel gewoon een 3 fase meter van SolarEdge zelf. Echter kun je deze via rs458 koppelen met de omvormer maar ook via hun nieuwe draadloze home energy protocol. Dat is nu het geval, maar wil wel is proberen als ik hem bekabel of het anders wordt.

wel heel apart dat die het als 1fase uitleest

Vdhham, 2023-11-30:
Ja heel apart, via de API app werkte die wel gewoon. Het is fysiek 100% 1 meter, originele SolarEdge inline meter. Alles in eigen beheer gemaakt dus die kant van het verhaal weet en ken ik. Het is alleen de vraag of die meter anders reageert dan, ik zal hem eens bekabelen en laat je dan is horen wat die doet. Maakt het daarvoor niet uit dat je die andere nu hebt uitgezet? Niet dat ik wat ga proberen wat dan geen effect gaat hebben.

Vdhham, 2023-11-30:
Blijft ook wel apart dat ik de eerste ben met dit meter systeem en fout, want bij de hybride met accu systemen van SolarEdge ben je deze meters ook nodig wil je de accu’s laten werken. Ga de kabel via rs485 eens proberen🙈

Vdhham, 2023-12-01:
Heb je als het goed is.