biemond/solaredge.modbus #65: Growatt MOD 6000 TL3-XH met bijbehorend batterijpakket 10 kW

fkoole, 2024-01-14:
Ook ik ben een beginnend gebruiker van Homy Pro. Veel van mijn apparatuur heb ik inmiddels kunnen toevoegen, maar mijn Growatt batterijpakket krijg ik helaas nog niet toegevoegd.
Volgens mij heb ik er wel alles aan gedaan, maar volgens de app wordt dit type nog niet ondersteund.

Ik heb een Growatt MOD 6000 TL3-XH met bijbehorende batterij (ARK, met in totaal 10.2 kW)
Ik heb een MODBUS TCP/RTU aangesloten en geconfigureerd. 
Tijdens de installatie van de app heb ik het IP-adres ingevoerd, maar nu krijg ik een melding dat dit type omvormer niet wordt ondersteund.
Uiteraard krijg ik geen data, maar alleen een rode driehoek met uitroepteken.

Is dit op te lossen, of doe ik zelf toch iets verkeerd?
Ik ben nog niet echt bekend met deze materie, maar zou het wel graag werkend willen hebben.

Ik installeer zelf zonnepanelen en batterijen en heb nu een batterijpakket staan welke nog niet helemaal slim kan laden. Er zit op deze batterij namelijk geen enkel paneel aangesloten. Hij wordt geladen vanuit het stroomnet en dat wil ik laten sturen. Ja, ik heb zelf een hele hoop zonnepanelen, maar die hebben allemaal micro omvormers van APSystems. 
Ook zoek ik hiervoor een app uiteraard, maar die vraag zal ik elders moeten stellen denk ik zo...

Met vriendelijke groet,

Ferdy Koole

mameijer, 2024-01-22:
Ik heb hier ook mee lopen stoeien maar het werkt nu enigszins. Heb je de TCP/RTU device op poort 502 staan? Nadat ik deze op poort 502 zette begon de RS485 te werken. Op de omvormer (Growatt SPA 10000 T3 UP) heb ik de mobdus/tcp op 485-3
aangesloten. Verder had ik veel aan de documentatie waaraan gerefereerd wordt in de Homey app:
https://homey.app/nl-nl/app/solaredge.modbus/SolarEdge-+-Growatt-TCP-modbus/ 
https://www.dropbox.com/s/584915enkxc508u/Setup%20Growatt.pdf?dl=0 

Vr Gr
Mark

biemond, 2024-04-01:
ja , ik heb zelf helaas geen 1 dus je kan het beste je vroeg op het forum stellen en mensen hierover taggen.