biemond/solaredge.modbus #69: No longer data from solaredge

whitefarmer1, 2024-03-25:
Hi Edwin,

Thanks for creating the app, unfortunately it no longer sends data to my homey. (6 days now).

I have restarted the app, but with no succes.

As far as i know, no changes have been made.

(Greetings John  Mierlo the Netherlands)

biemond, 2024-04-01:
hey , can you send me a report , probably your ip changed.

whitefarmer1, 2024-04-02:
Hoi Edwin,
Onderstaand de code:

e97e646f-3182-42e4-9bbd-a9530803498f

Laat me weten als je meer info nodig hebt.

Dank voor je App!

Laat me weten waar ik wat kan doneren aub.

Groet John

biemond, 2024-04-02:
yep 

```
[Driver:invertor] [Device:16033500-61bf-499f-8551-03c3d59090b2] 192.168.2.2
Error: connect EHOSTUNREACH [192.168.2.2:1502](http://192.168.2.2:1502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1157:16) {
  errno: -113,
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.2.2',
  port: 1502
}
Client closed
```

whitefarmer1, 2024-04-03:
DANK!

luccie007, 2024-04-04:
Just hijacking this thread as its IP related.

I use EVCC as a modbus relay due to solaredge accepting only 1 connection
sometimes EVCC goes offline and i lose solar data.
Is there a way to make the IPadres and Port available to set in a flowcard? So when i detect this i can route the ip directly (as EVCC then has no connection there is only homey that connects)

biemond, 2024-04-04:
I know there is some modbus proxy / docker is that not more stable.  

you can setup 2 devices , 1 on your evcc and 1 directly ( this will fail or work, depends on the 1st ). I retry it every 30 seconds.

there is also modbus app so checking some value  but I think you want to change the settings of a device. maybe there is homey api for this.

whitefarmer1, 2024-04-04:
Please take me out of the conversation. Mine works (wrong ip)

Thanks
John