biemond/solaredge.modbus #70: It would be great if this app also supported Solax inverters + battery

JorgenSeemann, 2024-03-27:
Note Solax supports both own Lithium battery as well as Lead Acid batteries :)

JorgenSeemann, 2024-03-27:
The latest document about I can find about Solax MODBUS protocol is located here: https://gbc-solino.cz/wp-content/uploads/2022/07/Hybrid-X1X3-G4-ModbusTCPRTU-V3.21-English_0622-public-version.pdf

biemond, 2024-04-01:
nice, looks doable. 

do you provide vpn access else it will be hard for me to test it.

JorgenSeemann, 2024-04-01:
Hi Edwin,
What do you need access to - Homey, Solax inverter or ?
Easy (if useful) would be to provide create an admin account on my homey.
That would allow you to play around (but I'm unsure if it's useful)?
Currently I'm using the "Solax Inverter", but I like the joined app better
(using it for my Huawei :))
Cheers
Jørgen

biemond, 2024-04-01:
hey,  only solax is ok. 

do you also have a Huawei?  like to try to do write access part.

JorgenSeemann, 2024-04-02:
Hi Edwin.
Not sure my system supports VPN, but will have a look tonight.
Yes. Have both Solax and Huawei (w. Battery)
Cheers
Jørgen

biemond, 2024-04-07:
I got this , but having a hard time to get your batt soc.
what can be use for this?

```

edwinbiemond@Edwins-MBP solaredge.modbus % node test9.js
-------------------
Connected ...
SolarChargerUseMode: 0
output power limits precent 0~100: 100
Manual mode: 0
Grid connected pull in power point: 2000
GridVoltage: 2366
GridCurrent: 91
GridPower: 6479
radiator temperature: 43
Powerdc1: 5977
Powerdc2: 5996
Batpower_Charge1: 5127
TemperatureBat: 0
Battery Capacity: 0
OutputEnergy_Charge_today: 26
InputEnergy_Charge_today: 83
feedin_energy_total: 1256587264
consum_energy_total: 2529492992
GridVoltage_R(X3): 2366
GridCurrent_R(X3): 91
GridPower_R(X3): 2140
GridVoltage_S(X3): 2386
GridCurrent_S(X3): 92
GridPower_S(X3): 2195
GridVoltage_T(X3): 2385
GridCurrent_T(X3): 90
GridPower_T(X3): 2143
FeedinPower_Rphase(X3): 132644864
FeedinPower_Tphase(X3): 132186112
EchargeToday: 0
EchargeTotal: 8978432
SolarEnergyTotal: 699990016
SolarEnergyToday: 107
energy to the grid: 3538944
energy from the grid: 4
BMS_UserSOC: 0
BMS_UserSOH: 0
ActivePowerTarget: 0
Charge_Discharg_Power: 0
SocUpper: 100
SocLower: 10
```

biemond, 2024-04-07:
```

registers = {
        "SolarChargerUseMode":       [0x008B, 1, 'UINT16', "SolarChargerUseMode"],
        "PowerLimitsPercent":        [0x0025, 1, 'UINT16', "output power limits precent 0~100"],
        "ManualMode":                [0x008C, 1, 'UINT16', "Manual mode"],
        "FeedinOnPower":             [0x0123, 1, 'UINT16', "Grid connected pull in power point"],

    }    

    inputRegisters = {

        "GridVoltage":              [0x0000, 1, 'UINT16', "GridVoltage"],
        "GridCurrent":              [0x0001, 1, 'INT16', "GridCurrent"],
        "GridPower":                [0x0002, 1, 'INT16', "GridPower"],

        // 0x0007 GridFrequency(X1) R GridFrequency 0.01Hz uint16 1

        "Temperature":             [0x0008, 1, 'INT16', "radiator temperature"],
        "Powerdc1":                [0x000A, 1, 'UINT16', "Powerdc1"],
        "Powerdc2":                [0x000B, 1, 'UINT16', "Powerdc2"],
        // 0x0016 Batpower_Charge1 R Batpower_Charge1 1W int16 1
        "Batpower_Charge1":        [0x0016, 1, 'INT16', "Batpower_Charge1"],
        // 0x0018 TemperatureBat R TemperatureBat 1℃ int16
        "TemperatureBat":          [0x0018, 1, 'INT16', "TemperatureBat"],
        // 0x001C Battery Capacity R Battery capacity 1% uint16 1
        "BatteryCapacity":         [0x001C, 1, 'UINT16', "Battery Capacity"],
        // 0x0020 OutputEnergy_Charge_today R OutputEnergy_Charge_today 0.1kWh uint16 1
        "OutputEnergy_Charge_today": [0x0020, 1, 'UINT16', "OutputEnergy_Charge_today"],
        // 0x0023 InputEnergy_Charge_today R InputEnergy_Charge_today 0.1kWh uint16 1
        "InputEnergy_Charge_today":  [0x0023, 1, 'UINT16', "InputEnergy_Charge_today"],

        // 0x0048
        // feedin_energy_total(meter) R
        // energy to the grid
        // (0x48:LSB,0x49:MSB)
        // 0.01kWh uint32 2
        "feedin_energy_total":  [0x0048, 2, 'UINT32', "feedin_energy_total"],
        // 0x004A
        // consum_energy_total(meter) R
        // energy form the grid
        // (0x4A:LSB,0x4B:MSB)
        // 0.01kWh uint32 2
        "consum_energy_total":  [0x004A, 2, 'UINT32', "consum_energy_total"],

        // 0x006A GridVoltage_R(X3) R GridVoltage_R 0.1V uint16 1
        "GridVoltage_R(X3)":        [0x006A, 1, 'UINT16', "GridVoltage_R(X3)"],
        // 0x006B GridCurrent_R(X3) R GridCurrent_R 0.1A int16 1
        "GridCurrent_R(X3)":        [0x006B, 1, 'INT16', "GridCurrent_R(X3)"],
        // 0x006C GridPower_R(X3) R GridPower_R 1W int16 1
        "GridPower_R(X3)":          [0x006C, 1, 'INT16', "GridPower_R(X3)"],
        // 0x006E GridVoltage_S(X3) R GridVoltage_S 0.1V uint16 1
        "GridVoltage_S(X3)":        [0x006E, 1, 'UINT16', "GridVoltage_S(X3)"],
        // 0x006F GridCurrent_S(X3) R GridCurrent_S 0.1A int16 1
        "GridCurrent_S(X3)":        [0x006F, 1, 'INT16', "GridCurrent_S(X3)"],
        // 0x0070 GridPower_S(X3) R GridPower_S 1W int16 1
        "GridPower_S(X3)":          [0x0070, 1, 'INT16', "GridPower_S(X3)"],
        // 0x0072 GridVoltage_T(X3) R GridVoltage_T 0.1V uint16 1
        "GridVoltage_T(X3)":        [0x0072, 1, 'UINT16', "GridVoltage_T(X3)"],
        // 0x0073 GridCurrent_T(X3) R GridCurrent_T 0.1A int16 1
        "GridCurrent_T(X3)":        [0x0073, 1, 'INT16', "GridCurrent_T(X3)"],
        // 0x0074 GridPower_T(X3) R GridPower_T 1W int16 1
        "GridPower_T(X3)":          [0x0074, 1, 'INT16', "GridPower_T(X3)"],

        // 0x0082
        // ~0x0083
        // FeedinPower_Rphase(X3)
        // R
        // FeedinPower_Rphase
        // (meter/CT)
[... 73 lines cut ...]
        // 0.01kWh uint16
        "consum_energy_today":  [0x009A, 1, 'UINT16', "energy from the grid"],
        // 2
        // 0x00BE BMS_UserSOC R BMS_UserSOC 1% Uint16 1
        "BMS_UserSOC":        [0x00BE, 1, 'UINT16', "BMS_UserSOC"],
        // 0x00BF BMS_UserSOH R BMS_UserSOH 1% Uint16 1
        "BMS_UserSOH":        [0x00BF, 1, 'UINT16', "BMS_UserSOH"],
        // 0x0102
        // ActivePowerTarget R ActivePowerTarget 1W int32 2
        "ActivePowerTarget":  [0x0102, 2, 'INT32', "ActivePowerTarget"],

        // 0x0114
        // Charge_Discharg_Power R
        // Charge_Discharg_Power
        // (0x114:LSB,0x115:MSB)
        // 1W int32 2
        "Charge_Discharg_Power":  [0x0114, 2, 'INT32', "Charge_Discharg_Power"],

        // 0x011C SocUpper R SocUpper 1% uint16 1
        "SocUpper":        [0x011C, 1, 'UINT16', "SocUpper"],
        // 0x011D SocLower R SocLower 1% uint16 1
        "SocLower":        [0x011D, 1, 'UINT16', "SocLower"],

    }    
```

JorgenSeemann, 2024-04-07:
Hi Edwin

You are right because currently I'm using the settings for Lead Acid
Battery. Hence you cannot get SOC/SOH.
There should be a battery voltage.

When you configure the battery for Lead Acid you basically enters min/max
voltage. It's defined as 4 different values due to the charge/discharge
parameters.

Cheers
Jørgen

biemond, 2024-04-07:
don't get the batt voltage.  all are 0. what register should I use for that.

JorgenSeemann, 2024-04-07:
Hi Edwin,

I think it would be very usefull to check
0x008D wBattery1_Type R 0：Lead Acid 1：Lithium 1 uint16 1

And if it 0 (lead acid) then use
0x008E Charge_floatVolt R Lead-acid battery charge_float voltage 0.1V
uint16 1
0x008F Battery_DischargeCutVoltage R Lead-acid battery discharge cut-off
voltage 0.1V uint16 1
0x0090 Battery_ChargeMaxCurrent R Lead-acid battery charge maximum current
0.1A uint16 1
0x0091 Battery_DischargeMaxCurrent R Lead-acid battery discharge maximum
Current 0.1A uint16 1
0x0092 absorpt_voltage R Lead-acid battery absorpt_voltage 0.1V uint16 1

I'm not a 100% sure but I imagine the current voltage would be the 0x0092
absorpt_voltage
It could also be
0x0014 BatVoltage_Charge1 R BatVoltage_Charge1 0.1V int16 1
0x0015 BatCurrent_Charge1 R BatCurrent_Charge1 0.1A int16 1
0x0016 Batpower_Charge1 R Batpower_Charge1 1W int16 1

To compare if you read values, them my setup is:
Charge equalization voltage = 735
Charge float voltage = 735
Discharge cut voltage 605
Discharge backup voltage = 605
Charge max current = 20
Discharge max current = 20

Currently my voltage is around 717 and dropping as I'm charging my car :)

Cheers
Jørgen

biemond, 2024-04-07:
```
SolarChargerUseMode: 0
output power limits precent 0~100: 100
Manual mode: 0
Grid connected pull in power point: 2000
Battery1 Type: 0
Charge_floatVolt: 7350
Battery_DischargeCutVoltage: 6050
Battery_ChargeMaxCurrent: 200
Battery_DischargeMaxCurrent: 200
absorpt_voltage: 7350
GridVoltage: 2376
GridCurrent: 11
GridPower: 366
radiator temperature: 46
Powerdc1: 2236
Powerdc2: 2628
BatVoltage_Charge1: 7275
BatCurrent_Charge1: 60
Batpower_Charge1: 4390
TemperatureBat: 0
Battery Capacity: 0
OutputEnergy_Charge_today: 94
InputEnergy_Charge_today: 129
Feedin power is obtained from Meter or CT: 0
energy to the grid: 1278214144
consum_energy_total: 2531590144
Today Energy (Inverter AC Port): 508
Total Energy (Inverter AC Port): 678821888
GridVoltage_R(X3): 2374
GridCurrent_R(X3): 11
GridPower_R(X3): 124
GridVoltage_S(X3): 2363
GridCurrent_S(X3): 13
GridPower_S(X3): 174
GridVoltage_T(X3): 2380
GridCurrent_T(X3): 10
GridPower_T(X3): 59
FeedinPower_Rphase(X3): 262144
FeedinPower_Tphase(X3): 65536
EchargeToday: 0
EchargeTotal: 8978432
SolarEnergyTotal: 729743360
SolarEnergyToday: 560
energy to the grid: 25165824
energy from the grid: 36
BMS_UserSOC: 0
BMS_UserSOH: 0
ActivePowerTarget: 0
Charge_Discharg_Power: 0
SocUpper: 100
SocLower: 10
Cell_Voltage_High: 0
Cell_Voltage_Low: 0
Off-gridVoltage(X1): 0
Off-gridCurrent(X1): 0
Off-gridPower(X1): 0
BusVolt: 0
```

biemond, 2024-04-07:
this indeed
```
BatVoltage_Charge1: 7275
BatCurrent_Charge1: 60
Batpower_Charge1: 4390
```

JorgenSeemann, 2024-04-07:
Pretty sure those 3 are correct with my app showing battery V is 728, and
charge W 4143 😀
Note its showing in 0.1V so 7275 would be 727.5V

biemond, 2024-04-07:
let me make a test version which you can try out.

JorgenSeemann, 2024-04-07:
Thanks, I'll be happy to check it out 😀

biemond, 2024-04-07:
can you check this 1 https://homey.app/a/solaredge.modbus/test/

JorgenSeemann, 2024-04-07:
I think the battery power, heatsink (inverter), and pv1+2 is correct.
Solar power looks more to be loaded power. Its for sure not equal to pv1+2.
Battery charge voltage and current is showing "-".
So some fine, some not 😀

biemond, 2024-04-07:
I think it is correct , it is also batt discharge

```
Inverter Power: 476
Powerdc1: 34
Powerdc2: 93
Batpower_Charge1: -358
```

biemond, 2024-04-07:
for discharge current is negative 

```
BatVoltage_Charge1: 7302
BatCurrent_Charge1: -4
Batpower_Charge1: -358
```

JorgenSeemann, 2024-04-07:
Let me show you how it looks

JorgenSeemann, 2024-04-07:
This is what I see in other app

biemond, 2024-04-07:
don't see it  but this is also what I see on our inverters.   probably in the app they do something. but the modbus ones looks correct with your data and what I know from other inverters.  

it is all about bookkeeping.   charge of batt is removed from solar power and with discharge it comes back.

JorgenSeemann, 2024-04-07:
The value shown as solar power is definitely load. I just checked by
putting the kettle on :)

biemond, 2024-04-07:
no no it is  including batt discharge

biemond, 2024-04-07:
looks like load because you don't need more than that

biemond, 2024-04-07:
tomorrow you will see  it is  pv minus  charging your battery.

biemond, 2024-04-07:
we can check tomorrow when the sun is up and maybe you can stop charging the battery . then we know for sure.

JorgenSeemann, 2024-04-07:
Ok. I'll check it again tomorrow. Sleep well :)

biemond, 2024-04-07:
yep , can add pvtotal based on pv1 and pv2 but curious if it is correct.  else need to calculate with pv and batt (dis)charge  . it need to be correct for the homey energy tab.

JorgenSeemann, 2024-04-08:
Good morning Edwin

We now have sun and the app is showing
Solar power = 496
Battery power = 5715
PV1 = 3151
PV2 = 3173
Battery charge current & voltage both just shows a "-"
Cheers
Jørgen

biemond, 2024-04-08:
now

I see this  7032 =   4154 + 4124 -  910. 
```
Inverter Power: 7032
radiator temperature: 44
Powerdc1: 4154
Powerdc2: 4124
BatVoltage_Charge1: 7349
BatCurrent_Charge1: 12
Batpower_Charge1: 910

```
2426  + 2289  + 2336
```
GridPower_R(X3): 2426
GridVoltage_S(X3): 2358
GridCurrent_S(X3): 97
GridPower_S(X3): 2289
GridVoltage_T(X3): 2381
GridCurrent_T(X3): 98
GridPower_T(X3): 2336

```

JorgenSeemann, 2024-04-08:
Sounds like the math is summing up correctly then. But strange I don't get
any numbers on Bat charge curr/vol
(I don't have bat power, nor any of the grid, but I guess this is just
limitations in the test version 😀)

biemond, 2024-04-08:
can you send me a diagnistic report for the batt vol /current. that should always display it.

biemond, 2024-04-08:
ah I see , forgot to add the data , will fix it tonight and add more.

JorgenSeemann, 2024-04-08:
Sounds good, thanks! 🙂

biemond, 2024-04-08:
can you try this https://homey.app/a/solaredge.modbus/test/

probably you need to remove and add the device again.

JorgenSeemann, 2024-04-08:
seems to be straight into working mode just doing the upgrade.
Now have solar+batt power. PV1+2=0 (of course - it black here now).
Battery charge current i 730A and battery charge voltage is 0A. I will
guess the parameters was swapped by accident.
the 3 meter power phase shows "-" so they are likely incorrect linked.
Besides of this - it's all looking great :)

biemond, 2024-04-08:
thanks indeed, fixed them  https://homey.app/a/solaredge.modbus/test/

for now just do an upgrade of the app.

biemond, 2024-04-08:
can you add a screenshot to the github thread.

JorgenSeemann, 2024-04-08:
[image]
[image]
[image]

JorgenSeemann, 2024-04-08:
with 3.0.101

JorgenSeemann, 2024-04-08:
Done :)

biemond, 2024-04-08:
yeah checking the big numbers. , can you add and remove again. looks like screenshots are from before.

JorgenSeemann, 2024-04-08:
Removed the device and added it again.
No it doesn't read any values at all
Claim all read last 54years ago
Homey rebooted as well but to no avail

biemond, 2024-04-08:
I think it was because both we reading it .  got this.

```
UserRequestError {
  err: 'OutOfSync',
  message: 'rejecting because of earlier OutOfSync error',
  response: undefined
}
```

biemond, 2024-04-08:
let me change the timing from 20 to 30 seconds

JorgenSeemann, 2024-04-09:
Good morning Edwin.
Upgraded to 3.0.102
Removed unit and added it again.
Same problem- all values are "-"
Cheers

JorgenSeemann, 2024-04-09:
> can you send me a report, else we need to restart the inverter somehow.

Sure, here you go!
52c92b14-1954-4619-b475-be9f00de6498

Note I also have the Huawei running on the code.

Br
Jørgen

biemond, 2024-04-09:
yeah send it to you, huawei I see indeed.   I think we need to restart solax. it is stuck or has a new ip or something.

JorgenSeemann, 2024-04-09:
Hm. Well. I can try. I don't think it will do anything as the other modbus
connection is working fine, but lets see.
Brb

biemond, 2024-04-09:
yeah, I think solax cache or something is acting weird.  clearing the memory can help here.

JorgenSeemann, 2024-04-09:
I tried the remote off/on. Didn't change anything. I'll try the proper
shutdown when I'm home to night, as I'm unsure if I can start it remotely
after a shutdown

JorgenSeemann, 2024-04-09:
Hi Edwin.
I have the inverter an my wifi rebooted.
App seems to be responding now 😀
Would be nice to have a decimal on the battery voltage, and somehow a
(calculated?) value for solar power.
I'll upload pictures as soon as possible
Cheers

biemond, 2024-04-09:
cool.  please show me pics indeed.  

and put the refresh to 30 sec instead of 20s.  maybe it was too much.  or it was doing things together.  

I need to do 1 more test with this big numbers.  can you try to retrieve 1 of the values so I can compare

JorgenSeemann, 2024-04-09:
Pictures uploaded
Re poll time - I don't see anywhere this can be changed
Re big numbers. Hm. Change how?

Cheers

biemond, 2024-04-09:
yeah , for the big numbers they use 2 bytes with big endian or little.  

Do you know 1 of those total values what is the real value so I can try to come close.

biemond, 2024-04-09:
will do a test now. to see if I get better or more realistics totals

biemond, 2024-04-09:
```
Connected ...
SolarChargerUseMode: 0
output power limits precent 0~100: 100
Manual mode: 0
Grid connected pull in power point: 2000
Battery1 Type: 0
Charge_floatVolt: 7350
Battery_DischargeCutVoltage: 6050
Battery_ChargeMaxCurrent: 200
Battery_DischargeMaxCurrent: 200
absorpt_voltage: 7350
Inverter Voltage: 2366
Inverter Current: 11
Inverter Power: 401
radiator temperature: 41
Powerdc1: 0
Powerdc2: 0
BatVoltage_Charge1: 7312
BatCurrent_Charge1: -4
Batpower_Charge1: -358
TemperatureBat: 0
Battery Capacity: 0
OutputEnergy_Charge_today: 144
InputEnergy_Charge_today: 162
Feedin power is obtained from Meter or CT: 0
energy today to the grid: 6915
energy today from the grid: 61453
energy to the grid: 57698
energy form the grid: 55972
Today's Solar Energy: 300
Total Energy (Inverter AC Port): 6956
GridVoltage_R(X3): 2366
GridCurrent_R(X3): 11
GridPower_R(X3): 122
GridVoltage_S(X3): 2354
GridCurrent_S(X3): 13
GridPower_S(X3): 179
GridVoltage_T(X3): 2363
GridCurrent_T(X3): 11
GridPower_T(X3): 89
FeedinPower_Rphase(X3): -257
FeedinPower_Tphase(X3): 256
EchargeToday: 0
EchargeTotal: 35328
SolarEnergyTotal: 28719
SolarEnergyToday: 332
BMS_UserSOC: 0
BMS_UserSOH: 0
ActivePowerTarget: 0
Charge_Discharg_Power: 0
SocUpper: 100
SocLower: 10
Cell_Voltage_High: 0
Cell_Voltage_Low: 0
Off-gridVoltage(X1): 0
Off-gridCurrent(X1): 0
Off-gridPower(X1): 0
BusVolt: 0
```

biemond, 2024-04-09:
new version https://homey.app/a/solaredge.modbus/test/

JorgenSeemann, 2024-04-09:
So perhaps the "total yield" value should be 1129.9kWh
or the "export control" that is set to 300000 (W) ?

[161 more replies]