biemond/solaredge.modbus #71: NO Data from the Huawei inverter

Daniels1959, 2024-03-28:
ik heb juist de App #SolarEdge + Growatt TCP modbus # geinstalleerd , maar er komt geen Data binnen van de Huawei inverter 
[image]

hansvanveer, 2024-04-01:
Dear all, 
Dear @biemond,

Maybe the following comment and error comm can help you further.
Thanks to @biemond, I was able to get both of my Huawei Solar inverters back into the Homey system through TCP ModBUS approach after a disrupt of a previously used Homey community app using the FusionSolar read-out.
Both of the inverters are connected through only one Dongle. I managed to get both of them at least initiated and able to read out on a LAN address, on port 502. The first inverter, where the dongle is plugged into - the master inverter, lives at InverterID '1', and the second inveter i could approach by adding a second device using the same ip address, same port 502 and inverterID '2'. Usually after a few seconds (i assume the ModBus readout interval as programmed in the app), all values appear.

Unfortunately,  after an initial succesfull day, the next morning, after returning from the night standby state, Homey seems to be able to reconnect fluently only to the slave inverter (ID '2'). The master inverter stays in the frozen state with the latest exchanged values of the last active days.
The only way to get the master inverter back nto the Homey-game is to delete that device and reinstall a new device on the same IP and deviceID - refreshing the device nor rebooting the app brings the master inverter back into Homey.

Using the Power by the Hour app to summarize the generated power produced by both inverters, in combination with a Shelly 3EM in my fuse box, i was hoping to be able to distract these results to do some power steering in my home. Unfortunately, due to the miscalculation of the skipped master inverter, this is not yet possible. 

I include the code of the error report for further reference: 1ef63009-ece0-44e9-bf48-be4a55ec0c3b and a screenshot illustrating the last loaded values date from now 3 days ago.

So to get a long story short: 
@Daniels1959: make sure you have the correct IP, port 502 (some white papers also suggest 6607) and device ID, depending on your local config; 
@biemond : is there any way to reinitiate the connection the next morning when the master inverter restarts? seems the communication with/through the dongle is not the issue, as the slave inverter can be read out correctly. would you able to check the log?

Many thanks for your efforts! 
[image]

biemond, 2024-04-01:
hey, 

so every 30 or 1 min I make a new connection to your device / sdongle, there is no re-use or caching of connection. 

only I can think of that s-dongle cannot handle 2 connections at some point . currently I don't support 1 connection over 2 devices.  
I think you can also do an inverter directly or not.  skip sdongle.

biemond, 2024-04-01:
@Daniels1959 can you send me a diagnostic report.

Daniels1959, 2024-04-01:
[image]

Daniels1959, 2024-04-01:
I have the error # Oeps the App is not working

Daniels1959, 2024-04-01:
2132056b-bd60-49dd-b07c-f747c2549886

biemond, 2024-04-01:
thanks 

```
[Driver:huaweiwithbatt] [Device:d34dbb7b-972d-4c6a-8024-8d642fee0925]  192.168.1.53
Error: getaddrinfo ENOTFOUND  192.168.1.53
    at Remote Process
 {
  code: 'ENOTFOUND'
}
Client closed

```

Daniels1959, 2024-04-02:
Hoi Edwin, kan ik hier nog iets aan doen ?

biemond, 2024-04-02:
heb je een firewall?  je kan proberen dit adres te pingen en kijken welke poorten open staan.
in ieder geval, homey kan dit ip niet bereiken.

Daniels1959, 2024-04-04:
[image]

Daniels1959, 2024-04-04:
hoi Edwin bij pingen naar het IP adres zie ik geen probleem , , ook Firewall staat uit.

biemond, 2024-04-04:
apart, lijkt erop dat homey het niet kan bereiken.   gebruik je sdongle en heb je geen ander programma die deze ook uitleest?
sdongle kan maar 1 connectie te gelijk aan.