biemond/solaredge.modbus #73: Can you implement "Production Limit" to earn money ;) ?

richard-e5, 2024-04-26:
Hello,

If the electricity is negative I would like to charge my battery. If I could set the "Production limit" to 0 Watt, then my battery can be charged with AC.

Kind regards,
Richard

[image]
[image]

biemond, 2024-04-26:
can you try these, it should already be supported

charge from pv, grid.
[image]

[image]

[image]

[image]

biemond, 2024-04-26:
need to check what production control does but should be almost the same 

```
      if (type == 'limitcontrolmode') {
        // 0 – Disabled
        // 1 – Export Control
        // 2 – Production Control
        // 11 – Minimum Import Control
        if (value == 1) {
          const limitcontrolmodeeRes = await client.writeSingleRegister(0xe000, Number(1));
          console.log('limitcontrolmode', limitcontrolmodeeRes);
          // side = 0
          const limitcontrolsideRes = await client.writeSingleRegister(0xe001, Number(0));
          console.log('limitcontrolside', limitcontrolsideRes);
          // 3000
          const limitcontrolWattRes = await client.writeMultipleRegisters(0xe002, [ 32768, 17723]);
          console.log('limitcontrolwatt', limitcontrolWattRes);
        } else if (value == 11) {
          const limitcontrolmodeeRes = await client.writeSingleRegister(0xe000, Number(2049));
          console.log('limitcontrolmode', limitcontrolmodeeRes);
          // 500
          const limitcontrolWattRes = await client.writeMultipleRegisters(0xe002, [0, 17402]);
          console.log('limitcontrolwatt', limitcontrolWattRes);
        } else {
          const limitcontrolmodeeRes = await client.writeSingleRegister(0xe000, Number(0));
          console.log('limitcontrolmode', limitcontrolmodeeRes);
        }
      }

```

richard-e5, 2024-04-26:
Thank you for the quick reply.

I tested your setup, but I can not charge the battery with AC.

With Export Limit the battery will charge with PV. With Product Limit the battery wil charge with AC. And if the price is negative I will earn some money ;)

[image]

richard-e5, 2024-04-26:
yes I can charge from AC, but alway first from PV. I can not choose for AC only. So turn off the SunCells and use only AC.

When I set "Production limit" to 0 Watt (in the SolarEdge install app) then there is no PV, so then the battery can only charge from AC.

[image]

biemond, 2024-04-26:
ok, why do you not use Charge from PV+AC?  
can you try active power to 0 , see if that works. 

have to check for production limit.

richard-e5, 2024-04-26:
I have checked that. Still charge from PV (as picture "With Export Limit" above)

richard-e5, 2024-04-26:
When I set "Production limit" to 0 Watt, it will charge from AC.

biemond, 2024-04-26:
nice, will look at it, at least you are unblocked.

by the way, I think price need to be , negative -14 c or not with the transport cost. 

but nice

richard-e5, 2024-04-26:
Yes I have read that too, but not all think the same. Some say €0 and some say -€0,14. So I have to test that also, when price is negative. But thank you for your information and your thoughs!!

richard-e5, 2024-05-01:
I asked Zonneplan for the price, when to stop de SolarEdge. The guy said by -€1 ;) 

I tested "Active Power Limit to 0":
1. For an inverter from 2019 it worked.
2. But the new one (2024) with the option "Production Limit" it did not work.

I hope you have an other idear ;) Thanx in advance.

biemond, 2024-05-01:
I think on the inverter you have to do something to enable active power or not. 

I did not implement production limit,  let me check it out.

biemond, 2024-05-01:
-1 euro is crazy , you get broke on it.  -14ct should be enough to stop and start importing it.

biemond, 2024-05-01:
can you try this https://homey.app/a/solaredge.modbus/test/

<img width="394" alt="Screen Shot 2024-05-01 at 20 07 04" src="https://github.com/biemond/solaredge.modbus/assets/1085004/28223ef4-d1ec-4720-8225-ea
[image]
dcb381e360">

biemond, 2024-05-01:
[image]

richard-e5, 2024-05-06:
THANX!! Ik ga het opnieuw testen ;)

[image]
[image]
[image]

green-us, 2024-05-21:
FYI: I got a working power limit on a 2022 HDwave (without battery), indeed by simply setting Active power limit to zero. To set it up I did need to reset the inverter to default settings and enable all modbus and power limiting functions but it was functioning perfectly. However (although nothing on my side changed) this specific option now stopped working. I think since I tested the improved tarif optimisation built in in the MySolaredge app. That works but now the modbus function doesn't anymore. May be related. (topic on tweakers: https://gathering.tweakers.net/forum/list_message/79053502#79053502 .

green-us, 2024-05-21:
here overview settings screenshots with actually tested and working active power limit manipulation using Homey:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

biemond, 2024-05-22:
Also heard this from others, was working fine until solaredge came with this feature.  I think you are the fourth person talking about this.  I think it is a bug with their new app eature or they want you force you the use the app. 

or when you do it once and go back , something is sticking and auto reverted.

green-us, 2024-05-22:
Solution (in dutch):

Wifi AP activeren (schakelaar anderhalve tel naar links)
inloggen
naar webgui navigeren (172.16.0.1)
naar 'power control'
vervolgens 'load defaults'
edit: That's enough. 
(de settings onder 'reactive power' mag ook gewoon op default blijven staan (cosphi) in tegenstelling tot mijn eerdere screenshots. Die deed dus niets. Simpelweg het naar default zetten van de power control does the trick.

biemond, 2024-05-22:
na dit is alles ok voor jou? en active power werkt weer

green-us, 2024-05-23:
Inderdaad. Lees nu je vraag en probeer het nu nog eens en ja, werkt nog steeds.

biemond, 2024-05-23:
[image]