biemond/solaredge.modbus #75: Totale dagopbrengst

MarcovBarneveld, 2024-05-05:
Beste,

Ik heb mijn SE3500 een reset naar fabrieksinstellingen gegeven.
Alle tellers van de inverter staan daardoor dus weer op nul.
Nu zie ik in insights op dat moment een hele grote dip bij de dagopbrengst ter waarde van de totale opbrengst op dat moment.
Kan dat komen doordat je deze waarde in de app berekent uit de totale opbrengst?
2 vragen:
1) Is het mogelijk om deze dag waarde uit de SolarEdge inverter uit te lezen om het beschreven probleem in het vervolg te voorkomen? Of op een andere manier oplossen?
2) is het mogelijk om de 'foutieve' data uit de homey te wissen zodat insights die dip niet meer toont en verticale as dus tussen de 0 en de 30 uitkomt?

Met vriendelijke groeten,
M. van Barneveld.

[image]

biemond, 2024-05-05:
hey,

ja die bereken ik , ik kijk naar de stand van 12uur. Solaredge heeft geen dag register.

insights is een database maar ik heb er geen toegang tot.  Met homey 2023 is er volgens mij wel meer mogelijk.  in ieder geval ik kan je helaas hier niet mee helpen.

Gr Edwin

MarcovBarneveld, 2024-05-08:
Beste Edwin,

 

Dank voor je antwoord.

Kan je dan in ieder geval voorkomen dat deze waarde opgeslagen wordt?

Het is nou niet echt een ‘realistische’ waarde.  

Voor een systeem als deze is een reële dagopbrengst misschien max 25kW(?) per dag?

 

Met vriendelijke groeten,

M van Barneveld.

MarcovBarneveld, 2024-05-19:
Beste Edwin,

 

Ik bedenk me ineens dat een negatieve opbrengst niet eens mogelijk is.

Zou je er voor kunnen zorgen dat negatieve opbrengsten zoals in dit geval niet worden weergegeven of als 0?

Ik wil de andere omvormer namelijk ook resetten, daar wil ik even mee wachten totdat dit is gefixt.

 

Marco

MarcovBarneveld, 2024-05-26:
Hoi Edwin,

 

Kan je bevestigen of ontkennen dat je hiervoor een aanpassing gaat doen.

 

Met vriendelijke groeten,

M van Barneveld.

biemond, 2024-05-26:
kan hem dan 0 of in ieder geval, niet negatief maken. 

wat is eigenlijk de reden voor de reset van de inverter?

MarcovBarneveld, 2024-05-27:
Hoi Edwin,

 

Dank je wel.

Na een van de laatste firmware updates zag ik rare uitval dips van 5 – 10 sec in het opwekken, meestal bij (vrijwel) vol vermogen en nog voordat de 253V langdurig werd aangetikt. Dit zag ik ook als ik op de omvormer zelf was ingelogt. Ook kwamen niet alle meldingen meer door, zowel in de omvormer zelf als naar de Homey.

Omdat ik wat had zitten spelen met de instellingen voor vermogens beperking heb ik besloten tot een reset. 

In de status logging van de omvormer zie ik nu na de reset ook weer de meldingen van >253V.

Hij lijkt het nu weer beter te doen, meldingen komen ook weer door in Homey.

Sowieso lijkt de status “Fault” (bij >253V --> netbescherming) verdwenen.

 

Marco

biemond, 2024-05-27:
interessant , maar ik vermoed dat er iets is veranderd in je buurt.  denk niet dat jij dat kan veroorzaken , tenzij je panelen uitzet. denk dat ze je buurt transformator iets moeten aanpassen.   

kreeg je 253 op alle fases ?  bij mij hadden ze een nieuwbouw wijk allemaal op fase 1 aangesloten.

MarcovBarneveld, 2024-05-30:
Ik zit ook nog op 1 fase met een oude meter, dat houdt ik nog even zo.

Zelfs zonder mijn panelen gaat het hier af en toe ruim boven de 253

Ik hoor graag wanneer je het hebt aangepast, dan ga ik resetten, proefondervindelijk kijken of het werkt 😊

MarcovBarneveld, 2024-06-05:
Beste Edwin,

 

Ik zie dat er wel als een béta klaar staat, maar zitten daar de aanpassingen al in waar we het hier over hebben?

Ik hoor graag, dan kan ik resetten.

 

Zo ziet de solar power er nu uit.

Vreemde dips met een trip op 13:00.

Interessant niet?

 

Marco

biemond, 2024-06-05:
hey,

kan het plaatje niet zien als je via email replied.  
zal vanavond eens kijken naar die dag opbrengst

MarcovBarneveld, 2024-06-05:
[image]

biemond, 2024-06-05:
ja heb je je fase spanning al gecheckt, plus was er toen een wolk .  Plus heb je 3 or 1 fase inverter. je moet namelijk er voor zorgen dat de spanning van een string niet onder 750v komt.  en het lijkt dat hij weer langzaam opschaalt to the max.

MarcovBarneveld, 2024-06-05:
Hoi Edwin,

 

Als er een wolk zou komen zie je geleidelijke afname niet geheel en direct tot 0 terug vallen zou je zeggen.

Als de SE3000H enkelvase omvormer tript op de spanning zie je inderdaad een langzame opbouw zoals bij de laatse dip, niet een plotselinge terugkeer zoals in de andere voorbeelden. Op dat moment zie ik overigens ook een piek op mijn verbruiksmeter, dus het is geen bug in de app of zo, hij valt echt helemaal weg.

Ik heb 9 x S500B voor korte string (vanaf 7 panelen) ik heb er 9.

Ook is de nom DC spanning van de omvormer 380V, die wordt zelfs al gehaald als de zon net op is.

Ik zie ook geen rare pieken of dalen in de grafiek van de optimizer spanningen. Hier en daar wel een dipje, maar dat wordt keurig opgevangen door de andere.

 

Bedankt voor je meedenken,

Ik hoor graag als je de aanpassing hebt gedaan.

Als het na de reset nog steeds gebeurt, zal ik SE eens contacteren, misschien weten zij het.

 

Marco

biemond, 2024-06-05:
kan. je deze proberen https://homey.app/a/solaredge.modbus/test/

als het > 0 dan gebuik ik het anders is hij 0

biemond, 2024-06-05:
we hebben maar 1 kans denk ik , het gaat goed of fout

MarcovBarneveld, 2024-06-08:
Hoi Edwin,

 

Hij doet het!!! Heel hartelijk dank.

Viel me wel op dat direct na de reset de rest van de dag de opbrengst 0 was en bleef, maar dat was te verwachten.

 

Volgens mij zit er ook iets nieuws in: “Power reduce limit”, of dat is mij niet eerder opgevallen. 

Wat is het verschil met “Active power limit”?

 

Zou je dan misschien de Cos Phi er dan ook nog eens bij kunnen zetten als je tijd hebt?

 

Met vriendelijke groeten,

M van Barneveld.

biemond, 2024-06-08:
mooi, power reduce was vanwege bij sommige de active power limit niet meer werkte. blijkt nu een bug te zijn nadat je in de solaredge app iets hebt aangepast.

ik zal een kijken naar Cos Ph