biemond/solaredge.modbus #76: Support for Huawei Battery Control

mackewigren, 2024-05-07:
Hi and thanks for a nice app. Just found it and worked at once with Modbus on my inverter. 

But I really would like to be able to control charging and discharging of the battery. Is it possible to add these functions for Huawei battery in the near future?

biemond, 2024-05-07:
yeah need to find out how to do authentication thing or is it allowed without a login and confirmation for you?

```
       async def _do_login():
            # Get challenge
            challenge_request = PrivateHuaweiModbusRequest(36, bytes([1, 0]), slave=slave or self.slave)

            challenge_response = await self._client.execute(challenge_request)

            assert challenge_response.content[0] == 0x11
            inverter_challenge = challenge_response.content[1:17]

            client_challenge = secrets.token_bytes(16)

            encoded_username = username.encode("utf-8")
            hashed_password = _compute_digest(password.encode("utf-8"), inverter_challenge)

            login_bytes = bytes(
                [
                    len(client_challenge) + 1 + len(encoded_username) + 1 + len(hashed_password),
                    *client_challenge,
                    len(encoded_username),
                    *encoded_username,
                    len(hashed_password),
                    *hashed_password,
                ]
            )
            await asyncio.sleep(0.05)
            login_request = PrivateHuaweiModbusRequest(37, login_bytes, slave=slave or self.slave)
            login_response = await self._client.execute(login_request)

            if login_response.content[1] == 0:
                # check if inverter returned the right hash of the password as well
                inverter_mac_response_lengths = login_response.content[2]

                inverter_mac_response = login_response.content[3 : 3 + inverter_mac_response_lengths]

                if not _compute_digest(password.encode("utf-8"), client_challenge) == inverter_mac_response:
                    LOGGER.error(
                        "Inverter response contains an invalid challenge answer. This could indicate a MitM-attack!"
                    )

                return True
            return False

        async with self._communication_lock():
            LOGGER.debug("Logging in")
            return await _do_login()
```

mackewigren, 2024-05-12:
HiI am not sure about this. I have never used Modbus to communicate (except via your app). What I can find is this link about Modbus communication. And I know it works with Home Assistant. I read a little about that integration. Seems like we need login for changing settings. Most inverters have the login password 000000a but not all. And as you can see in the Huawei link I just choose Installer from a dropdown. forum.huawei.comHope this helps. Best regards /Marcus Wigren7 maj

biemond, 2024-05-12:
from the official app , can you just change something and does it ask for some login again?

I can enable it again , it is already there

mackewigren, 2024-05-12:
HiAs if today I change timer settings every day and never need to login again.

mackewigren, 2024-05-15:
If you can activate it again, I will be happy to try it as soon as possible. Just let me know when it’s done in that case

mackewigren, 2024-05-17:
> (Issue #76)
>
>
>
> can you try this one https://homey.app/a/solaredge.modbus/test/
>
> it on the device
>
> 
> <https://github.com/biemond/solaredge.modbus/issues/76#issuecomment-21131657
> ORNDAVCNFSM6AAAAABHLSI6D2VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDCMJTGE3DK
> NZSG4> .
> <https://github.com/notifications/beacon/ANZQJILXYY7CNOMWODYMY33ZCORNDA5CNFS
> M6AAAAABHLSI6D2WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT
>
>
> <https://djXrJB04.eu1.hs-salescrm-engage.com/Cto/5H+23284/djXrJB04/R5R8b40lN
> N7p53sk2fJctW1T_X-g3H34lTW20ZrBB1X2cQ_W1Gy2yZ1WZd6qW1X2cGr24TV8bW3GJsPW24W6F
> Vn22X81-4W1>

Hi

 

I have downloaded it. I cannot see that I can control anything from flows
yet. But it looks like I can control it from the device.

 

But there are many ways that look very similar. Do you know how I should do
the settings? It looks like the images below (but guess you know that
offcourse).

 

Best regards

 

Marcus

biemond, 2024-05-17:
can you try something, see if it sticks and not being reverted.

mackewigren, 2024-05-22:
HiI have done sporadic tests but have had too much to do at work. Going away over the weekend but scheduled to test on Monday. Best regards/Marcus Wigren17 maj

mackewigren, 2024-05-29:
I have now tested several options and cannot find any way to get the settings to work right now.

biemond, 2024-05-29:
so api handles this auth for you and probably modbus it will denied , let me revert it for now.

mackewigren, 2024-05-29:
Do not trust my tests fully. Would be nice if someone else could try too