biemond/solaredge.modbus #77: Piek in Solar Power

MarcovBarneveld, 2024-05-08:
Beste Edwin,

Toen ik maandag morgen de inverter 'opende' om te kijken wat de status was zag ik dat hij exact op dat moment 3000 W produceerde, heel bijzonder om 7:32. 3000W is het ingestelde maximum in de app.
Ik heb dit wel vaker gezien op het moment dat ik het apparaat opende denk ik, en dan ook naar 0 toe, terwijl hij gewoon aan het opwekken is.

Misschien kan je hier iets mee.

Met vriendelijke groeten.
[image]

M van Barneveld.

biemond, 2024-05-08:
volgens mij zijn dat de optimizers icm een soort relais. soms kun je dat ook krijgen bij zware bewolking.

je kan het beperken via je settings.

[image]

MarcovBarneveld, 2024-05-08:
Hoi Edwin,

 

Dat heb ik gedaan, 3000W.

Het is alleen wel heel toevallig dat deze piek optreedt op het moment dat ik het apparaat in Homey open, en volgens mij niet de eerste keer dat ik dit zie.

Ik had in de changelog iets gezien over onrealistische waarden daar vond ik dit wel bij horen, vandaar dat ik dit met je wilde delen. 

Ik houdt het in de gaten.

 

Marco

 

Screen.Shot.2024-05-08.at.15.09.22.png (view on web) <https://github.com/biemond/solaredge.modbus/assets/1085004/07b569af-f6ef-4b83-b5d3-de79f2dd1bcd>

biemond, 2024-05-08:
goede  timing denk ik en je override werkt goed  anders was het 10k of zo.
maar ik geef alleen maar door wat solaredge mij aanreikt. Ik verzin geen waarden.

maar ik heb het nooit , maar het kan met die relais te maken hebben en de optimizers die opstarten.  misschien een bepaalde batch met bepaalde onderdelen zijn gevoelig voor dit.